realsenseai/librealsense #7612: D435i cannot access IMU data on Mac

Jinghe-mel, 2020-10-19:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |   MacOS  | 
| Platform                                 | PyCharm|          | 
| Language                            |  opencv/python                          | 

### Each time I try to access that IMU data, there will be an error: 
#### "Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)"
#### It works well on colour/depth frame. I am using a Macbook and working on Pycharm.

#### The code I am using:
import sys
import numpy as np
import cv2
import pyrealsense2 as rs

def initialize_camera():
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 250)
    config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)
    pipeline.start(config)
    return pipeline

def gyro_data(gyro):
    return np.asarray([gyro.x, gyro.y, gyro.z])

def accel_data(accel):
    return np.asarray([accel.x, accel.y, accel.z])

p = initialize_camera()
try:
    while True:
        f = p.wait_for_frames()
        accel = accel_data(f[0].as_motion_frame().get_motion_data())
        gyro = gyro_data(f[1].as_motion_frame().get_motion_data())
finally:
    p.stop()

MartyG-RealSense, 2020-10-19:
Hi @Jinghe-mel   There was a past case that used a script very similar to your one and also had problems.  A solution was suggested in the link below that corrected the problem by making edits to the **accel =** and the **gyro =** lines near the bottom of the script.

https://github.com/IntelRealSense/librealsense/issues/4018#issuecomment-520827544

It is also a widely practised  pyrealsense2 programming convention to have **import pyrealsense2 as rs** as the first line above all the other import statements.

MartyG-RealSense, 2020-10-19:
Hi @Jinghe-mel   The RealSense SDK release notes state that D435i is not supported on Mac yet.

https://github.com/IntelRealSense/librealsense/releases

[image]

Jinghe-mel, 2020-10-20:
Hi, @MartyG-RealSense Thank you for the reply.

closed 2020-10-20 by Jinghe-mel

MartyG-RealSense, 2021-02-15:
Hi again.  The new librealsense version **2.42.0** has introduced IMU support for MacOS.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0