realsenseai/librealsense #7614: config::enable_device breaks RS_OPTION_GLOBAL_TIME_ENABLED

martomi, 2020-10-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                              | D455 | 
| Firmware Version                         | (Open RealSense Viewer --> Click info) | 
| Operating System & Version       |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)         |  5.4.0-51                                         | 
| Platform                                         | PC |
| SDK Version                                  |  2.39.0                          | 
| Language                                      |   C++                           | 
| Segment			                 |  Robot                          | 

### Issue Description

Using the pipeline API, I'm calling `config.resolve(pipeline)` to get the device and apply some configurations before streaming. 

I noticed that when I call `frame.get_frame_timestamp_domain()` inside the pipeline callback, I'm always getting `Global` regardless of how I configure the setting `RS_OPTION_GLOBAL_TIME_ENABLED`.

I've tried making the configuration before or after `config.enable_device(...)` call. As well as before and after starting the stream. The only way I found to "fix" this is to avoid calling `config.enable_device(...)`.

Other options (e.g. RS2_OPTION_EMITTER_ENABLED) don't seem to be affected by this issue.

MartyG-RealSense, 2020-10-20:
Hi @martomi   I note that you quote the setting **RS_OPTION_GLOBAL_TIME_ENABLED**.   In C++ the function is called **RS2_OPTION_GLOBAL_TIME_ENABLED**.  May I confirm please that you have defined it as RS2 in your application? 

http://docs.ros.org/en/kinetic/api/librealsense2/html/rs__option_8h.html#a8b9c011f705cfab20c7eaaa7a26040e2a19afe71bbf8d3531d036005055e35199

martomi, 2020-10-20:
Hi @MartyG-RealSense , yes I have used the correct enum name in my application. It seems I made a typo while writing the JIRA issue - will correct that now :-)

MartyG-RealSense, 2020-10-20:
I researched your problem further and found a C++ example that a RealSense user developed for disabling RS2_OPTION_GLOBAL_TIME_ENABLED with callback:

https://github.com/IntelRealSense/librealsense/issues/5067#issuecomment-544548652

pipe.start(callback) starts the pipeline with a callback, whilst pipe.start(callback, cfg) does the same but also makes use of the cfg configurations.

https://github.com/IntelRealSense/librealsense/issues/3320#issuecomment-466882443

martomi, 2020-10-20:
Thanks for taking a look into it. I'm aware of these methods for setting the option. They do work but only as long as I don't call `config.enable_device(...)` prior to running `pipe.start(...)`. And this is also the essence of the reported issue. It's the unexpected behaviour/side-effect caused by `config.enable_device(...)`.

martomi, 2020-10-20:
To be a bit more specific, I would expect this order of actions to result in working configuration:

```
rs2::pipeline pipe;
rs2::config cfg;

cfg.disable_all_streams();
cfg.enable_stream(...);
cfg.enable_stream(...);

auto profile = cfg.resolve(pipe);
auto device = profile.get_device();

// This code snippet only works if I remove this call to enable_device(...)
config.enable_device(device.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

for (auto& sensor : device.query_sensors()) {
    sensor.set_option(RS2_OPTION_GLOBAL_TIME_ENABLED, 0);
}

pipe.start(cfg, callback);
```

Note: This is pseudo-code which might have some small typos and is meant just for high-level explanation.

MartyG-RealSense, 2020-10-20:
Noting that the above script is psuedo code and there is the possibility for typos ... I believe that **config.enable_device** would be **cfg.enable_device** if you have been using the cfg term up until this point.

If you only have one camera attached then I would not expect to have to query for the serial number, since serial numbers enable programs to differentiate between devices if more than one is attached.  The code sample below for getting the first RealSense device (the only device if only one camera is attached) is quoted from the documentation:

`rs2::context ctx;`
`auto list = ctx.query_devices(); // Get a snapshot of currently connected devices`
`if (list.size() == 0) `
`throw std::runtime_error("No device detected. Is it plugged in?");`
`rs2::device dev = list.front();`

MartyG-RealSense, 2020-10-20:
In any situation that involves multiple cameras and their serial numbers, the multicam querying code of **rs-multicam** is an excellent reference to check against your own code:

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L21

martomi, 2020-10-20:
@MartyG-RealSense, I appreciate your efforts to provide me with workaround to the issue. 

However, I already have a workaround which works for me at the moment so that's not what I'm looking for here. I'm reporting a flaw in the API.

The approach I have outlined in the pseudo-code snippet above should intuitively work, don't you agree? At least according to the documentation of the methods that I'm using (and which I've thoroughly read).

It is a bad API design if something behaves unexpected and is not documented, even worse if it fails silently (which it does). So I'm hoping you recognize that and consider investigating the issue to the bottom rather than focusing on workarounds. 

If it'll help move it forward, I can extend the above pseudo-code to a working code example. Let me know.

MartyG-RealSense, 2020-10-20:
I understand your frustration.   I am not on the development team, so a member of the RealSense team who is a specialized developer may be able to better advise you.  I will seek assistance for your case.  Thanks for your patience in the meantime.

RealSenseSupport, 2020-10-21:
It should work if configure this option after pipeline is started. 
Please see if the following also works on your side. Thank you.

```
    rs2::pipeline pipe;
    rs2::config cfg;

    cfg.enable_stream(RS2_STREAM_DEPTH);

    auto profile = cfg.resolve(pipe);
    cfg.enable_device(profile.get_device().get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));

    //starting the pipeline
    auto pipe_prof = pipe.start(cfg);
    auto depth_sensor = pipe_prof.get_device().first<rs2::depth_sensor>();
    //Disable Global Time Domain
    if (depth_sensor.supports(RS2_OPTION_GLOBAL_TIME_ENABLED))
        depth_sensor.set_option(RS2_OPTION_GLOBAL_TIME_ENABLED, 0.f);
    //Verify clock domain in metadata
    auto data = pipe.wait_for_frames();
    printf("%s\n", rs2_timestamp_domain_to_string(data.get_depth_frame().get_frame_timestamp_domain()));
```

closed 2020-11-07 by MartyG-RealSense