realsenseai/librealsense #7616: T265 outdoors drift

dronus-lincio, 2020-10-20:
hello,
I have questions on how to minimize the drift experienced with t265 outdoors, particularly when the camera is quite still. 
- Which are the options available? Is mapping feature enabled by default?
- Is it possible to get numeric values for tracking confidence, instead of a qualitative indication (low, medium, high)?
- Are there other indicators of the quality of SLAM?
Thanks,

zainmehdi, 2020-10-22:
Sadly Realsense Team isnt really active on supporting T265 related issues. They have kind of dumped it. What you can do is to turn off pose jumping and re localization and make sure there are good enough features around you. Some drift in long running experiment is inherent and cant be avoided until you close the loop which I wont recommend in T265's case.

dronus-lincio, 2020-10-26:
@zainmehdi thanks for your comments. Anyway I was already with pose jumping and relocalization turned off, and features were coming from a garden and a wall, so I would expect good keypoints.
What I am looking for is the presence of more knobs to turn, so to make my system more robust. That's why I need to understand if I can get more information out from the sensor.

rlabs-oss, 2020-11-19:
Over a 30 meter track (width) returning to the same point (expecting loop closure) I saw an error of 5.9 meters which is just unacceptable for Lidar mapping in daylight (overcast). 
[image]

closed 2026-10-23 by MartyG-RealSense