realsenseai/librealsense #7617: Intel d455 is not working

naruto-uzumakki, 2020-10-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  16.04                                         | 
| Platform                                 | NVIDIA Jetson TX2   |
| SDK Version                            |   2.39.0                     | 
| Language                            | c++                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
i cannot open realsense-viewer and when try to read thr video stream of the camera through my c++ code , it shows No Device Connected but  my d415 is working properly . iam using usb 3.0 connector and realsense - viewer version is v2.39.0

naruto-uzumakki, 2020-10-21:
i installed as per the git page of intel realsense sdk installation https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

dorodnic, 2020-10-21:
Please un-install using `sudo make uninstall` and follow the Jetson guide - 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

naruto-uzumakki, 2020-10-22:
Hi ,@dorodnic i tried to work with my ubuntu -18.04 PC , it woked  with sdk v2.36.0 , suddenly error i below mentioned came and i cant solve that error still , please help me solve this error 

2020-10-22 11:15:26,329 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:15:32,448 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:15:38,568 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:15:50,863 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:15:56,964 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:16:03,085 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:16:15,316 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:16:21,427 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
2020-10-22 11:16:27,536 WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55950c8ee510, config:0x55950c8c5d70):
    No device connected

mohanen, 2020-10-23:
On Connecting Camera

```
$ realsense-viewer
 23/10 12:18:59,168 ERROR [140566390667008] (types.h:313) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid argument
 23/10 12:18:59,168 WARNING [140566390667008] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 23/10 12:18:59,176 ERROR [140566390667008] (types.h:313) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid argument
 23/10 12:18:59,176 WARNING [140566390667008] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 23/10 12:19:04,466 WARNING [140566476357056] (backend-v4l2.cpp:1267) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
```
On trying to turn on depth stream

```
 23/10 12:19:04,466 WARNING [140566476357056] (backend-v4l2.cpp:1267) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 23/10 12:20:13,314 WARNING [140566209713920] (backend-v4l2.cpp:1032) Frames didn't arrived within 5 seconds
 23/10 12:20:18,317 WARNING [140566209713920] (backend-v4l2.cpp:1032) Frames didn't arrived within 5 seconds
 23/10 12:20:20,764 ERROR [140566276855552] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Device or resource busy
 23/10 12:20:21,765 ERROR [140566276855552] (types.h:313) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 23/10 12:20:21,765 ERROR [140566276855552] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 23/10 12:20:22,766 ERROR [140566276855552] (types.h:313) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 23/10 12:20:22,766 ERROR [140566276855552] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 23/10 12:20:23,322 WARNING [140566209713920] (backend-v4l2.cpp:1032) Frames didn't arrived within 5 seconds
 23/10 12:20:23,767 ERROR [140566276855552] (types.h:313) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 23/10 12:20:23,767 ERROR [140566276855552] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 23/10 12:20:24,767 ERROR [140566276855552] (types.h:313) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
```

MartyG-RealSense, 2020-10-23:
Hi @naruto-uzumakki and @dorodnic    I have a Jetson TX2 and D455, so I will take a look at the issue.

MartyG-RealSense, 2020-10-25:
@naruto-uzumakki   @dorodnic    I ran tests with my own TX2.  The RealSense Viewer was able to successfully detect D415 and D435i.   But every time the D455 was inserted the camera did not appear in the options side-panel and an **UNKNOWN in rs2_create_device ... failed to set power state** option appeared, along with a red pop-up error in the top corner that suggested that the udev rules were out of date.  Reloading the udev rules in the terminal did not corect the problem.

MartyG-RealSense, 2020-10-27:
Hi again @naruto-uzumakki   May I confirm please that you are experiencing this error on both Jetson TX2 and your Ubuntu PC?

**WARNING [default] Could not open device set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable**

@mohanen   The errors that you reported look as though they may be a different problem to those reported by @naruto-uzumakki   Could you create a new case for your problem please by visiting the front page of the GitHub forum at the link below and clicking on the **New Issue** button please?   Thanks!

https://github.com/IntelRealSense/librealsense/issues

naruto-uzumakki, 2020-10-27:
Hi @MartyG-RealSense  the error , I and @mohanen  facing right now are  the same. Yes iam facing this in both TX2 and PC , i even tried to upgrade the firmware version but it didn't helped.

MartyG-RealSense, 2020-10-27:
@mohanen   Another RealSense team member has stated that the **Exception was thrown when inspecting Raw RGB Camera property** error that you experience may be generated by a non-RealSense camera that is attached to the computer, such as a USB webcam or a built-in webcam, and the message can be ignored.  Do you have such a camera on your computer in addition to the RealSense camera please (for example, laptops have built-in cameras).

https://github.com/IntelRealSense/librealsense/issues/6351#issuecomment-664553055

MartyG-RealSense, 2020-10-27:
@naruto-uzumakki   Is the D455 detectable if you use the command **rs-enumerate-devices**

naruto-uzumakki, 2020-10-27:
@MartyG-RealSense  yes . D455 is detectable while using rs-enumerate-devices

MartyG-RealSense, 2020-10-27:
Okay, thank you very much.   Next, could you try launching the **rs-hello-realsense** example program with your D455 as a test of whether C++ programs can work at all?  If the program works, it should provide a simple real-time text display of the current detected depth without any graphics.  

I just successfully tested the program in Windows with a D455.

[image]

MartyG-RealSense, 2020-11-02:
Hi @naruto-uzumakki   Can you confirm please if you are still experiencing the same problems in tis case?  

You mentioned earlier that you upgraded your firmware.  Can you confirm if you are now using firmware version **5.12.8.200**

MartyG-RealSense, 2020-11-10:
Hi @naruto-uzumakki   Do you still require assistance with this case, please?    Thanks!

FrankCreen, 2020-11-11:
> Required Info	
> Camera Model	D455
> Firmware Version	5.12.06.00
> Operating System & Version	ubuntu 16.04
> Kernel Version (Linux Only)	5+
> Platform	NVIDIA Jetson TX2
> SDK Version	2.39.0
> Language	c++
> Segment	{Robot/Smartphone/VR/AR/others }
> ### Issue Description
> when i open realsense-viewer it shows ,connect realsense camera or add source and when i tried to read the video stream of the camera through my c++ code , it throws No Device Connected error but the d415 is working properly with both the realsense viewer and code. iam using usb 3.0 connector and realsense - viewer version is v2.39.0

Hi ！I test the code below with a 75FPS, could you show your result? Thanks very much!

## Test Program
```
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

#include <opencv2/opencv.hpp>

#include <chrono>

using namespace std;
using namespace cv;

int main()
try
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 480, 270, RS2_FORMAT_BGR8, 90);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

    //Sensor Settings
    auto sensor1 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor1.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    auto sensor2 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor2.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

    // Display in a GUI
    const auto window_name = "Display Video";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    //用于计时
    //show time
    auto start = chrono::system_clock::now();
    int frame = 0;

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        //获取彩色图像帧
        rs2::frame color = data.get_color_frame();

        //保存彩色图像
        Mat image(Size(480, 270), CV_8UC3, (void *)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);

        frame++;
        //one loop end
        auto end = chrono::system_clock::now();
        auto duration = chrono::duration_cast<chrono::microseconds>(end - start);
        double timecost = double(duration.count()) * chrono::microseconds::period::num / chrono::microseconds::period::den;

        cout << frame / timecost << " FPS" << endl;
    }

    return 0;
}
catch (Exception ex)
{
    cout << "Error!!!" << endl;
    cerr << ex.msg << endl;
}
```

MartyG-RealSense, 2020-11-11:
Hi @FrankCreen    Please post your question as a new issue using the **New Issue** button at the link below and do not post it on multiple D455 cases if the question is unrelated to that case.   Thank you.

https://github.com/IntelRealSense/librealsense/issues/

MartyG-RealSense, 2020-11-16:
@naruto-uzumakki   I did further testing with the D455 on another case with similar errors to yours and found further details about what may trigger the **Exception was thrown when inspecting Raw RGB Camera property** error.  

https://support.intelrealsense.com/hc/en-us/community/posts/360052144833/comments/360013618754

closed 2021-11-16 by MartyG-RealSense