realsenseai/librealsense #7620: RealSense T265 information after ~1 minute without an error

Chris-Bee, 2020-10-20:
Hello everybody,

I posted this issue for the realsense-ros node a week ago but I believe that this issue is probably more related to the driver itself. (https://github.com/IntelRealSense/realsense-ros/issues/1445) I will ask to close the issue for the ROS wrapper.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T256 | 
| Firmware Version                   | v2.39.0 | 
| Operating System & Version |   Ubuntu 18.04 64bit | 
| Kernel Version (Linux Only)    |  5.4.65-v8                                         | 
| Platform                                 | Raspberry PI 4 |
| SDK Version                            |  RealSense ROS v2.2.17                          | 
| Language                            |                            | 
| Segment			|  Robot }                          | 

### Issue Description

All topics published by the realsense-ros node stop publishing after a certain time usually 1-2 minutes. The realsense-ros node does not print an error, journalctl and dmesg also do not show an error. This error is reproducible and occurs if the data throughput on the USB is high, due to other devices.

The interesting part is that, if this error happened and I restart the ROS node, then everything is publishing except for the two fisheye image streams. If I disconnect the RealSense USB connection, reconnect it, and start the ROS node again, then everything is publishing again. It is a bit difficult to debug this since there is no error message.

To prevent power issues, the RealSense is connected to an external USB3 Hub with a standalone power supply [Digitus USB3 Hub](https://www.digitus.info/de/produkte/computer-und-office-zubehoer/computer-zubehoer/usb-komponenten-und-zubehoer/hubs-und-kartenleser/da-70231/)

Let me know if there is another way on how I can further debug this issue.

Iko-Git, 2020-11-02:
I did face the same issue and did extensive debugging to find the problem. As based on the debugging messages, I ended up that the bellow _**default**_ part in  [_realsense-ros/realsense2_camera/src/base_realsense_node.cpp_](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp) ​isn't joined by the _**switch (stream)**_ which belongs to 
https://github.com/IntelRealSense/realsense-ros/blob/b1c02bdf689c2e1b47c88fac098608c1e7964b91/realsense2_camera/src/base_realsense_node.cpp#L1687-L1703
As it's also shown in the same part of the code, we see how _stream_ is defined as following 
**_auto stream = frame.get_profile().stream_type();_**
Questioning what is **_frame_**?!
Particularly it's an object of **_frame_** class which belongs to **_rs2_** library (already in the link below) 
**_rs2::frame frame_**
http://docs.ros.org/en/kinetic/api/librealsense2/html/rs__frame_8hpp_source.html
Consequently, I see the issue now from API level

stephanweiss, 2020-12-01:
Here is the answer from Intel:
" We want to thank you for your inquiry with regards to T265. We are aware of this issue but unfortunately, there is no fix for this. We have moved our focus to our next generation of products and, consequently, we will not be addressing this issue in the T265."

Always happy to be an end-user beta-tester! ;-)

closed 2026-10-23 by MartyG-RealSense