realsenseai/librealsense #7621: Cannot receive depth frames consistently

msaliharslan, 2020-10-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |   Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  5.4.0-51-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2.39.0                          | 
| Language                            |  C++                          | 
| Segment			|  General                     | 

### Issue Description
I was able to run my code and collect depth information from the device then after a few days I was not. During those few days I did not use the camera. On realsense-viewer I am rarely able to receive depth frames for certain image size and frame rate configurations  but not for others. In other words, most of the time realsense-viewer says no frames received regardless of configuration. I tried with other versions of realsense-viewer(2.39, 2.38, 2.27) and on a windows pc the problem persists on those platforms as well. On the other hand, I can consistently receive color and motion frames on realsense-viewer. However, I cannot receive color frame with my C++ code with which I used to.

MartyG-RealSense, 2020-10-21:
Hi @msaliharslan   The fact that you have problems on both Ubuntu and Windows computers would suggest that something external to the computers is involved that is used with both computers - for example, the camera firmware driver (stored inside the camera hardware, not on individual computers) or the USB cable.

If the camera is able to at least be detected, a good first step would be to perform a **Gold Reset** of the camera to return it to its factory-new configuration.  If the **calibration table** stored inside the camera is corrupted then a Gold Reset can also correct tha problem.

You can perform a Gold Reset using a program called **CustomRW** that is part of the **Dynamic Calibrator** camera calibration tool.   You can install this tool on Linux by following the instructions on page 14 onward of the tool's user guide.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Once the tool is installed, you can launch CustomRW in the Gold Reset mode with the command below:

**Intel.RealSense.CustomRW -g**

msaliharslan, 2020-10-21:
I followed the instructions and made a gold reset to the device, but, unfortunately it did not work. Nothing has changed.

MartyG-RealSense, 2020-10-21:
Thank you very much for the tests.  When you use the RealSense Viewer, do you have the camera plugged into a USB 3 port?  If you do, is the camera being correctly detected as USB 3?

You can tell this by looking at the number beside the camera name at the top of the options side-panel.  If it says '3.2' then it is a USB3 connection.  If it says '2.1' then it is being mis-identified as being a slower and more limited USB2 connection.

msaliharslan, 2020-10-21:
[image]
 
Yes

MartyG-RealSense, 2020-10-21:
Thank you.  I see that you have depth, color and IMU all enabled at the same time in your image, and a **Frames not received** error on the depth stream.

Problems when all three of these stream types are active is a known-about issue.  It can be corrected in Python and C++ applications written by RealSense users, but I am not aware of a solution for the RealSense Viewer.  

You should be able to successfully display color, depth and IMU simultaneously in the **rs-multicam** example program though, as tests that I have conducted have consistently shown that it does not experience this problem.  The rs-multicam program can display color, depth and IMU panels from a single attached D435i camera too.

https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

msaliharslan, 2020-10-22:
Screenshot is just a sample, I had tried only enabling the stereo module too, it did not work as well. Also, my own c++ application cannot receive any frames either.  rs-multicam example says "please connect one or more device", however, other examples (rs-align, rs-capture, C/depth, C/distance) see the device but says "Frame didn't arrive within 15000". C/color example runs and gets RGB  frames.

msaliharslan, 2020-10-22:
[image]
[image]

This time I was lucky enough to receive depth frames(still the problem remains). However, I noticed another issue which is the left side of the depth frame is missing as if depth frame is slightly slided to right. Do you think this is related to the original problem? Also, on graph it says that on average 55 frames drop per second, is this normal?

MartyG-RealSense, 2020-10-22:
There have been a couple of recent cases involving **Frames didn't arrived within 15000** that were resolved by updating to the currently latest SDK and firmware.  I see you have the latest SDK **2.39.0**.  Are you also using the latest camera firmware **5.12.8.200** please?

msaliharslan, 2020-10-24:
It is already running with **5.12.8.200**

msaliharslan, 2020-10-24:
I do not know if this has any significance, I hope it is a clue: 848 x 480 resolution is more problematic than the others. I had received depth frames for all other resolutions(still not consistent) but this one. Also, note that after trying to enable depth stream with this resolution, the others do not work either from then on.

MartyG-RealSense, 2020-10-25:
I have investigated your case carefully.  The fact that the camera has problems on both Ubuntu and Windows and on multiple Viewer versions leads me to suspect the camera firmware driver that is stored inside the camera.

As the camera can be detected by the Viewer, could you try the steps below please:

1.  Download the firmware bin file for **5.12.7100** from the link below.

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

2.  Launch the RealSense Viewer program and go to the **More** icon near the top of the options side-panel.  Select the **Update Firmware** menu option.

[image]

3.  A file selection window will open.  Select the **Signed_Image_UVC_5_12_7_100.bin** file from the location that you downloaded it to.  The camera will then downgrade to the 5.12.7.100 firmware version.

Bear in mind that RealSense cameras can only have their firmware downgraded 20 times before they can only be upgraded to a newer version once the limit is used up.  So the downgrade operation should not be performed multiple times.

4.  Test the camera to see whether it now works normally with the RealSense Viewer and your C++ program.  If it does work normally, you can use the **Install Recommended Firmware** option with Viewer version 2.39.0 to re-upgrade to firmware 5.12.8.200.  If it then stops working then you will still have the ability to re-downgrade to 5.12.7.100, as you will have used very little of the camera firmware downgrade limit at this point.

msaliharslan, 2020-10-25:
I am sorry to say that downgrading to 5.12.7.100 via realsense-viewer as you suggested did not change anything.

MartyG-RealSense, 2020-10-25:
Thanks again for the further update.  On Windows, can you confirm whether you experience the problems if you download the standalone executable .exe version of the Viewer from the **Assets** list at the bottom of the details listing for SDK 2.39.0?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.39.0

[image]

The **Intel.RealSense.Viewer.exe** file should be launched from the location that you downloaded it to by double-leftclicking on the file, and not the Viewer launch icon on the Windows desktop.

msaliharslan, 2020-10-27:
When I said I tried on windows I already used that version of the realsense-viewer. Sorry for misleading you.

MartyG-RealSense, 2020-10-27:
It is okay!  :)

May I also ask whether you connect the camera to a USB hub when you use the camera with your Ubuntu and Windows PCs, or do you connect the camera directly to the USB port on the computer?

msaliharslan, 2020-10-27:
Directly to computer

MartyG-RealSense, 2020-10-27:
Thank you very much for your patience.   Next, please see whether the camera is detected if the command **rs-enumerate-devices** is used.

msaliharslan, 2020-10-28:
[output.txt](https://github.com/IntelRealSense/librealsense/files/5454548/output.txt)

Camera can be detected. I added the output of the program also, copied here.

Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	850312070185
    Firmware Version              : 	05.12.07.100
    Recommended Firmware Version  : 	05.12.08.200
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	915323050566
    Firmware Update Id            : 	915323050566

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 25Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 1	  848x100	@ 300Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 15Hz	   Y16
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 1	  640x360	@ 15Hz	   Y8
    Infrared 1	  640x360	@ 6Hz	   Y8
    Infrared 1	  480x270	@ 90Hz	   Y8
    Infrared 1	  480x270	@ 60Hz	   Y8
    Infrared 1	  480x270	@ 30Hz	   Y8
    Infrared 1	  480x270	@ 15Hz	   Y8
    Infrared 1	  480x270	@ 6Hz	   Y8
    Infrared 1	  424x240	@ 90Hz	   Y8
    Infrared 1	  424x240	@ 60Hz	   Y8
    Infrared 1	  424x240	@ 30Hz	   Y8
    Infrared 1	  424x240	@ 15Hz	   Y8
    Infrared 1	  424x240	@ 6Hz	   Y8
    Infrared 1	  256x144	@ 300Hz	   Y8
    Infrared 1	  256x144	@ 90Hz	   Y8
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 30Hz	   Y8
[... 239 lines cut ...]
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 250Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 63Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F

MartyG-RealSense, 2020-10-29:
The next test is to unplug the USB cable's micro-size end from the side of the camera, turn the cable upside down and re-insert it into the side of the camera (the micro-size USB C port has two-way insertion, unlike the large connector that only inserts one way up).

msaliharslan, 2020-10-29:
Well it is not affected by orientation of the cable. Also, I have an update, my c++ code works with librealsense version 2.28.1 but not with 2.38 an 2.39. However, there are regular frame drops both for RGB and depth. On realsense-viewer 2.28.1 depth frames arrive except for 848 x 480 resolution. Moreover, after trying to enable this resolution, frames does not arrive for any other resolutions as well and my c++ code stops working until I un-plug & re-plug the device.

MartyG-RealSense, 2020-10-30:
In the RealSense Viewer, are you able to receive **infrared** frames successfully in the same way that you can with color and motion frames?

Also in the Viewer, could you test whether performing a **hardware reset** before you begin streaming makes a difference?   you can do this by going to the **More** option at the top of the Viewer's options side-panel and selecting **Hardware Reset** from its drop-down menu.

[image]

msaliharslan, 2020-10-31:
Hardware reset has no effect, but after trying to enable at 848 x 480 hardware reset makes other resolutions work just like re-plugging. For infrared frames, they behave like depth, i.e. they work fine for resolutions other than 848 x 480. At 848 x 480 one set of infrared frames (one from each) arrives after about 5-7 seconds. However, IR pattern is not visible in the images. Disabling emitter or enabling just one infrared does not change anything.

I believe my issue is very similar to #3376, for that case 640 x 480  is problematic, for me it is 848 x 480.

MartyG-RealSense, 2020-11-01:
At this point it may be worth trying to build librealsense from source code with the **RSUSB** method that is not dependent on kernel versions or Linux versions and does not need patching.  This should determine whether or not the problem is related to the kernel, since RSUSB does not rely on the kernel to operate.  RSUSB is enabled by using the CMake build flag **-DFORCE_RSUSB_BACKEND=true**

 https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

msaliharslan, 2020-11-03:
848 x 480 resolution still is not working

MartyG-RealSense, 2020-11-03:
I have researched the case completely again from the beginning but cannot find an explanation for the behavior.  So I will consult with Intel about it.   Thanks again for your patience.

MartyG-RealSense, 2020-11-03:
Hi again @msaliharslan   I have received guidance for your case.

1.  Please use the latest SDK and firmware versions.

2.  Please try running **Dynamic Calibration** on your camera.

As you have access to a Windows PC, it will be easiest to use the Dynamic Calibrator tool if you download the Windows version from the link below.  The user guide for the tool can also be found at this link.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

If you would prefer to use the Linux version of the Dynamic Calibrator tool, page 14 onward of the user guide has Linux installation instructions.

msaliharslan, 2020-11-05:
Dynamic calibration did the trick, now it works!! Thank you.

closed 2020-11-05 by msaliharslan

MartyG-RealSense, 2020-11-05:
Great news - thanks for the update!   :)