realsenseai/librealsense #7624: Get depth of only one aligned pixel without aligning every pixel of the depth image

feefro, 2020-10-21:
Since aligning funcion is very time consuming is possible to get only one (or a little group) of depth aligned pixels without aligning all the pixels depth image to the RGB one?

MartyG-RealSense, 2020-10-21:
Hi @feefro   If you want to map a color pixel to depth pixel without using rs2::align, you could use **rs2_project_color_pixel_to_depth_pixel**

The link below provides example scripting in the Python language.

https://github.com/IntelRealSense/librealsense/issues/5603

The discussion in the link below is very good for a C++ perspective:

https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-614261704

MartyG-RealSense, 2020-10-30:
Hi @feefro   Do you still require assistance with this case, please?   Thanks!

feefro, 2020-10-30:
No, thanks!

MartyG-RealSense, 2020-10-30:
Thanks again for the update!

closed 2020-10-30 by MartyG-RealSense