realsenseai/librealsense #7626: One USB3.1 Hub for Multi-cameras

JANGSOONMYUN, 2020-10-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |   Win 10   | 
| Platform                                 | PC |          

### Issue Description

I'm going to use 4 realsense cameras and 4 FILR cameras with one USB3.1 Hub.
The USB hub support 5Gbs data speed. And external power is connected.

However, 5Gbs is enough to control all the cameras?
Or, should I use more hubs? If I need to use more hub, how many hubs are suitable for 8 cameras?

MartyG-RealSense, 2020-10-21:
Hi @JANGSOONMYUN   The USB standard allows up to 5 hubs to be linked together on a single computer.   If you want to keep your setup simple and compact, you can purchase USB 3 hubs that have up to 10 ports.  You can search Amazon for **hub powered 10 port** for a range of different manufacturers' 10-port hubs.

Intel's multiple camera white-paper document provides a table for estimates of bandwidth use on a single hub for different numbers of non-synced cameras at various streaming configurations.  The streaming configuration will likely determine whether there is enough bandwidth within a single hb's 5 Gbs allowance for 8 cameras.

[image]

JANGSOONMYUN, 2020-10-22:
@MartyG-RealSense Thank you so much. It helped a lot.

closed 2020-10-22 by JANGSOONMYUN

MartyG-RealSense, 2020-10-22:
Great to hear @JANGSOONMYUN  I'm pleased that I could be of help.   :)