realsenseai/librealsense #7628: D435 GenLock mode generates frames without trigger (with start of streaming)

daziFR, 2020-10-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | SDK 2.0 (build 2.39.0) }                          | 
| Language                            |  cpp             | 
| Segment			|  others                           | 

### Issue Description
When using D435 camera with the Inter Cam Sync Mode set to i.e. 124 (or any to enable GenLock feature generating burst of frames), with start of streaming, camera generates burst of frame without external trigger being applied on synchronization interface.

Is this a known/expected behavior?
From the "External Synchronization of Intel® RealSense™ Depth cameras" White Paper, when the Genlock mode is selected, the camera will wait indefinitely for a trigger. 
###

agrunnet, 2020-10-21:
That should not happen. Do you have something plugged into the trigger port? I want to rule out electrical noise on the line.

daziFR, 2020-10-21:
No. Camera housing is closed. With the Inter Cam Sync Mode set to 124 camera generates 118 frames with stream start.

agrunnet, 2020-10-22:
@daziFR thanks for the alert. We are able to reproduce it.

daziFR, 2020-10-23:
@agrunnet Thank you for the information. Will this be solved in some of the next firmware releases?

agrunnet, 2020-10-23:
Yes

YoshuaNava, 2021-01-14:
@daziFR I'm experiencing a similar issue. I use a D430 board, follow the instructions provided by Intel for setting up a trigger signal, and configure the cameras to run with inter cam sync mode 4 (genlock). This issue does not seem to happen with D435x cameras. In both cases I'm employing the realsense-ros package with a few modifications: https://github.com/ANYbotics/realsense

@agrunnet Has this issue been solved in the latest firmware?

Thank you for your help in advance.

agrunnet, 2021-01-14:
Currently we are still trying to understand by talking to sensor engineers.  To be specific, if you set the trigger to only capture a few frames you should not see this issue. The root cause of the pulse limit is not quite clear yet. 

I am very curious though, as to what the usage is that requires a single sync to trigger that many frame captures, if it is possible to share it?  

@YoshuaNava you are seeing this when you set it to a small number of pulses, correct? (Basically 1). But for the d430 only I presume.

YoshuaNava, 2021-01-20:
Hi @agrunnet

Thank you for your prompt reply.

> I am very curious though, as to what the usage is that requires a single sync to trigger that many frame captures, if it is possible to share it?

In our case we have a trigger running at 16 Hz, and we want the cameras to sample images at 16 Hz as well. So, we have set the inter_cam_sync_mode parameter to 4. We want one image per trigger pulse. Can you confirm if this setting is correct?

> you are seeing this when you set it to a small number of pulses, correct?

We have tested setting the trigger frequency to 16 Hz only (with inter_cam_sync_mode=4), but varied the values of `infra_fps`, `depth_fps` to 8, 5, 30, 60 and 90. In all cases we still see that the first time we run the driver with genlock, the frame rate is that the camera follows is not the one from the trigger. If we stop the driver and run it again N times, then it does sample at the trigger frequency. 

We have further tried to get the parameter of inter_cam_sync_mode from the sensor handle, which seems correct (4 when set it to 4 for genlock)

> But for the d430 only I presume.

Exactly, we have only observed it with the D430 module. This is not the only difference we have observed, though. It seems to have a different initialization procedure compared to the D435, as explained in [my comment in another issue](https://github.com/IntelRealSense/realsense-ros/pull/1313#issuecomment-760168899)

agrunnet, 2021-01-22:
@YoshuaNava we have tried to reproduce your issue with D430 and are not able to. Please note that the prescription for setting this mode is:

1.	Set “Inter Cam Sync Mode” to 4 (BEFORE STREAMING)
2.	Select the desired streaming mode, e.g. 848x480x60fps
3.	Start streaming
      a.	Trigger signal can be started either before or after step 3
     b.	Trigger frequency should be half of native frame rate or less

jnsweets, 2021-02-12:
@YoshuaNava 
Did the suggestions provided by @agrunnet help? Are you still having issues? We have not been able to reproduce the problem on D430.

@daziFR 
Your issue is still being worked on. We will update when new info or a solution is available.

YoshuaNava, 2021-02-21:
Hi @agrunnet and @jnsweets 

We updated our realsense-ros wrapper, and it solved our issues. Sorry for the hassle.

Thank you very much for the support.

agrunnet, 2021-02-21:
Glad to hear it!

closed 2021-02-21 by agrunnet