realsenseai/librealsense #7629: pyrealsense2 get_vertices() default argument

MechatronixX, 2020-10-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |   Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.4.0-51-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.23.0                           | 
| Language                            |  Python 3.7.9             | 
| Segment			|  Others                           | 

### Issue Description
I am following this example to get the raw point cloud as a Numpy array: 
https://github.com/IntelRealSense/librealsense/issues/1466

Trying back and forth for quiet some time, I just couldn't get data out that I could do anything with except printing it. While completing writing issue I found out that I need to call get_vertices() with an integer argument. Then useful data is returned. From experiments I found out that this returns the point cloud in different array shapes. 

It would be nice to clarify this in the [documentation](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.points.html#pyrealsense2.points.get_vertices).  

Here is some sample code and the printouts I got when not calling get_vertices() _without_ an integer argument: 

```
import numpy as np
import pyrealsense2 as rs

pc = rs.pointcloud()
pipe = rs.pipeline()

pipe.start()
frames = pipe.wait_for_frames()
depth = frames.get_depth_frame()
color = frames.get_color_frame()
pc.map_to(color)
points = pc.calculate(depth)

# >> TypeError: Cannot cast array data from dtype([('f0', '<f4'), ('f1', '<f4'), ('f2', '<f4')]) to dtype('float64') according to the rule 'unsafe'
#point_cloud = np.asfarray(points.get_vertices())

# Also tried  and asarray()
point_cloud = np.asanyarray(points.get_vertices())  # Needs to called with integer 1,2,3 to return useful data it seems  

# >> [(-0., -0., 0.) (-0., -0., 0.) (-0., -0., 0.) ... ( 0.,  0., 0.)  (0.,  0., 0.)(0.,  0., 0.)]
print(point_cloud)

# >> [('f0', '<f4'), ('f1', '<f4'), ('f2', '<f4')]
print(point_cloud.dtype)

# >> True
print(point_cloud.flags['C_CONTIGUOUS'])

# >> (0.88558733, -2.0034473, 3.2540002)
print(point_cloud[2343])

# Basic operations that failed

# >> "Error: cannot perform reduce with flexible type"
# test1 = np.amax(point_cloud)

# >> TypeError: ufunc 'isnan' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''
#test2 = np.isnan(point_cloud)

# >> TypeError: invalid type promotion
# test3 = point_cloud[0]*1.2344

# >> TypeError: invalid type promotion
test4 = np.copy(point_cloud[0])*1.12344
```

MartyG-RealSense, 2020-10-21:
Hi @MechatronixX   May I confirm please that you have found a solution and would like the Python documentation changed to make it easier for others to solve the same problem that you did?  Thanks!

MechatronixX, 2020-10-21:
Yes, that is a precise summary. 
Maybe it would also be a good idea to remove whatever default argument get_vertices() has, as that default argument makes get_vertices() return something that appears not to be useful.

MartyG-RealSense, 2020-10-21:
I will label this discussion as Documentation so that your suggestion for a documentation edit can be considered by the RealSense team at a future date. This case will be kept open in the meantime.  Thanks very much.

hwy1992129, 2021-05-28:
@MartyG-RealSense Hello, I am running some examples. In pyglet_pointcloud_viewer.py.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/pyglet_pointcloud_viewer.py#L444
```
verts = np.asarray(points.get_vertices(2)).reshape(h, w, 3)    
texcoords = np.asarray(points.get_texture_coordinates(2))
```
I don't understand the meaning of the argument "2" there. It seems that it allows an empty argument, 1,2,3... I don't find anything about it in the documentation....

MartyG-RealSense, 2021-05-28:
Hi @hwy1992129   **points.get.vertices()** retrieves the vertices of the point cloud.  In 99% of cases though the brackets are left empty when using this instruction.  

A long version of the (2) command is **np.asanyarray(points.get_vertices(dims=2))**.  According to the pyrealsense2 official documentation, the bracket is **dims=1** by default in regard to get_vertices and get_texture_coordinates.  So leaving the bracket empty could signify to just use the default value.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.points.html

What the supported dims settings 1-3 do seems to be defined in the link below as 'case' 1 to 3, with the appropriate response being applied depending on the number used in the brackets.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/pyrs_frame.cpp#L211

I do not know what difference each setting makes though unfortunately.

hwy1992129, 2021-05-28:
@MartyG-RealSense Thank you

closed 2022-01-22 by RealSenseCustomerSupport