realsenseai/librealsense #7631: Depth resolution below 0.1 mm (reproducing the Intel whitepaper)

vmazlin, 2020-10-22:
Intel whitepaper **Stereo depth cameras for mobile phones** demonstrates impressive depth precision of 60 micrometers at 6 cm range. We would like to reproduce these results, however the details about the device are missing. The white paper only mentions that "Intel RealSense D405 mobile module was modified for focus at 6cm range". How was the optics modified? Can we achieve the same results with the modern stereo depth cameras (D455, D435, D415)?

[image]

Thanks

Slava

MartyG-RealSense, 2020-10-22:
Hi @vmazlin   A good starting point in achieving very close range sensing is to use a D435 or D435i camera model, as these have a default minimum depth sensing range of 0.1 meters / 10 cm.  The D415 minimum depth sensing range is 0.3 m and D455 is 0.4 m.  

You can reduce minimum distance by increasing the value of a setting called **Disparity Shift**, though this has the effect of reducing the **maximum** depth sensing distance.

vmazlin, 2020-10-22:
Thanks @MartyG-RealSense , we will start with D435 or D435i although we expect that such close focusing will also require different optics
Are there some guidelines or someone can share experience with using different lenses?

MartyG-RealSense, 2020-10-22:
Intel have published a white-paper document on physical optical filters for RealSense 400 Series cameras.  

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

agrunnet, 2020-10-22:
This was a special camera with smaller baseline and lens with optics that allowed image to remain in focus down to 6cm.  You are correct in saying that your image will blur on the d435 when you move it that close and also the baseline is so big that you will be trying to triangulate on objects seen from opposite sides, so it will only work with nearly flat objects. In short, you can definitely get closer with the d435 as recommended by @MartyG-RealSense but you cannot retrofit optics on there to change focus or baseline.

closed 2020-11-07 by MartyG-RealSense