realsenseai/librealsense #7633: Whether the bag file is completion?

zhang942, 2020-10-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRea

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

config.enable_record_to_file('../shiyanbag/object10.bag')

Above is my program to get the bag file.But,I have a question, I get the bag file before the image is aligned,so, whether I get the bag file is complete?

MartyG-RealSense, 2020-10-22:
Hi @zhang942   Are you asking please how to check if the bag file has as many frames in it as you believe that there should be?

zhang942, 2020-10-23:
Hi，yes.I want to know how to parse the bag file and get the data in the bag file.

zhang942, 2020-10-23:
@MartyG-RealSense Yes.I want to know how to parse the bag file and get the data in the bag file.I just want to recover the three-dimensional coordinates of any point from inside.But I don't know how to achieve it?

MartyG-RealSense, 2020-10-23:
The link below contains a Python tutorial for reading a depth and RGB stream from a bag file and rendering the data as a 3D point cloud.  The point-cloud data is then exported to a **ply** format file.

https://github.com/dorodnic/binder_test/blob/master/pointcloud.ipynb

zhang942, 2020-10-25:
Thank you very much for your answer, I have solved this problem. Now I have another question. What is the difference between bag files recorded by ROS and function config.enable_record_to_file?

MartyG-RealSense, 2020-10-25:
Bag files recorded in the RealSense SDK are based on the ROS rosbag standard.  The SDK version of rosbags focuses on storing only camera data though.

https://github.com/IntelRealSense/librealsense/issues/3020

closed 2020-11-04 by zhang942