realsenseai/librealsense #7634: Unable to resolve the name realsense.pipeline.

saifiasif, 2020-10-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC windows 10  |
| SDK Version                            |  {2.0 }                          | 
| Language                            |  {MATLAB }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
hello, i tried running depth_example in MATLAB and i got the following issue,

>> depth_example
Unable to resolve the name realsense.pipeline.

Error in depth_example (line 3)
    pipe = realsense.pipeline();

MartyG-RealSense, 2020-10-22:
Hi @saifiasif    This problem may be occurring because some files need to be copied into the **+realsense** folder.

https://github.com/IntelRealSense/librealsense/issues/5376

saifiasif, 2020-10-22:
i was using matlab 2020a then i used matlab 2017b and it worked.

saifiasif, 2020-10-22:
> Hi @saifiasif This problem may be occurring because some files need to be copied into the **+realsense** folder. Please visit the link below for further details.
> 
> #5376

w

saifiasif, 2020-10-22:
thank you for your swift assistance. I installed using the windows installer would any files still be missing?

MartyG-RealSense, 2020-10-22:
After the Windows installer has been run, there is a final step:

:************

The package will be installed to C:\Program Files (x86)\Intel RealSense SDK 2.0\matlab\+realsense\. You can copy it from here to a place where Matlab can find it or add it to Matlab's path

saifiasif, 2020-10-22:
> After the Windows installer has been run and completed installation, there is a final step:
> 
> The package will be installed to C:\Program Files (x86)\Intel RealSense SDK 2.0\matlab+realsense. You can copy it from here to a place where Matlab can find it or add it to Matlab's path

Thanks for the clarification

closed 2020-10-22 by saifiasif