realsenseai/librealsense #7636: Python wrapper

ELEGER, 2020-10-22:
Hello,
Thanks for developping and open source SDK. I have a question regarding the python wrapper.
I'm trying to follow the (installation) [https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python] procedure foran unsupported kernel (5.0.0-1070-oem-osp1) on ubuntu 18.04.

> cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true

gives: 

`-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/gcc
-- Check for working CXX compiler: /usr/bin/gcc -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /home/el/miniconda3/bin/python (found version "3.8.3") 
-- Found PythonLibs: /home/el/miniconda3/lib/libpython3.8.so (found version "3.8.3") 
-- Found PythonLibs: /home/el/miniconda3/lib/libpython3.8.so
-- Performing Test HAS_CPP14_FLAG
-- Performing Test HAS_CPP14_FLAG - Success
-- pybind11 v2.2.1
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- LTO enabled
CMake Error at CMakeLists.txt:45 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at CMakeLists.txt:82 (if):
  if given arguments:

    "STREQUAL" "RS2_USE_V4L2_BACKEND"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
`

Any help would be greatly welcome

Thanks

MartyG-RealSense, 2020-10-22:
Hi @ELEGER    Are you building the Python wrapper whilst in the topmost directory (**librealsense**) please?

https://github.com/IntelRealSense/librealsense/issues/4335

ELEGER, 2020-10-22:
Hi @MartyG-RealSense ,
Thanks for your answer.
I was in:

> /librealsense/wrappers/python/build

I ran it in  /librealsense/
>cmake -DBUILD_PYTHON_BINDINGS:bool=true        
CMake Error: The source "/librealsense/CMakeLists.txt" does not match the source "/librealsense/wrappers/python/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.

I don't see what I am missing. Could you be a bit more explicit ?

Thanks

e

MartyG-RealSense, 2020-10-22:
It can admittedly sometimes be confusing to say that CMake should be run from the top directory.  The full process is to do **mkdir build && cd build** whilst in the **librealsense** directory.  This creates a directory called **build** and takes you into that directory.  THEN you run the CMake command from within **build**

ELEGER, 2020-10-22:
Thanks for your quick answer,

I understood correctly then. I installed the sdk yesterday #1462 . The build folder in the top directory already exist and isn't empty:

> CMakeFiles/
> common/
> config/
> examples/
> Makefile
> src/
> third-party/
> tools/
> wrappers/
> cmake_install.cmake
> cmake_uninstall.cmake
> realsense2Config.cmake
> realsense2ConfigVersion.cmake
> udev-rules.h
> compile_commands.json
> CMakeCache.txt

thanks

MartyG-RealSense, 2020-10-22:
Okay, then please run the command inside **build** that you tried already in **librealsense**

cmake -DBUILD_PYTHON_BINDINGS:bool=true

ELEGER, 2020-10-22:
So, inside the build, I ran the command you mentionned and got:

```
CMake Error: The source directory "/librealsense/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

```

MartyG-RealSense, 2020-10-22:
Try:

cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true

ELEGER, 2020-10-22:
I did:

```
-- pybind11 v2.2.1
CMake Error at CMakeLists.txt:45 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

CMake Error at CMakeLists.txt:82 (if):
  if given arguments:

    "STREQUAL" "RS2_USE_V4L2_BACKEND"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!

```

Thanks

MartyG-RealSense, 2020-10-22:
It is probably going back to the original subject of a non-standard kernel being used.  Is it possible to install librealsense with RSUSB like I suggested yesterday in your other case please?

ELEGER, 2020-10-22:
But this is what I did yesterday, and it went through #1462
However the wrapper is not installed or is installed ? Or do we need to perform an other installation following in addition, following [this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md).

My folder in  librealsense/wrappers/python/ contains is not empty at all, meaning the wrapper has been installed ?

Sorry I am really lost into the installation steps

MartyG-RealSense, 2020-10-22:
Okay, I see what has happened now.  In yesterday's case, Python was not mentioned and so the instructions to build the Python wrapper at the same time as librealsense were not included.

To build the Python wrapper too,  -DBUILD_PYTHON_BINDINGS:bool=true would have to be included:

 cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_PYTHON_BINDINGS:bool=true

ELEGER, 2020-10-22:
Ok, thanks.
Ran the command in build,
-- pybind11 v2.2.1
CMake Error at CMakeLists.txt:45 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

> CMake Error at CMakeLists.txt:82 (if):
>   if given arguments:
> 
>     "STREQUAL" "RS2_USE_V4L2_BACKEND"
> 
>   Unknown arguments specified
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/el/bin/Intel/librealsense/CMakeFiles/CMakeOutput.log".

The CMakeOutput.log is enclosed
[CMakeOutput.log](https://github.com/IntelRealSense/librealsense/files/5425353/CMakeOutput.log)

thanks

MartyG-RealSense, 2020-10-22:
Past history with this problem suggests that running the CMake command from the **build** folder of the librealsense source code directory should solve this error every time.  So I am not sure why it is not doing so in this case.  Using RSUSB should mean that your non-standard kernel should not matter.

And your librealsense build is otherwise working correctly, yes?  And you can run programs such as the RealSense Viewer?

ELEGER, 2020-10-22:
It seems they are not compiled ? in:

> /librealsense/tools/realsense-viewer/
I should have a compiled program, but have only:

> realsense-viewer.cpp
> readme.md
> openvino-face-detection.cpp
> CMakeLists.txt
 and 
> res/

After the run command: cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true 

I should have to do something else, or did I miss a step?

Thanks

MartyG-RealSense, 2020-10-22:
After the CMake build statement has been used, I have seen this command used as a final step:

**sudo make uninstall && make clean && make && sudo make install**

ELEGER, 2020-10-22:
ok... So recloned the repo, 

followed the steps to install:
```
 ./scripts/setup_udev_rules.sh
 cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
```
Went smooth, but the 

> sudo make uninstall && make clean && make && sudo make install

did not go through. So you were right, I did not install it properly. The log is enclosed
[CMakeOutput.log](https://github.com/IntelRealSense/librealsense/files/5425714/CMakeOutput.log)
, you can tail it to see where it fails.

As a reminder:
uname -a
Linux cima 5.0.0-1070-oem-osp1 #76-Ubuntu SMP Thu Oct 15 14:25:40 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Thanks

MartyG-RealSense, 2020-10-22:
Using **scripts/setup_udev_rules.sh** is not part of the RSUSB process.  Once the source code folder has been obtained and the build directory has been created, you go to build and use the CMake instruction.

The key difference with the RSUSB process is that dependencies are set up over the internet connection instead of being patched in.

ELEGER, 2020-10-23:
Ok, I did. Still cannot install it.

The log is enclosed
[CMakeOutput.log](https://github.com/IntelRealSense/librealsense/files/5427744/CMakeOutput.log).

Thanks

MartyG-RealSense, 2020-10-23:
There is no need to supply the CMake log each time.  I appreciate the thought though!

Let's go through the RSUSB process from the start as though we were doing it for the first time.

1.  Go the the SDK's **Releases** page and download the SDK source file from the **Assets** list at the bottom of the release notes.

https://github.com/IntelRealSense/librealsense/releases

[image]

2.  Once the source code folder is downloaded, go to the parent **librealsense** folder and do **mkdir build && cd build** to create a directory called **build** and navigate to it.

[image]

3.  Once in the **build** directory, ensure that you have an internet connection and then run this CMake instruction to build librealsense and the Python wrapper together:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_PYTHON_BINDINGS:bool=true**

4.  Once the above instruction has finished running, input the instruction below to finish the installation process.  There should be no more steps after this, as you do not need to d any patching with the RSUSB method.

**sudo make uninstall && make clean && make && sudo make install**

ELEGER, 2020-10-23:
Thanks,
This is exactly what I was doing before, except that I was cloning the repo.

The last commands are not going through, I cannot link the executable:

> [ 60%] Linking CXX executable rs-hello-realsense
> 
> collect2: error: ld returned 1 exit status
> examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:101: recipe for target 'examples/hello-realsense/rs-hello-realsense' failed
> make[2]: *** [examples/hello-realsense/rs-hello-realsense] Error 1
> CMakeFiles/Makefile2:568: recipe for target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all' failed
> make[1]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Error 2
> 

Thanks

MartyG-RealSense, 2020-10-23:
I have seen these errors before.  The information below solved the error for the previous person who experienced it.

*************

Can you delete the content of the build directory please and then try re-running both CMAKE and MAKE using the instructions below:

cd ~/librealsense

sudo rm -rf ./build/

mkdir build && cd build

cmake .. -DBUILD_EXAMPLES=true && make -J && sudo make install

ELEGER, 2020-10-23:
Went to 79% this time.

> collect2: error: ld returned 1 exit status
> tools/enumerate-devices/CMakeFiles/rs-enumerate-devices.dir/build.make:95: recipe for target 'tools/enumerate-devices/rs-enumerate-devices' failed
> make[2]: *** [tools/enumerate-devices/rs-enumerate-devices] Error 1
> CMakeFiles/Makefile2:2031: recipe for target 'tools/enumerate-devices/CMakeFiles/rs-enumerate-devices.dir/all' failed
> make[1]: *** [tools/enumerate-devices/CMakeFiles/rs-enumerate-devices.dir/all] Error 2
> [ 79%] Linking CXX executable rs-convert

> ../../librealsense2.so.2.39.0: undefined reference to `__cxa_throw_bad_array_new_length'
> ../../librealsense2.so.2.39.0: undefined reference to `std::basic_ofstream<char, std::char_traits<char> >::~basic_ofstream()'
> collect2: error: ld returned 1 exit status
> tools/terminal/CMakeFiles/rs-terminal.dir/build.make:121: recipe for target 'tools/terminal/rs-terminal' failed
> make[2]: *** [tools/terminal/rs-terminal] Error 1
> CMakeFiles/Makefile2:2141: recipe for target 'tools/terminal/CMakeFiles/rs-terminal.dir/all' failed
> make[1]: *** [tools/terminal/CMakeFiles/rs-terminal.dir/all] Error 2
> [ 79%] Linking CXX shared library librealsense2-gl.so
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Error 2

MartyG-RealSense, 2020-10-23:
Ok, repeat the process please and leave out -J this time:

cd ~/librealsense

sudo rm -rf ./build/

mkdir build && cd build

cmake .. -DBUILD_EXAMPLES=true && make && sudo make install

3dsf, 2020-10-23:
I just started looking into this 
and as I built it last night using a distro specific modified script (jetson hacks) for the Jetson Nano L4T 32.4.4 (latest) -- Linux kernel 4.9.14 tegra.
It was late but I didn't notice any errors.  
- realsense-viewer opens
- pyrealsense2 does not import into python 

`pip check pyrealsense2`
> uff 0.6.9 has requirement protobuf>=3.3.0, but you have protobuf 3.0.0.

I don't know if that is relevant to @ELEGER , but I presume that is my issue on why it failed for me

_tag:_ @MartyG-RealSense

ELEGER, 2020-10-23:
Same error than before, 56%:

> 
> collect2: error: ld returned 1 exit status
> examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:101: recipe for target 'examples/hello-realsense/rs-hello-realsense' failed
> make[2]: *** [examples/hello-realsense/rs-hello-realsense] Error 1
> CMakeFiles/Makefile2:454: recipe for target 'examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all' failed
> make[1]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Error 2

MartyG-RealSense, 2020-10-23:
Thanks very much @3dsf for your advice.  Pip install of pyrealsense2 does not work on Arm-processor devices such as Jetson and Raspberry Pi as the pip pyrealsense2 packages are for x86 and x64 devices such as PCs.  The Python wrapper has to be built from source on Arm.

MartyG-RealSense, 2020-10-23:
@ELEGER   Could you provide more details please about the computer with the non-standard kernel that you are installing on please?

ELEGER, 2020-10-23:
> Linux cima 5.0.0-1070-oem-osp1 #76-Ubuntu SMP Thu Oct 15 14:25:40 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

> cat /proc/cpuinfo
11 of them
> processor	: 0
> vendor_id	: GenuineIntel
> cpu family	: 6
> model		: 165
> model name	: Intel(R) Core(TM) i9-10885H CPU @ 2.40GHz
> stepping	: 2
> microcode	: 0xc8
> cpu MHz		: 1988.733
> cache size	: 16384 KB
> physical id	: 0
> siblings	: 16
> core id		: 0
> cpu cores	: 8
> apicid		: 0
> initial apicid	: 0
> fpu		: yes
> fpu_exception	: yes
> cpuid level	: 22
> wp		: yes
> flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp pku ospke md_clear flush_l1d arch_capabilities
> bugs		: spectre_v1 spectre_v2 spec_store_bypass swapgs itlb_multihit
> bogomips	: 4800.00
> clflush size	: 64
> cache_alignment	: 64
> address sizes	: 39 bits physical, 48 bits virtual
> power management:

lspci

lspci 

> 00:00.0 Host bridge: Intel Corporation Device 9b44 (rev 02)
> 00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 02)
> 00:02.0 VGA compatible controller: Intel Corporation Device 9bc4 (rev 05)
> 00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 02)
> 00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
> 00:12.0 Signal processing controller: Intel Corporation Device 06f9
> 00:13.0 Serial controller: Intel Corporation Device 06fc
> 00:14.0 USB controller: Intel Corporation Device 06ed
> 00:14.2 RAM memory: Intel Corporation Device 06ef
> 00:14.3 Network controller: Intel Corporation Device 06f0
> 00:15.0 Serial bus controller [0c80]: Intel Corporation Device 06e8
> 00:15.1 Serial bus controller [0c80]: Intel Corporation Device 06e9
> 00:16.0 Communication controller: Intel Corporation Device 06e0
> 00:16.3 Serial controller: Intel Corporation Device 06e3
> 00:1b.0 PCI bridge: Intel Corporation Device 06ac (rev f0)
> 00:1c.0 PCI bridge: Intel Corporation Device 06b8 (rev f0)
> 00:1c.5 PCI bridge: Intel Corporation Device 06bd (rev f0)
> 00:1f.0 ISA bridge: Intel Corporation Device 068e
> 00:1f.3 Audio device: Intel Corporation Device 06c8
> 00:1f.4 SMBus: Intel Corporation Device 06a3
> 00:1f.5 Serial bus controller [0c80]: Intel Corporation Device 06a4
> 00:1f.6 Ethernet controller: Intel Corporation Device 0d4c
> 01:00.0 VGA compatible controller: NVIDIA Corporation Device 1fb8 (rev a1)
> 01:00.1 Audio device: NVIDIA Corporation Device 10fa (rev a1)
> 

I think I'll install it to a Rasp, or an other machine, this is too complicated on this one.

MartyG-RealSense, 2020-10-23:
@ELEGER   Okay, good luck.  I'm sorry that you have had a bad experience on your current machine.

3dsf, 2020-10-23:
@MartyG-RealSense 
I built from source (cloning git repo) as @ELEGER did too.  
I used the `pip check` to see if I was missing dependencies.  I've upgraded my protobuf version and will re-build.  
`protobuf has caused issues for me in the past for other programs...`

MartyG-RealSense, 2020-10-23:
@3dsf   Did you get the error **ImportError: No module named 'pyrealsense2'**

3dsf, 2020-10-23:
> @3dsf Did you get the error **ImportError: No module named 'pyrealsense2'**

**yes**, in 2.7 and 3.6  
I can see the pyrealsense2.so files at `/usr/local/libs` and the directory is in my `$PYTHONPATH`  

I'll follow your build advice here (cmake clean, etc.), I just checking on some unrelated build flags

3dsf, 2020-10-24:
**pyrealsense2 fixed**
_looks like I was causing my own trouble... or this is just how it installs now_
`sudo ln -s /usr/lib/python3/dist-packages/pyrealsense2/python-3.6/pyrealsense2.cpython-36m-aarch64-linux-gnu.so /usr/local/lib/pyrealsense2.so`

_I did rebuild (from git), but I also did the above to make it work in python3.  It did work in python2 though (though, I'm not saying rebuilding did it, because I ran an install script instead of the build script in between...) _

MartyG-RealSense, 2020-10-24:
Thanks again @3dsf for sharing your tips!

MartyG-RealSense, 2020-10-30:
Hi @ELEGER   Do you have an update about this case that you can provide, please?   Did you have success with installation on another machine?   Thanks!

closed 2020-11-07 by MartyG-RealSense