realsenseai/librealsense #7638: realsense2 library - Clean compile free of warnings

JTrantow, 2020-10-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { all} | 
| Firmware Version                   | N/A | 
| Operating System & Version | My focus is Win10, but applies to all.  | 
| Kernel Version (Linux Only)    | N/A                                         | 
| Platform                                 | Primarily PC, but applies to all|
| SDK Version                            |  master                       | 
| Language                            |  {C/C#}                          | 
| Segment			|  | 

### Issue Description
The default development build currently yields 275 Warnings and 122 Messages using the MSVC Compiler 19.27.29912.0 with the /W3 warning level. I achieved a clean build on my fork from months ago but I'd like to get these changes into the master so I can keep sync'd with improvements.

I am using MSVC /std:C++14 but these same issues carry over to other compilers and linters.

Cleaning up the warnings will improve the code and reduce bugs. Many of the changes are casting issues brought on by sloppy type decisions. Using the correct variable type reduces the amount of casting required and improves bug quashing. Other warnings indicate best practices for initializing variables. 

I decided to start with realsense2 library which currently yields 35 Warnings and 15 messages. I will break the pulls into multiple requests in case my choices conflict with the design intent.  I won't waste any time on third party files.

MartyG-RealSense, 2020-10-23:
Hi @JTrantow   Thanks very much for your commitment to improving RealSense!  Yes, **Pull Requests** would be the proper channel to submit fixes for consideration for merging into the RealSense SDK.

How many separate pull submissions do you anticipate might be required, please?  I believe that less would be better in order to keep the PR page manageable and avoid forcing the pulls of other contributors off the front page.  As such, it may be best to submit what you consider to be the most important fixes first and then submit subsequent PRs once the previous ones are merged.

JTrantow, 2020-10-23:
The current development branch builds the realsens2 library with 35
Warnings/15 Messages with the MSVC compiler /W3.

To remove the warnings, there are four or five different fixes across 28
files. *I was planning on 4-5 pulls *each of which would touch 2-8 files.
Some of the fixes are simple and should be noncontroversial, I think these
would be merged quickly and off the front page.
One or two pull requests will involve variable type changes that might
generate discussion before the merge.

As an example, several objects use int or uint32_t variables to hold the
size of the object or frame or transfer request. These size member
variables are used with std::vector, map, etc.. functionality that uses
size_t types. So many compiler warnings are generated by assigning int <->
size_t.
In my opinion:

   - size variables should always be unsigned.
   - If size if a known fixed size firmware sizes or hardware restrictions
   a fixed int type should be used.
   - In most cases, size_t is a good choice and causes no warnings when
   used with std:: functions.

I think it makes sense to break the pulls for realsense2 into multiple
requests. Once I have a better feel for how the changes are accepted, I can
consolidate further code cleanup into fewer requests. (the solution builds
with 275 Warnings, but applying the same types of fixes as to realsense2
probably accounts for > 80% of the warnings.)

MartyG-RealSense, 2020-10-23:
Thanks very much for the great amount of detail that you have provided in your commentary.  I will tag @dorodnic the RealSense SDK Manager into this discussion for his input.

JTrantow, 2020-10-27:
I'd like to resolve the discrepancy between how different classes declare _bpp, _width, _height, _stride. Consistency makes it much easier to improve the code and cleanup compiler warnings.

Various classes use int, size_t, and uint32_t for these members which generally refer to image pixels. I'd argue against int as these values should not be signed. size_t seems overkill for these values. I like fixed ints for hardware restricted dimensions like pixels. Is uint32_t the way to go?  
Or would smaller types like:
```
uint16_t _width, _height, _stride;
uint8_t _bpp;
```
or 
```
uint_least16_t _width, _height, _stride;
uint_least16_t _bpp;
```
be better?

MartyG-RealSense, 2020-11-07:
Reminder to @dorodnic to please take a look at the issue above.   Thanks!

MartyG-RealSense, 2020-11-21:
@RealSenseSupport    Could you help with the question in the link below please?   Thanks!

https://github.com/IntelRealSense/librealsense/issues/7638#issuecomment-717603891

MartyG-RealSense, 2020-11-29:
Hi @JTrantow   I'm very sorry that you did not get a response to your question.  At this point I will close the case.   If you would still like an answer, I recommend re-posting it as a new issue to move it back to the top of the forum's front page and draw attention to it.

closed 2020-11-29 by MartyG-RealSense