realsenseai/librealsense #7640: L515 (RGB) Dynamic Calibration Tool

Faaux, 2020-10-23:
### Issue Description

For the D400 the tool the Dynamic Calibration Tool (specifically Intel.Realsense.CustomRW.exe) exists to read and write custom calibrations onto the device. I was wondering if the same tool or an equivalent is available for the L515 RGB Camera.

Best,
Faaux

RealSenseSupport, 2020-10-23:
Hi @Faaux 

The tool that you reference is only for the D400 product line.  It does not extend to the L500 product line.  Also to complete an answer to your question, there is no equivalent tool for L515.

Faaux, 2020-10-26:
Thanks for the quick answer. I would like to ask another question then (if I should open a new issue for this please tell me).

### Context
My goal is to calibrate a D435 to a L515. I then want to either transform the D435s left IR channel to match the L515 depth channel or the other way round. The extrinsic calibration of this is already done using 3d3d pose estimation of the point clouds. As I want to use the IR image for segmentation of the scene I'd like to map the pixels to the more accurate L515 depth values.

### Current approach
The current approach was to map the IR image to the L515s RGB sensor and then use the internal align between the RGB to Depth of the L515 (calibration via OpenCV of IR ot L515s RGB). For this I would like to get the L515s RGB intrinsics as well as the extrinsics between the L515s RGB sensor and the depth sensor.

### Questions
- Is this a sensible approach?
- What would you recommend? 
- Is there a way to get the intrinsics and extrinsics of the L515?
- Are there specialized functions to project the point cloud back down to the image in librealsense (essentially rs2_project_point_to_pixelfor a whole cloud)?

RealSenseSupport, 2020-11-13:
Hi @Faaux 

Your approach seems reasonable.  Although we have not done this calibration ourselves between D435 and L515.

With regards to intrinsic and extrinsic, you can use https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices to get the detailed information from the cameras.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-23 by RealSenseSupport