realsenseai/librealsense #7641: Is librealsense thread-safe when controlling several cameras?

michael201021, 2020-10-23:
| Required Info               |                            |
|-----------------------------|----------------------------|
| Camera Model                | D435                       |
| Firmware Version            | 05.12.06.00                |
| Operating System & Version  | Linux (Ubuntu 18.04.5 LTS) |
| Kernel Version (Linux Only) | 4.15.0-122-generic         |
| Platform                    | PC                         |
| SDK Version                 | 2.36.0                     |
| Language                    | python, c++                |
| Segment                     | others                     |

##### Issue Description

We use six extrinsically calibrated D435 devices to get a 3D representation of an object. The devices are connected to a powered USB hub, which itself is connected to a PC. All connections are reported to be USB 3.2. To lower the minimum depth as well as keeping the data rate low, we use depth images of 640x360 pixels with 6 frames per second (the object is close and does not move).

In a C++ implementation we observed that some operations lead to longer delays (several hundreds of milliseconds) and therefore tried to avoid pauses in our program by using multi-threading. Especially setting of sensor options and processing frames ran in different threads, so that 12 threads were accessing `librealsense` at the same time, but strictly separated by sensors.

This causes several strange errors and sets the devices to a state where they do no longer react to anything. Resetting by removing power supply was necessary. We checked different sets of devices and cables to eliminate a hardware problem. We therefore suspected that there is a severe synchronization bug in the code. We tried to simplify our code and finally got this small Python source:

```python
import concurrent.futures
import datetime
import logging
import sys
import time

import pyrealsense2 as rs

log = logging.getLogger(__file__)
log.setLevel(logging.INFO)
log.addHandler(logging.StreamHandler())

def select_stream_profile(profiles):
    for profile in profiles:
        profile = profile.as_video_stream_profile()
        if (
            profile.format() == rs.format.z16 and
            profile.fps() == 6 and
            profile.width() == 640 and
            profile.height() == 360
        ):
            return profile

def get_all_devices():
    ctx = rs.context()
    devices = ctx.query_devices()
    print("Found %i devices" % len(devices))
    if devices:
        return devices
    print("Device not found")
    sys.exit(-1)

def get_device(device_id):
    ctx = rs.context()
    devices = ctx.query_devices()
    for device in devices:
        if device.get_info(rs.camera_info.serial_number) == device_id:
            return device
    print("Device not found")
    sys.exit(-1)

def get_frame_handler(device):
    def handle_frame(frame):
        # This line causes a segfault if run under multi-threading,
        # but not with multiple processes.
        device
    return handle_frame

def capture(device_id):
    device = get_device(device_id)
    depth_sensor = device.first_depth_sensor()
    profiles = depth_sensor.get_stream_profiles()
    profile = select_stream_profile(profiles)
    if profile is None:
        print("Device not found")
        sys.exit(-1)
    try:
        depth_sensor.open(profile)
        depth_sensor.start(get_frame_handler(device))
        time.sleep(2)
    finally:
        depth_sensor.stop()
        depth_sensor.close()

def test():
    devices = get_all_devices()
    device_ids = [
        device.get_info(rs.camera_info.serial_number) for device in devices
    ]
    # Replace ProcessPoolExecutor with ThreadPoolExecutor to get a segfault.
    with concurrent.futures.ProcessPoolExecutor(max_workers=6) as executor:
        futures_to_device_ids = {executor.submit(capture, device_id): device_id for device_id in device_ids}
        for future in concurrent.futures.as_completed(futures_to_device_ids):
            print("Completed %s" % futures_to_device_ids[future])

if __name__ == "__main__":
    test()
```

By changing the commented line this program will cause a segmentation fault.

Our questions are:
1. Is `librealsense` thread-safe?
2. Is it allowed to control the options of a sensor from one thread and concurrently process the frames in another thread?
3. Is it allowed to control the options of different sensors in different threads concurrently?
4. Is there a difference in using multi-threading opposed to multi-processing?
5. Why does the execution of e.g. option setting vary between a few milliseconds and several hundreds of milliseconds for the same option (e.g. exposure is just a single number)?

These are some references that are close to this issue, but do not fully answer our questions:
- <https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam>
- <https://community.intel.com/t5/Items-with-no-label/librealsense2-is-it-thread-safe/td-p/485383>
- <https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model>

MartyG-RealSense, 2020-10-24:
Hi @michael201021  I hope that the further-reading links provided below for each of your questions will be useful to you.

**1.  Is librealsense thread-safe?**

https://dev.intelrealsense.com/docs/frame-management#section-frames-and-threads

To quote the above documentation link:

"Callbacks are invoked from an internal thread to minimize latency.  If you have a lot of processing to do, or simply want to handle the frame in your main event loop, librealsense provides rs2::frame_queue primitive to move frames from one thread to another in a **thread-safe** fashion".

**2.  Is it allowed to control the options of a sensor from one thread and concurrently process the frames in another thread?**

https://github.com/IntelRealSense/librealsense/issues/6039

**3.  Is it allowed to control the options of different sensors in different threads concurrently?**

@dorodnic the RealSense SDK Manager has said in a previous case: "Different instances of pipeline or device objects can co-exist on different threads, and you can send frames from thread to thread using frame_queue primitive".

https://github.com/IntelRealSense/librealsense/issues/2005#issuecomment-402734472

**4.  Is there a difference in using multi-threading opposed to multi-processing?**

Whilst not specifically about librealsense, the guide in the link below provides a good explanation of the difference between multi-threading and multi-processing.

https://levelup.gitconnected.com/diy-multithreading-vs-multiprocessing-in-python-fb93698ca7f3

**5.  Why does the execution of e.g. option setting vary between a few milliseconds and several hundreds of milliseconds for the same option (e.g. exposure is just a single number)?**

Quoted from the link below:

"The controls and video streams run asynchronously, which means that setting the exposure will definitely take several frames to propagate back to the user, given the nature of the actual control chain and the USB pipeline".

 https://github.com/IntelRealSense/librealsense/issues/3467#issuecomment-483410178

michael201021, 2020-10-28:
Hi @MartyG-RealSense . Thank you very much for your fast reply. I read through your answers carefully and found that synchronized transfer of frames is well described. But in the short Python program I provided above, a segmentation fault occurs even without any kind of frame handling (the frame handler does not touch the `frame` parameter). The segmentation fault disappears as soon as the program is switched from multi-threading to multi-processing. I do not understand why this happens.

Also all sources I found combined with the sources you provided state that it is absolutely okay to create pipelines or devices in different threads. But from my experience controlling sensor options from different threads either require long pauses in the program or will crash the devices. Unfortunately to demonstrate this a longer program is necessary. I will try to provide one.

MartyG-RealSense, 2020-10-29:
There is no need to provide a test program for me, but thank you for the offer.

This subject is somewhat outside of my programming experience, unfortunately.   My research did find a Python multi-camera case though where a segmentation fault when extracting frames was handled by implementing null pointer checking:

https://github.com/IntelRealSense/librealsense/issues/885#issuecomment-349831693

MartyG-RealSense, 2020-11-07:
Hi @michael201021   Do you still require assistance with this case, please?   Thanks!

michael201021, 2020-11-12:
I read through issue #885. Several commits lie between this and the version we are using, so maybe the error is caused by something else now. If the error is still the same, the described fix requires to change the library source code and to rebuilt it (and to apply the same fix to all newer versions of the library from now on). Also the author of the fix suspects this would create memory leaks. This does not sound very promising.

MartyG-RealSense, 2020-11-12:
Given that setting FPS to 6 may result in timeouts whilst waiting for new frames to arrive, do you experience slowdown / freezes if you test the program in your opening comment with 15 FPS instead?

michael201021, 2020-11-16:
This sounds interesting. Because of the number of used devices (6) and the possible need to transfer several streams per camera, we tried to lower the required bandwidth as much as possible. But maybe a frame rate of 6 is slower than necessary. I will try this again with 15 FPS. Maybe this solves the suspected synchronization issues.

MartyG-RealSense, 2020-11-23:
Hi @michael201021   Do you have an update for me please?   Thanks!

closed 2020-12-01 by MartyG-RealSense