realsenseai/librealsense #7642: 3D depth does not work (2d works)

happyCodingSusan, 2020-10-23:
| | |
|---|---|
|**librealsense**|2.39.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense SR300|
|**Serial Number**|618206003544|
|**Firmware Version**|3.21.0.0|
|**Camera Locked**|YES|
|**Product Line**|SR300|
|**Asic Serial Number**|618206003544|
|**Firmware Update Id**|618206003544|

Please provide a description of the problem
Coded light depth sensor cannot be set to on.
Debug hello-realsense, in line "rs2::depth_frame depth = frames.get_depth_frame();", catched the following error,
RealSense error calling rs2_pipeline_wait_for_frames(pipe:000002227F86EB20):
    Frame didn't arrive within 15000

MartyG-RealSense, 2020-10-23:
Hi @happyCodingSusan   I confirmed that this error can appear when launching the **rs-hello-realsense** example program from the Windows **Command Prompt** interface with an SR300 camera attached.

[image]

If you launch the **RealSense Viewer** in 3D mode and the camera details in the options side-panel momentarily disappear and then re-appear as though the camera is resetting, this can indicate that the camera is being supplied with insufficient power by the USB port when the stream activates and it begins drawing a higher amount of power than when it is in its off state.

Do you have the options of using a mains electricity powered USB hub with the camera, please in order to provide the camera with additional power stability?

happyCodingSusan, 2020-10-23:
@MartyG-RealSense , thank you very much for your fast reply! I urgently need it work for a project. 

When I lunched it in Realsense viewer in 3D mode, it does not work as well. When I ran camera explorer, it showed "error, device failed due to malfunction". 

>>"Do you have the options of using a mains electricity powered USB hub with the camera, please in order to provide the >>camera with additional power stability?" 
I do not have one now. I was wondering if you could please show me how it looks and if you know whether it is sold on amazon.ca or other websites so that I can purchase it.

Many thanks for your kind advice!

happyCodingSusan, 2020-10-23:
@MartyG-RealSense,

I do not have a mains electricity powered USB hub. I was wondering if you could please show me how it looks and if you know whether it is sold on amazon.ca or other websites so that I can purchase it.

Thanks!

MartyG-RealSense, 2020-10-24:
Ypu are very welcome.   :)

When Intel wrote their paper about use of multiple RealSense cameras, they tested with an **AmazonBasics** 4-port mains powered USB hub.  I bought the 7-port version myself based on that recommendation and have had no problems with it.  Other brands of hub are available of course, though it is worth mentioning AmazonBasics first as a model that has been proven to work with RealSense

You can find a range of mains-powered hubs, including AmazonBasics, on Amazon Canada by searching the site for **powered usb 3 hub**.  Bear in mind that they may have full-size "USB-A" cables or micro-sized "USB-C" that will not fit your computer if it does not have a USB-C micro port.  So the safest action will be to choose a model with a USB-A connector.  They are easy to tell apart at a glance in product images, as USB-A is the normal sized USB cable plug and USB-C is much smaller and thinner.

happyCodingSusan, 2020-10-25:
@MartyG-RealSense ,  

3D works now with a powered usb 3 hub.
Thank you very much for your advice.

MartyG-RealSense, 2020-10-25:
Great news @happyCodingSusan  - thanks very much for the update!   :)

happyCodingSusan, 2020-10-26:
@MartyG-RealSense ,

I tested my new camera in the "camera explorer" tool and it works. 
However, when I debug hello-realsense, when hitting the line of code "p.start();", it catches  the following error"
RealSense error calling rs2_pipeline_start(pipe:000001D438294890):

Failed to resolve the request:
        Format: Z16, width: 640, height: 480

Into:
        Formats:
         INZI
"
I found SR300 virtual device is installed under "other devices" with a yellow triangle mark in device manager. I removed it a few times but it is still showing up there. I remembered it is under imaging devices in my first installation. Later I removed the camera and reinstalled it, now it is under "other devices" with yellow triangle mark. Do you think the hello realsense error may be related to it? 
Thanks.

MartyG-RealSense, 2020-10-27:
Hi @happyCodingSusan  The yellow triangle mark indicates that there is a problem with the camera in the Device Manager, likely related to the camera driver.  Problems with the Virtual Driver can make the RealSense functions of the camera (e.g depth) stop working.

Could you try uninstalling first the Virtual Driver and then the Depth and RGB SR300 drivers if you can find them (they should either be under **Imaging Devices** or under **Camera** in the Device Manager).  Right-click on each driver and select **Uninstall device** from the menu.   

Below is an image from my own Device Manager with an SR300 plugged in.

[image]

After the three drivers are uninstalled, unplug the camera, wait a couple of seconds and plug it back in.   If you have an active internet connection, the drivers should automatically download and reinstall, appearing in the Device Manager when the process is complete.

MartyG-RealSense, 2020-11-02:
Hi @happyCodingSusan   Do you require any further assistance with this case please?

happyCodingSusan, 2020-11-03:
@MartyG-RealSense ,

Thank you very much for follow-up.
I did try uninstall and install for a few times, but the virtual device driver is still installed under other device with a yellow bang. The strange thing is the Microsoft gesture application works on this depth camera even though its virtual device driver is not installed correctly.

MartyG-RealSense, 2020-11-03:
Microsoft has a hidden hardware Troubleshooter option in Windows 10 that can scan for hardware problems and try to automatically correct them.

https://support.intelrealsense.com/hc/en-us/community/posts/360049169174/comments/360012666273

If the troubleshooter does not work, running the SR300 **Depth Camera Manager (DCM)** driver installation program has sometimes fixed a broken Virtual Driver.  The program does not always work in modern Windows versions as it was created in 2016, but it is worth a try.

If you left-click on the link below then it will automatically launch a download of the DCM program's installer file in your browser.  Then run the file to install the DCM and hopefully correct the Virtual Driver when doing so.

https://downloadmirror.intel.com/25044/eng/intel_rs_dcm_sr300_3.3.27.5718.exe

MartyG-RealSense, 2020-11-11:
Hi @happyCodingSusan   Do you still require assistance with this case pleae?  Thanks!

happyCodingSusan, 2020-11-12:
@MartyG-RealSense ,
Thanks for asking. I tried the solutions in the link, but the yellow triangle warning cannot be fixed. However, I do not need assistance at the moment.

closed 2020-11-12 by happyCodingSusan

MartyG-RealSense, 2020-11-12:
Okay, thanks very much for the update!