realsenseai/librealsense #7643: Cmake issue when white space in username

jdagdelen, 2020-10-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | NA } | 
| Firmware Version                   | NA                                                         | 
| Operating System & Version |  Win  10                                                 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |PC                                                           |
| SDK Version                            |    latest                                                  | 
| Language                            |  NA  | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                        NA
### Issue Description

When trying to build librealsense on Windows 10 with CMake3.17.5, generation fails if user name has a whitespace in it. 

generate produces errors like:
```
CMake Error at build/unit-tests/log/test-env-log-level-off/CMakeLists.txt:16 (add_executable):
  Cannot find source file:

    C:/Users/John

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
```

MartyG-RealSense, 2020-10-24:
Hi @jdagdelen   Problems with whitespaces in CMake paths is a general issue with CMake rather than liberalsense specifically.   Information on the subject can be found by googling for **path whitespace cmake**.

The most popular recommendations for a solution are to put semicolons - ; - or backslashes - \ - in the path spaces instead of having whitespace.

Point 3 of the guide in the link below discusses whitespace versus semicolons in CMake:

https://samthursfield.wordpress.com/2015/10/20/some-cmake-tips/

jdagdelen, 2020-10-24:
Cool cool, having read more into it, this makes a lot of sense. I just moved it to the root of my C:// directory to solve the problem.

closed 2020-10-24 by jdagdelen

MartyG-RealSense, 2020-10-24:
I'm pleased to hear that you found a solution - thanks for the update!