realsenseai/librealsense #7646: Capture Color and Depth on Google Coral Dev Board

MinPopp, 2020-10-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 |
| Firmware Version                   | 05.12.08.200 |
| Operating System & Version |   Linux Mendel  |
| Kernel Version (Linux Only)    |  4.14.98-imx                                         |
| Platform                                 | Google Coral Dev Board |
| SDK Version                            | v2.39.0                          |
| Language                            |  python3 / -                          |
| Segment			|  others                          |

### Issue Description

I would like to capture depth and color frames with the D435 on the Google Coral Dev board. I sucessfully built librealsense from source, but i do not succeed in patching the kernel for the version 4.14.98. Is there any way of getting the camera to run on the board? My goal is to be able to interact with the camera using python3. [Here](https://github.com/IntelRealSense/librealsense/issues/1335#issuecomment-372622247) is described that the camera should be able to run without the kernel patch. How can I achieve this?

Below you see the log output of `modprobe ucvideo` the Coral Dev Board:

```
mendel@orange-goose:~$ sudo modprobe ucvideo
modprobe: FATAL: Module ucvideo not found in directory /lib/modules/4.14.98-imx
```

I assume that this is where the problem is. Nonetheless below is some more info from the board:

<details>
  <summary>rs-enumerate-devices</summary>
  
  ```
  mendel@orange-goose:~$ rs-enumerate-devices
  Device info:
      Name                          : 	Intel RealSense D435
      Serial Number                 : 	815412070329
      Firmware Version              : 	05.12.08.200
      Recommended Firmware Version  : 	05.12.08.200
      Physical Port                 : 	/sys/devices/platform/usb@38200000/38200000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/video4linux/video1
      Debug Op Code                 : 	15
      Advanced Mode                 : 	YES
      Product Id                    : 	0B07
      Camera Locked                 : 	YES
      Usb Type Descriptor           : 	3.2
      Product Line                  : 	D400
      Asic Serial Number            : 	812113021027
      Firmware Update Id            : 	812113021027

  Stream Profiles supported by Stereo Module
   Supported modes:
      stream       resolution      fps       format
      Infrared 1	  1280x800	@ 30Hz	   Y8
      Infrared 1	  1280x800	@ 25Hz	   Y16
      Infrared 1	  1280x800	@ 15Hz	   Y16
      Infrared 1	  1280x800	@ 15Hz	   Y8
      Infrared 1	  1280x720	@ 30Hz	   Y8
      Infrared 1	  1280x720	@ 15Hz	   Y8
      Infrared 1	  1280x720	@ 6Hz	   Y8
      Infrared 1	  848x480	@ 90Hz	   Y8
      Infrared 1	  848x480	@ 60Hz	   Y8
      Infrared 1	  848x480	@ 30Hz	   Y8
      Infrared 1	  848x480	@ 15Hz	   Y8
      Infrared 1	  848x480	@ 6Hz	   Y8
      Infrared 1	  848x100	@ 300Hz	   Y8
      Infrared 1	  848x100	@ 100Hz	   Y8
      Infrared 1	  640x480	@ 90Hz	   Y8
      Infrared 1	  640x480	@ 60Hz	   Y8
      Infrared 1	  640x480	@ 30Hz	   Y8
      Infrared 1	  640x480	@ 15Hz	   Y8
      Infrared 1	  640x480	@ 6Hz	   Y8
      Infrared 1	  640x400	@ 25Hz	   Y16
      Infrared 1	  640x400	@ 15Hz	   Y16
[... 298 lines cut ...]
  	 Z16
  ```
</details>

<details>
  <summary>tail of dmesg</summary>
  
  ```
  [ 2408.093707] usb 2-1: new SuperSpeed USB device number 4 using xhci-hcd
  [ 2408.134139] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
  [ 2408.141499] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
  [ 2408.149603] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
  [ 2408.181295] uvcvideo: Unable to create debugfs 2-4 directory.
  [ 2408.187720] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
  [ 2408.197473] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
  [ 2408.205573] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
  [ 2408.214003] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/usb@38200000/38200000.dwc3/xhci-hcd.0.auto/usb2/2-1/2-1:1.0/input/input3
  [ 2408.229370] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
  [ 2408.236762] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
  [ 2408.257388] uvcvideo: Unable to create debugfs 2-4 directory.
  [ 2408.264154] uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
  [ 2408.272301] uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
  [ 2408.280387] uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!
  ```
</details>

MartyG-RealSense, 2020-10-25:
Hi @MinPopp   It is possible to build librealsense from source code with a method called **RSUSB** that is not dependent on Linux versions or kernel versions and does not require patching.  The link below has instructions for building librealsense and the Python wrapper from source at the same time using the RSUSB method.  Performing the method requires an internet connection.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

There are both advantages and disadvantages to using the RSUSB method.  For example, it enables quick setup and prototyping but may not be suited to a commercial project.   For more information, please visit the link below and scroll down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

MinPopp, 2020-10-25:
Thank you for your reply. I started `make` and will report back once i (hopefully) worked. It's pretty slow on the board so it might be a day or two

MartyG-RealSense, 2020-10-25:
Thanks very much for the update - good luck, and I look forward to your report.

MinPopp, 2020-10-28:
Thank you so much @MartyG-RealSense , that worked like a charm!

Because I spent quite some time figuring out how I get the D435 camera to run on the Coral Dev Board I share what worked for me, hoping it helps somebody else. 

#### Step 0: prerequisites

Before you start, be sure to **place the board somewhere it doesn't bother you even if the fan is spinning**!

After the board runs as described [here](https://coral.ai/docs/dev-board/get-started/), install the following prerequisites. I am not sure if all of these are needed. It is what I installed, so it might not work if one of them is missing

``` bash
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev7
sudo apt-get install libusb-1.0-0-dev
sudo apt-get install xorg-dev libglu1-mesa-dev
sudo apt-get install libssl-dev
```

#### Step 1: swapfile

The building process requires more than 1GB of ram. So you have to create a swapfile, as it is described [here](https://github.com/f0cal/google-coral/issues/61). I made mine 8G just to be sure it is big enough. I did not move my home folder on the sd-card.

#### Step 2: cmake

The cmake version on the coral board did not work for me and there was no newer version available with apt. So I had to build it myself:

```bash
wget https://github.com/Kitware/CMake/releases/download/v3.18.4/cmake-3.18.4.tar.gz
tar -xf cmake-3.18.4.tar.gz
cd cmake-3.18.4/
./configure                       # takes a while
make                              # takes a while
sudo make install
```

#### Step 3: librealsense

Build realsense how it is described [here](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908), but with a few changes to the cmake command because I did not want the graphical examples and because cuda is not compatible with the board (i think?)

``` bash
wget https://github.com/IntelRealSense/librealsense/archive/master.zip
unzip master.zip
cd librealsense-master
mkdir build
cd build
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DFORCE_RSUSB_BACKEND:bool=true -DBUILD_WITH_CUDA:bool=false -DBUILD_GRAPHICAL_EXAMPLES:bool=false
make -j4 #might take almost a day
sudo make install
```

For the python-stuff to work properly I had to copy the \_\_init\_\_.py  file from the git-repo to the python3-folder as follows. The first path depends on where you unpacked the git repo.

``` bash
sudo cp ~/librealsense-master/wrappers/python/pyrealsense2/__init__.py /usr/lib/python3/dist-packages/pyrealsense2/
```

#### Step 4: opencv (optional)

OpenCV is in not required for this to work but I wanted to install it and it didn't work with pip, so here is how I did it. I followed mainly [this article](https://medium.com/@balaji_85683/installing-opencv-4-0-on-google-coral-dev-board-5c3a69d7f52f) but without the virtual environment (not recommended) and minus some steps that were already done before:

``` bash
wget -O opencv.tar.gz https://github.com/opencv/opencv/archive/4.4.0.tar.gz
wget -O opencv-contrib.tar.gz https://github.com/opencv/opencv_contrib/archive/4.4.0.tar.gz
tar -xf opencv.tar.gz
tar -xf opencv-contrib.tar.gz
cd opencv-4.4.0
mkdir build
cd build
 # OPENCV_EXTRA_MODULES_PATH has to be replaced with your download path!
cmake -D CMAKE_BUILD_TYPE=RELEASE \
 -D INSTALL_PYTHON_EXAMPLES=ON \
 -D INSTALL_C_EXAMPLES=OFF \
 -D OPENCV_ENABLE_NONFREE=ON \
 -D OPENCV_EXTRA_MODULES_PATH=/home/mendel/mnt/installation_cv2/opencv_contrib-4.4.0/modules \
 -D ENABLE_FAST_MATH=1 \
 -D ENABLE_NEON=ON -D WITH_LIBV4L=ON \
 -D WITH_V4L=ON \
 -D BUILD_EXAMPLES=ON ..
 make #takes >5h
 sudo make install
 sudo ldconfig
```

closed 2020-10-28 by MinPopp

MartyG-RealSense, 2020-10-28:
Thanks so much @MinPopp - guides such as these that are kindly contributed by RealSense community members are really helpful to others in future who are using the same equipment.   :)   It's great news that you were successful.