realsenseai/librealsense #7652: Install librealsense2 in RPI4 with Raspberry PI OS x64

AlessandroMelino, 2020-10-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | Last | 
| Operating System & Version | Raspberry PI OS 64 bits  | 
| Kernel Version (Linux Only)    | 5.4.51-v8+                                         | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  Last                         | 
| Language                            |                          | 
| Segment			|  Robot                      | 

### Issue Description

Hello.

I am trying to install from source realsense-ros in my Raspberry Pi4 with Raspberry PI OS x64 and ROS Noetic. When I try to resolve dependencies it notices that librealsense2 is not installed. So, how can I do to install librealsense2? I have not found anything similar.

Best regards.
Alessandro

MartyG-RealSense, 2020-10-26:
Hi @AlessandroMelino   If you are aiming to install librealsense and ROS on a Pi 4 with Raspbian OS then the resources in the link below will hopefully be of help to you.

https://github.com/IntelRealSense/realsense-ros/issues/1375#issuecomment-691482673

AlessandroMelino, 2020-10-26:
> Hi @AlessandroMelino If you are aiming to install librealsense and ROS on a Pi 4 with Raspbian OS then the resources in the link below will hopefully be of help to you.
> 
> [IntelRealSense/realsense-ros#1375 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/1375#issuecomment-691482673)

Following [librealsense installation guid](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md ) I got the following error:

```
sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbg libdrm-dev libdrm-exynos1 libdrm-exynos1-dbg libdrm-freedreno1 libdrm-freedreno1-dbg libdrm-nouveau2 libdrm-nouveau2-dbg libdrm-omap1 libdrm-omap1-dbg libdrm-radeon1 libdrm-radeon1-dbg libdrm-tegra0 libdrm-tegra0-dbg libdrm2 libdrm2-dbg
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libdrm-amdgpu1-dbg
E: Unable to locate package libdrm-exynos1-dbg
E: Unable to locate package libdrm-freedreno1-dbg
E: Unable to locate package libdrm-nouveau2-dbg
E: Unable to locate package libdrm-omap1-dbg
E: Unable to locate package libdrm-radeon1-dbg
E: Unable to locate package libdrm-tegra0-dbg
E: Unable to locate package libdrm2-dbg

```

In what repository are them?

Best regards.
Alessandro

MartyG-RealSense, 2020-10-26:
On a case where two other RealSense users experienced the same set of errors, a set of instructions to resolve the problems was posted.

https://github.com/IntelRealSense/librealsense/issues/5092#issuecomment-554190266

AlessandroMelino, 2020-10-27:
Hello @MartyG-RealSense .

That steps helped me and now it works. Thanks a lot.

Best regards.
Alessandro

closed 2020-10-27 by AlessandroMelino

MartyG-RealSense, 2020-10-27:
Great news @AlessandroMelino  - thanks for the update!