realsenseai/librealsense #7654: Is there any example about slam? Using T265 and re435

JimXu1989, 2020-10-26:
I remember there are some example for slam using T265 and rs435i, and the slam example is only for linux, while I can not find it now, could you show me where can I find it?

Thanks a lot!

RealSenseSupport, 2020-10-27:
This may be a resource for you:  https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

T265 has SLAM that runs on the device, that is built into the device.  The D435i does not which means that SLAM used for D435i must run on host platform and that slam is not supplied by Intel.  One must use a 3rd party SLAM algo to function with the D435i camera.

rlabs-oss, 2020-11-14:
Do you have an example where the T265 is translating and rotating points from a Lidar point cloud ? (or any point cloud)

JimXu1989, 2020-11-16:
> This may be a resource for you: https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i
>
> T265 has SLAM that runs on the device, that is built into the device. The D435i does not which [...]

I want use the position return by T265 and the 3D information return by D435i, the two devices is attatched together, is there any way to acheive this?

rlabs-oss, 2020-11-16:
@JimXu1989 yes use RTAB-Map standalone in dual mode. Ive used that for the T265 + D455 / T265 + D415. (or the ros route via the slam with d435i url)

closed 2026-10-23 by MartyG-RealSense