realsenseai/librealsense #7656: n Chip Calibration not able to recover corrupted data uploaded using the Intel.Realsense.CustomRW.exe

denis75, 2020-10-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |                          D415                              | 
| Firmware Version                   |                    05.12.07.100                        | 
| Operating System & Version |                          Win 10                           | 
| Kernel Version (Linux Only)    |                                                               | 
| Platform                                 |                                PC                            |
| SDK Version                            |                           2.38.1 / 2.39.0              | 
| Language                            |                                   C++ }                         | 
| Segment			|                                      Robot}                          | 

### Issue Description
<On Chip Calibration not able to recover corrupted data uploaded using the Intel.Realsense.CustomRW.exe>

Starting from Gold settings

[image]

Read data back using CustomRW.exe

[image]

Use printed pattern Random_image_10_90_10_1920_1080.png from Intel texture-pattern-set.zip.

Use Depth Quality Tool V 2.39.0 to verify the performance:

[image]

Upload corrupted data using Intel.Realsense.CustomRW.exe that modifies the focal length and principal point:

[image]

[image]

Use Depth Quality tool V 2.39.0 to check the distance and RMS error again:

[image]

Error in distance is now -97mm.

Use On-Chip Calibration in SLOW setting to recover the corrupted calibration data.
Target is printed pattern Random_image_10_90_10_1920_1080.png from Intel texture-pattern-set.zip.

[image]

Result of the calibration is reported as Good.

[image]

Calibrate also TARE => measured 694mm

[image]

Result is still Not as before  => 630mm

agrunnet, 2020-10-26:
On-chip self-calibration improves the depth map, not the accuracy. For accuracy you will need to run the on-chip Tare function.  
Two other things to note: 1. On-chip functions adjust the right camera (not the left). This does not mean that it does not recover changes that you made to the left, we just point it out if you are looking for the post-calibration changes, 2. You are adjusting the focal length of Fx, but not Fy. This is not something that physically will happen. Note that adjusting the Fx&Fy will "tilt" the depth map. This is not currently compensated for in the OCC, but is actually a feature that will be added soon 3. When adjusting Px, you are deliberately primarily changing the Z - the intrinsic OCC command will adjust the Py. These are almost orthogonal, as noted above. So you would not expect OCC to fix that, but Tare will.

denis75, 2020-10-26:
I used also the TARE calibration to try to recover. 
But reading back the data after On-Chip Calibration using TARE doesn't seem to change anything. Same values that I uploaded.
[image]

denis75, 2020-10-26:
[image]

denis75, 2020-10-28:
@agrunnet 
Thanks a lot for your feedback.  

After running the Tare calibration I have still quite some error left in the distance reported by the Depth Quality Tool (about 10%):

629mm after Tare calibration
694mm original value (same value measured with a Laser meter)

Is there something else I can try or should I conclude that for this specific case I would need to run the "OEM calibration" based on targets to restore the performances?

[image]

agrunnet, 2020-10-28:
@denis75 that is strange. Is you measure with a laser range finder the distance to a wall, and enter that distance during the tare procedure, then the camera should calibrate to that value. So presumably in your setup you would enter the number 694mm. On the screen it says the wall is tilted 22deg. Please try to mount it perpendicular and make sure to measure the center. If you feel the unit is pointed perpendicular then something else has gone wrong with the unit.

denis75, 2020-10-28:
The tilt is introduced artificially to simulate a bad unit in production (modifying Fx and Px). As you can see above the first image with the Gold settings was measuring 5deg. 
After the perturbation of the parameters Fx and Px I got this tilt in the depth map (and you kindly explained that this is not currently compensated for in the OCC but will be added soon).
The unit and the pattern are in the same position (we would like to have a fixed setup in production).

This is what I see with the Viewer (tried with version 2.38.1 but it doesn't make a difference)

[image]

The Health check reports Good:

[image]

And those are the final parameters after the 2 calibrations (On-Chip + Tare):

[image]

agrunnet, 2020-10-28:
@denis75  I see. The Tare function really only works well when there is no tilt. It is looking at the average depth near the center but slightly shifted to the right (which seems consistent with the fact that the right side is closer). So when it tries to align the depth map to the ground truth it is matching up those range numbers on-chip.   I emphasize that the On-chip calibration functions and Tare are meant to correct/heal units that have deviated a small amount from their perfect calibration. It is not a replacement for full-calibration for units that have experienced large perturbations, like strong defocus of one lens.  

So right now, one thing that is important is that when you adjust Fx please also adjust Fy by the same ratio. Second, in the future, the right process flow will be to run 1. OCC (On-chip self-calibration), then 2. A new yet-to-be-release function that tunes the focal length imbalance (coming soon...), and 3. Tare. That way tare has a properly aligned depth plane.

But I do want to emphasize that these techniques are meant to "re-tune" a unit that has deviated only slightly from perfect. If you are purposefully changing calibration numbers then at some point all these "small delta" assumptions are no longer valid and it really is a multi-variable problem that can only be solve through proper calibration.

denis75, 2020-10-28:
@agrunnet 
Many thanks for your time and explanations. I will close the issue (looking forward to the new function).

closed 2020-10-28 by denis75