realsenseai/librealsense #7658: The Depth data and RGB data  not match .The Depth data is wrong

hqindong, 2020-10-27:
1、The Depth data and RGB data  not match
2、The Depth data is wrong

[image]
[image]
[image]

MartyG-RealSense, 2020-10-27:
Hi @hqindong   Could you provide more detail about your project please before we start to look at why depth might be wrong or why depth and RGB might not match up?   

It looks as though you are using a Python script to read a bag file.  You also show a list of post-processing filters in the RealSense Viewer program for a live D435i camera.

Setting post-processing filters for the live D435i camera would have no effect on a pre-recorded bag file that is being read by a Python script.  If you wish to apply post-processing in a Python script then you have to add code to your script to apply the individual post-procesing filters that you want to use.

If you want to test the bag file in the RealSense Viewer, you can play the bag in the Viewer and apply post-processing filters to it from the options side-panel by drag and dropping the bag file into the central panel of the Viewer, or by using the **Add Source** button and selecting the **Load Recorded Sequence** drop-down menu option.

hqindong, 2020-10-28:
My purpose is to get a picture and its corresponding  more accurate depth  value through d435 like the picture below. What should I do?
[image]

First， I found that the depth map seen through the SDK is very inaccurate. There are black holes or afterimages on the boundary. The depth values at different distances are similar, and there are many holes.
[image]

 So I parsed the picture and depth value matrix from the BAG file, and found that the depth value matrix is like the picture above, with almost no color.
[image]

MartyG-RealSense, 2020-10-28:
Images in applications created in the RealSense SDK can look different to those in the RealSense Viewer because the Viewer applies a range of **post-processing filters** by default.  

Programs that you create yourself have no post-processing filters applied by default, and you have to deliberately code them into your application yourself.  For example, the Viewer has a **hole-filling filter** that can close up the black dots that you see on the image in your own application.  

The link below provides a tutorial for adding post-processing filters in Python.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

If you are not familiar with the SDK's post-processing filters, I recommend reading Intel's documentation on the subject.

https://dev.intelrealsense.com/docs/post-processing-filters
https://dev.intelrealsense.com/docs/depth-post-processing

If you find that you are getting similar depth readings from different distances then **aligning depth to color** can help with this, as it makes it easier to distinguish between background and foreground pixels.  Intel also provide a Python tutorial for alignment:

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

If you wish to change how depth is rendered as color then you can change the **color scheme** in Python with the colorizer.  Below is a link for scripting in the Python language to set the colorizer color scheme with **rs.option.color_scheme**

https://github.com/IntelRealSense/librealsense/issues/3363

closed 2020-11-13 by MartyG-RealSense