realsenseai/librealsense #7660: Install pyrealsense on ARM

RarDay, 2020-10-27:
Hello! 
I read a lot of questions, but they are all old. Actually I want to ask how can I install pyrealsense on my ARM processor (Nvidia Jetson Nano, python 3.6 or 3.7)?

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version | Ubuntu 18.04 | 
| Platform                                 | NVIDIA Jetson |
| Language                            | Python 3.6 / 3.7 | 

### Issue Description
Yes, you all know this error)

```
desktop:~$ python3.7 -m pip install pyrealsense2
Collecting pyrealsense2
  Could not find a version that satisfies the requirement pyrealsense2
No matching distribution found for pyrealsense2
```

MartyG-RealSense, 2020-10-27:
Hi @RarDay   The pip pyrealsense2 packages do not work with devices that have Arm processors such as Jetson and Raspberry Pi, as the packages require x86 or x64 type procesors.  On Jetson, the Python wrapper therefore should be built from source code.

The link below contains installation instructions that have worked for Nano users during the past week:

https://github.com/IntelRealSense/librealsense/issues/7480#issuecomment-713498309

RarDay, 2020-10-27:
@MartyG-RealSense oh, it's no easy, i do last point:
update your PYTHONPATH environment variable to add the path to the pyrealsense library
`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`
And what i must do next?

MartyG-RealSense, 2020-10-27:
If you are following **Option 2** of the instructions in the link to install the Python wrapper only:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

`export PYTHONPATH=$PYTHONPATH:/usr/local/lib` is the final step of the instructions.  So you would then test whether the Python wrapper is working correctly by creating a Python test script, such as the example one in the documentation:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

RarDay, 2020-10-29:
Yes, i do Option 2, i do all step without error, i do the final step and create on desctop .py file with example, then i get this error:
```
Traceback (most recent call last):
  File "/media/282156b7-5133-4579-aea8-4e88b73e9891/home/r415/test.py", line 1, in <module>
    import pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2'
```
mb i must create .py file in another directory?

MartyG-RealSense, 2020-10-29:
If you are using Python 3.7, could you try the PYTHONPATH below please:

**export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.7/**

yosoufe, 2020-11-02:
run the cmake like this:
```
cmake ../ \
    -DBUILD_EXAMPLES=true \
    -DFORCE_LIBUVC=true \
    -DBUILD_WITH_CUDA=true \
    -DCMAKE_BUILD_TYPE=release \
    -DBUILD_PYTHON_BINDINGS=bool:true \
    -DCMAKE_INSTALL_PREFIX=~/librealsense_binary \
    -DPYTHON_INSTALL_DIR=~/librealsense_binary/python \
    -DPYBIND11_INSTALL=ON \
    -DCMAKE_CUDA_COMPILER=/usr/local/cuda/bin/nvcc \
    -DPYTHON_EXECUTABLE=$(python3.6 -c "import sys; print(sys.executable)")
```

If you have `python 3.7` installed, change the last line to `python3.7`. Then it compiles for `python 3.7`
Then add `export PYTHONPATH=$PYTHONPATH:$HOME/librealsense_binary/python` to your `~/.bashrc `

MartyG-RealSense, 2020-11-02:
@yosoufe  Thanks for your advice for @RarDay   :)

RarDay, 2020-11-02:
@yosoufe Hi, i do cmake, its run without error, but i dont have no one file "pyrealsense2.cpython..."
About version i have 3.5 3.6 3.7, version not important, only work lib important )

RarDay, 2020-11-02:
And i dont have librealsense_binary, only librealsense with:
```
appveyor.yml           code-of-conduct.md  doc       package.xml  tools
build                  common              examples  readme.md    unit-tests
CMake                  config              include   scripts      wrappers
CMakeLists.txt         connectivity_check  LICENSE   src
cmake_uninstall.cmake  CONTRIBUTING.md     NOTICE    third-party

```

yosoufe, 2020-11-02:
@RarDay what is the exact cmake you run? And also make sure that your build directory is completely empty before running cmake. Because with pybind11 usually you cannot change the interpreter when you ran it once. So the only way I know, is to remove and recreate the build directory from scratch.

yosoufe, 2020-11-02:
please run  `find -name "*pyrealsense2*"` in `librealsense` directory after the compile. And let me know the output.

RarDay, 2020-11-02:
@yosoufe hmm good question, i run it but i dont create folder, now i create folder 'librealsense_binary' and after start i have(my folder librealsense_binary is empty after run):
```
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_FW_Image-5.12.8.200.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for SR3XX_FW_Image-3.26.1.0.bin
-- http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for target-0.2.0.951.mvcmd
-- http://realsense-hw-public.s3.amazonaws.com/Releases/L5xx/FW/L5XX_FW_Image-1.5.1.3.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for L5XX_FW_Image-1.5.1.3.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/r415/librealsense/CMake
```
It is 
```
./wrappers/python/pyrealsense2
./CMake/wrappers/python/pyrealsense2ConfigVersion.cmake
./CMake/wrappers/python/CMakeFiles/pyrealsense2.dir
./CMake/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2
./CMake/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2/pyrealsense2Targets-release.cmake
./CMake/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2/pyrealsense2Targets.cmake
./CMake/wrappers/python/pyrealsense2Config.cmake
./CMake/pyrealsense2Config.cmake.in
./CMake/pyrealsense2-netConfig.cmake.in
./build/wrappers/python/pyrealsense2ConfigVersion.cmake
./build/wrappers/python/CMakeFiles/pyrealsense2.dir
./build/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2
./build/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2/pyrealsense2Targets-noconfig.cmake
./build/wrappers/python/CMakeFiles/Export/lib/cmake/pyrealsense2/pyrealsense2Targets.cmake
./build/wrappers/python/pyrealsense2.so.2.39.0
./build/wrappers/python/pyrealsense2.so.2.39
./build/wrappers/python/pyrealsense2.so
./build/wrappers/python/pyrealsense2Config.cmake
```

yosoufe, 2020-11-02:
@RarDay  
- I guess you compiled for python 2.7, you can test with
```
cd ./build/wrappers/python/
python2 -c "import pyrealsense2" # if there is no error, it is imported and built correctly but for python 2
```  
- `make` should create the `librealsense_binary` directory, not you.
- I think you didn't run the cmake command as I told you. Or you did not delete and recreate the build directory.

We can try something else which I do not know if it is going to work. I  compiled the library for Jetson Nano, JP 4.4.1 yesterday and you can download my install (binary) directory [from here](https://drive.google.com/file/d/1AupByIRo8kuOn9l6aMy6OH-09u45RxTe/view?usp=sharing) to your jetson nano. Just unzip it somewhere and it has `python/pyrealsense2.cpython-*` files for both 3.6 and 3.7 (compiled in two different times, you cannot create both of them simultaneously). you can test by
```
cd <location of unzip>/python
python3.6 -c "import pyrealsense2"
```
no errors mean it works. if it worked you can set the PYTHONPATH to this directory.

If you want to compile it yourself (**recommended**), This is the procedure I followed. You can follow it but please do it exactly:
https://github.com/yosoufe/SelfStudyRobotics/blob/master/hardware/realsense.md#compile-for-jetson-nano

RarDay, 2020-11-02:
@yosoufe 
yes, python2 -c "import pyrealsense2" - started without errors, its work
Start download your zip

RarDay, 2020-11-02:
Oh, Yes, your library is working. You and the genius. why no one has ever taken it off before. Unbelievable of course. Thanks!

```
                                         .hXWWWWWh:             
                                         BWWWWWWWWWW:           
                                         BWWWWWWWWWWW           
                                         WWWWWWWWWWWW:          
                                        .WWWWWWWWWWWWW.         
                                         BWWWWWWWWWWWX          
                                         XWWWXBXWWXX:           
                                         .WWn:WWWWWWWh:..       
                                          :n:WWXWWWWWWWWWWB:::  
                                      .:nhXB.hWWnWWWXWWWWWWWXXWh
                                   nXWWWWWWWWBXWWWh  WWWWWWWWBXW
                                 hWWWWWWWWWWWWWWWWWX .BWWWWWWWWX
                               .WWWWWWWWWWWWWWWWWWWWh  XWWWWWWWW
                             .hWWWWWWWWWWWWWWWWWWWWWWh .XWWWWWWW
                           :BWWWWWWWWWWWWWWWWWWWWWWWWWX:.:WWWWWW
                        :hWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWn :WWWWW
                  .nhBBWWWWWWWWWWXnnnhWWWWWWWWWWWWWWWWWWW .WWWWW
             .nhXWWWWWWWWWWWWWWX:     BWWWWWWWWWWWWWWWWWWX. hWWW
       .:nBXWWWWWWWWWWWWWWWWWh.        XWWWWWWWWWWWWWWWWWWB. hWW
       WWWWWWWWWWWWWWWWWWWXn           nWWWWWWWWWWWWWWWWWWWW. nW
       WWWWWWWWWWWWWWWWWWWBnhBXXXXBBBhhBWWWXXXXXWWWWWWWWWXWWX: .
       BWWWWWWWWWWWWWWXXXWhhBBBXWWWWWWWWWWWWWWWWXWXWWhhn.   hXn 
```

edwin9610, 2020-11-06:
> @RarDay
> 
> * I guess you compiled for python 2.7, you can test with
> 
> ```
> cd ./build/wrappers/python/
> python2 -c "import pyrealsense2" # if there is no error, it is imported and built correctly but for python 2
> ```
> 
> * `make` should create the `librealsense_binary` directory, not you.
> * I think you didn't run the cmake command as I told you. Or you did not delete and recreate the build directory.
> 
> We can try something else which I do not know if it is going to work. I compiled the library for Jetson Nano, JP 4.4.1 yesterday and you can download my install (binary) directory [from here](https://drive.google.com/file/d/1AupByIRo8kuOn9l6aMy6OH-09u45RxTe/view?usp=sharing) to your jetson nano. Just unzip it somewhere and it has `python/pyrealsense2.cpython-*` files for both 3.6 and 3.7 (compiled in two different times, you cannot create both of them simultaneously). you can test by
> 
> ```
> cd <location of unzip>/python
> python3.6 -c "import pyrealsense2"
> ```
> 
> no errors mean it works. I can see a good chance that you see some sort of `refrence not found` error by linker. if it worked you can set the PYTHONPATH to this directory. If it didn't, you have to compile it yourself as follow.
> 
> If you want to compile it yourself (**recommended**), This is the procedure I followed. I documented what I did in the following link, section `Compile for Jetson Nano`:
> https://github.com/yosoufe/SelfStudyRobotics/blob/master/hardware/realsense.md#compile-for-jetson-nano

@yosoufe Hi! I did the same by running python3.6 -c "import pyrealsense2" and it showed no problem. But when I run the script it's still showing 
ModuleNotFoundError: No module named 'pyrealsense2'

I also added export PYTHONPATH=$PYTHONPATH:/home/nano/librealsense_binary/python to my ~/.bashrc but it still does not work. I'm still new to Linux and I don't know what else I can do. This problem has been bothering me for few days. :(

MartyG-RealSense, 2020-11-06:
Hi @edwin9610   A RealSense user posted last month a guide for the method that worked for them for solving the **No module named 'pyrealsense2'** error on Nano:

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

yosoufe, 2020-11-07:
@edwin9610 please follow @MartyG-RealSense 's recommendation. 

For your case I guess you have not sourced the ~/.bashrc. Still let's do some checking:
1. Is it the same on a new terminal? check with `echo $PYTHONPATH` in the same terminal you are running your script and see if the directory of the `librealsense` python binary is there.
2. How do you run your python script? If you are running with pycharm, you should know that pycharm does not read PYTHONPATH environment variable and you should manually add that path to your pycharm. https://stackoverflow.com/a/53439589/9516701

edwin9610, 2020-11-08:
The pathi is already there. 

nano@nano-desktop:~$ echo $PYTHONPATH
:/home/nano/librealsense_binary/python:/home/nano/librealsense/build/wrappers/python:/usr/local/lib:/usr/local/lib/python3.6/pyrealsense2:/home/nano/librealsense_binary/python:/home/nano/librealsense/build/wrappers/python:/usr/local/lib:/usr/local/lib/python3.6/pyrealsense2

MartyG-RealSense, 2020-11-23:
Case closed due to no further comments received.

closed 2020-11-23 by MartyG-RealSense

traversaro, 2024-06-26:
In case someone find this discussion looking for a `arm64` binary package for `pyrealsense2`, if you are using conda a suitable package is build by conda-forge, see https://github.com/conda-forge/librealsense-feedstock .

**IMPORTANT**: This build is community maintained and not maintained by realsense developers, so if you find any issue in that, please open an issue at https://github.com/conda-forge/librealsense-feedstock .

MartyG-RealSense, 2024-06-26:
Thanks so much for the share, @traversaro 

The pyrealsense2 wrapper can now be installed on Arm devices via a .whl (wheel) package with the **pip install pyrealsense2** command and supports Python 3.7 to 3.9, though this installation method does not work on Raspberry Pi and Nvidia Jetson boards.

traversaro, 2024-06-26:
> The pyrealsense2 wrapper can now be installed on Arm devices via a .whl (wheel) package with the **pip install pyrealsense2** command and supports Python 3.7 to 3.9, though this installation method does not work on Raspberry Pi and Nvidia Jetson boards.

Thanks for the info! I ended up here as one of my colleagues had indeed this problem on a Nvidia Jetson board for a Python version >= 3.10 I guess.

jbcpollak, 2024-07-16:
is there any chance of getting an updated pyrealsense for python 3.11 or newer soon? It's kind of a drag to be limited to python 3.9.

MartyG-RealSense, 2024-07-17:
Hi @jbcpollak  The Arm wheel files for the 2.55.1 version of the pyrealsense2 wrapper support Python 3.10.  

[image]

I do not currently have information about when a wheel file for Python 3.11 might be released.  However, if the pyrealsense2 wrapper is built from source code on Arm instead of using **pip install pyrealsense2** then newer Python versions such as 3.11 and 3.12 are supported.