realsenseai/librealsense #7662: L515 not showing up in Unity #6859 [part2]

AndreV84, 2020-10-27:
is there any update since https://github.com/IntelRealSense/librealsense/issues/6859 ?

dorodnic, 2020-10-28:
Support for recent versions of Unity was merged lately - 
https://github.com/IntelRealSense/librealsense/pull/7625
It will be part of upcoming release.

AndreV84, 2020-10-28:
what about the support of recent version of realsense models?
we are trying on unity 2019.4
what worked was d435i
what dud nit work was L515. d455
what do we do to get L515 working?

AndreV84, 2020-10-28:
does it mean that until the upcoming release it won't be possoble to use L515?

dorodnic, 2020-10-28:
You can build the wrapper from the development branch. 
L515 and D455 are supported for Unity 2017, that was the only version we supported until now.

AndreV84, 2020-10-28:
@dorodnic 
Could you clarify exact steps to get L515 in Univty working, please?
1. set up 2017 unity? 2019.4 unity? both? neither? any of the two?
2. downbload the repository using clone method? like 
```
git-clone https://github.com/IntelRealSense/librealsense/
cd librealsense
git -b checkout dev 
```
Then the branch could be build for windows ? for linux? for both? neither of the two?
with exact steps? which exact steps?
could you extend, please?

AndreV84, 2020-10-28:
so do I clone the realsense like that?
```
git clone -b dev https://github.com/IntelRealSense/librealsense/i
```
then start unity 2019? or 2017? then import the realsense unity package into it?
or some build provcedure also needs to be made?

dorodnic, 2020-11-04:
Please read the instructions here - https://github.com/IntelRealSense/librealsense/tree/development/wrappers/unity
Every github release includes unitypackage for Unity 2017
Both L515 and D400 are supported

AndreV84, 2020-11-05:
Hi @dorodnic 
How is it going?
Thank you fo following up!
Do you imply that while there is the unity package supplied for unity 2017
there is no unity package for unity 2019 supplied via github?
Also that the instructions for building do not need to be applied as long as the suppolied unity package is used with unity 2017?
But will the build instructions be sufficient to build the unitypackage for unity 2019?
If so could you build the unity package with the build instructions for unity 2019, please?

leeprobert, 2020-12-14:
The L515 was recognised in the wrapper that is bundled with SDK 2.0 (v2.40.0) but you only get access to the depth and infrared stream.

RealSenseSupport, 2021-01-13:
@leeprobert 

Which version of Unity package did you try?

closed 2021-02-10 by RealSenseSupport