realsenseai/librealsense #7663: Capturing frames without frame drops

a-m-p-m, 2020-10-27:
Hello, 

I hope this is an appropriate place for the following question: 

Given that there is a substantial level of frame drops in RealSense Viewer at high fps (e.g., 90 fps for 848x480 frame size: https://github.com/IntelRealSense/librealsense/issues/7488), did anyone had a success capturing and saving frames from the camera (e.g., D435) using there own piece of software **without** frame drops for 848x480 @ 90 fps setting?  

Thanks.

MartyG-RealSense, 2020-10-28:
Hi @a-m-p-m   If you are saving the frames to disk in real-time as they are captured then there is the possibility of bottlenecks occurring with the computer hardware, such as a storage device with slow access speeds (e.g a non-SSD hard drive).  

Also, are you using the official 1 meter cable supplied with the camera please?   If you use your own cable, the choice of cable can have a bearing on data transfer due to varying quality of third-party USB cables.  Intel's officially recommended USB cable supplier for the 400 Series cameras is **Newnex**, who can provide industrial-grade USB equipment.

https://www.newnex.com/realsense-3d-camera-connectivity.php

If you are recording for a short period (e.g up to 30 seconds) then an SDK function called **Keep()** may suit you.   It records frames to the computer's memory instead of writing them to file.  When the pipeline is closed, you can do batch-processing on the frames such as align and post-processing and then save the data to file.  The technique consumes computer memory capacity though, which is why it is best suited to short recording durations.

Another way to reduce the amount of bandwidth is to apply depth image compression via colorization, as described in the Intel white-paper on the subject in the link below:

https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras

MartyG-RealSense, 2020-11-13:
Case closed due to no futher comments received.

closed 2020-11-13 by MartyG-RealSense