realsenseai/librealsense #7664: What is considered the origin on the d435i camera?

blakat360, 2020-10-28:
What is considered the origin on the d435i camera? When we get distances, from which point are they calculated? Does this change when aligning the depth feed to the colour feed?

MartyG-RealSense, 2020-10-28:
Hi @blakat360   On the 400 Series cameras, the origin is the center of the left imager.

https://github.com/IntelRealSense/librealsense/issues/2688

When measuring distance, the cased USB models measure depth from the front glass and the caseless Depth Kit Module models measure from the front of the lens or the back of the module.   Details about this can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/6169#issuecomment-606487195

The origin would remain the center of the left imager during depth to color alignment.

The link below provides interesting further reading about the origin.

https://github.com/IntelRealSense/librealsense/issues/3072

blakat360, 2020-10-29:
Thanks for this. When you say left imager, is that left looking from behind the camera (where the observer just sees intel logo on camera back) or left when I'm looking from in front of the camera (when the observer sees the lenses)?

MartyG-RealSense, 2020-10-29:
It is on the right side when looking at it from the front.  agrunnet the Chief Technical Officer for the RealSense Group at Intel confirms this in the link below:

https://github.com/IntelRealSense/librealsense/issues/3662

closed 2020-10-30 by blakat360