realsenseai/librealsense #7666: Error type: Access Violation!

meamm, 2020-10-28:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2020-10-28:
Hi @meamm   There was a previous case of this error in the link below. 

https://github.com/IntelRealSense/librealsense/issues/7227

Following the lines of enquiry applied by the RealSense team member in that case, may I ask these questions please:

1.   Which camera firmware driver version are you currently using.

2.  Are you using the camera on a USB 2 or a USB 3 connection?

dorodnic, 2020-10-28:
Please try https://github.com/IntelRealSense/librealsense/releases/tag/v2.39.0 , this issue should be resolved there

MartyG-RealSense, 2020-11-05:
Hi @meamm   Do you still require assistance with this case, please?   Thanks!

meamm, 2020-11-06:
Thank you so much for your help! I've solved this problem already.It turns out the problem of using wrong type of interface. Thank you!

MartyG-RealSense, 2020-11-06:
Excellent news that you found a solution @meamm - thanks so much for the update!

closed 2020-11-06 by MartyG-RealSense