realsenseai/librealsense #7667: Calibration procedure in Viewer and Python wrapper questions

BobrG, 2020-10-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I have some troubles with understanding depth cameras calibration procedure. I was doing both Python calibration (described here https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/depth_auto_calibration_example.py) and Realsense Viewer calibration. Each procedure left some questions.  
1. What data contains calibration_table? Can I get updated intrinsic/extrinsic from this table? Is it possible to format it in some more informative view instead of just list of numbers?
2. Should there be any changes in Python code after on-chip calibration performed in Realsense Viewer or with Python wrapper? Where should I look then, cause depth_profile.get_intrinsics() and depth_profile.get_extrinsics_to(color_profile) methods return the same numbers before and after applying new calibrations in Realsense Viewer or setting new calibration table in wrapper? 
3. I can examine Calibration data in Realsense Viewer and verify that the calibration procedure makes some difference in the device parameters, but I do not understand how to find the same data in Python wrapper afterward. If these changes are burnt to the device, then I suppose I should have the ability to verify the same changes in wrapper, but I cannot find how to do it.
Thanks in advance!

MartyG-RealSense, 2020-10-28:
Hi @BobrG   I must emphasize that calibration tables are not one of my specialist areas of knowledge, so I will attempt to address your initial questions and research any follow-up questions that emerge from that.

1.  As far as I know, there is not a reference that explicitly states the stored contents of a calibration table.  There is documentation of extrinsic and intrinsic **calibration parameters**.

https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide#section-2-1-calibration-parameters

The comment by @dorodnic the RealSense SDK Manager in the link below describes how you can retrieve extrinsic and intrinsic calibration information with Python:

https://github.com/IntelRealSense/librealsense/issues/3986#issuecomment-492758545

I would think that the values retrieved by the methods described in that link could be substituted into easy to read printed statements, such as **"The focal length is", FocalLength**  (not actual Python code!)

2.  If you are asking whether calibration values will be updated after a calibration has been performed then I would think the answer would be yes, if you have written the changes to the camera.  If you are getting the same table values before and after calibration, a recent case where this was happening to another RealSense user may be helpful.

https://github.com/IntelRealSense/librealsense/issues/7562

3.  The calibration table can be read with **get_calibration_table()**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/depth_auto_calibration_example.py#L56

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.auto_calibrated_device.html#pyrealsense2.auto_calibrated_device.get_calibration_table

**********

I hope that I have been able to provide information that assists with the majority of your questions.   Please do ask further questions if there are points that you would like further clarification on.

BobrG, 2020-10-29:
Hi @MartyG-RealSense Thank you for providing information! I'm satisfied with the answer to my first question, but about the second and third I would like to ask for more details. 
You are telling that changes in calibration data are burnt to the device and consequently they are updated everywhere - in Viewer and in py-wrapper as well. My concern is that I don't understand how to check this in py-wrapper. In the link you've presented only Viewer is discussed and there are no mentions of py-wrapper. I understand that I can compare to Calibration data tables saved as jsons from Viewer and verify that calibration does smth. What I do not understand is how to do same comparison in py-wrapper after calibration in Viewer. Should I compare lists from get_calibration_table? 
Another thing - you've provided a link to a procedure for obtaining intrinsic and extrinsic parameters in py-wrapper. However, when I follow this procedure before and after calibration - I see no changes in this data. Intrinsic and extrinsic stay the same. At the same time, intrinsic data from Viewer calibration data table has some changes. Can you please comment on this?  Why some data changes and another stays the same?

MartyG-RealSense, 2020-10-29:
As mentioned in the @dorodnic link, you can get extrinsics through Python with the instruction **get_extrinsics_to**.  Point **2. 
 Is there a direct way I can get the extrinsic parameters for the camera?** of the comment linked to below will hopefully be a useful reference:

https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888

In regard to getting intrinsics with the Python wrapper, I refer you to this comment:

https://github.com/IntelRealSense/librealsense/issues/869#issuecomment-523071259

An alternative method of obtaining intrinsics in Python with **get_intrinsics()** is referenced here:

https://github.com/IntelRealSense/librealsense/issues/2869

If you are calibrating with the Dynamic Calibrator tool, it is worth bearing in mind that the normal free version of this tool does not calibrate intrinsics, only extrinsics.  A special OEM version of the Dynamic Calibrator tool is required to calibrate both intrinsics and extrinsics.  @dorodnic is better able to explain the workings of the calibration tools though.

BobrG, 2020-10-30:
@MartyG-RealSense I forgot to specify that I am using On-Chip calibration. Can you please tell me should this code 
```depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics```
produce **different** outputs **before** and **after On-Chip calibration**. Mine returns **same** intrinsic despite doing on-chip calibration.

MartyG-RealSense, 2020-10-30:
When the On-Chip calibration was completed, did you click the **Apply Now** button to write the new calibration to the camera please?

BobrG, 2020-10-31:
Yes, I did this. Calibration data has changed in Viewer but in Python everything stays the same.

MartyG-RealSense, 2020-10-31:
I researched your question about Python carefully but it is outside of my current programming knowledge unfortunately.  Another test you could do to compare calibration before and after On-Chip calibration to confirm whether the calibration has changed is to use the command **rs-enumerate-devices -c** (where 'c' refers to calibration).

BobrG, 2020-10-31:
Ok, I will try this. But still, I'm interested in py-wrapper. Do you know someone who can help me with Python part? Intrinsics and extrinsics stay the same in python even after on-chip calibration in py-wrapper and this is kinda weird for me.

MartyG-RealSense, 2020-10-31:
I wll tag @RealSenseSupport into this conversation.

MartyG-RealSense, 2020-11-07:
Reminder to @RealSenseSupport to please take a look at this case.  Thanks!

BobrG, 2020-11-13:
still require expertise @RealSenseSupport, thanks

MartyG-RealSense, 2020-11-13:
@BobrG  It may be worth following this new case:

https://support.intelrealsense.com/hc/en-us/community/posts/360051257494/comments/360014851593

closed 2020-11-26 by MartyG-RealSense