realsenseai/librealsense #7669: How to connect more than one camera with Thread?

JANGSOONMYUN, 2020-10-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| Language                            |  C++ / MFC                 | 

I'm trying to connect multi-cameras with Thread.
But I don't use the example code of 'multicam'. 

Here are functions that I used for camera connection and acquiring images.

```
void myRealSense::init(std::string cam_id)
{
	cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
	cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8, 30);
	cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
	cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);

	cfg.enable_device(cam_id);
	serial_num = cam_id;
	selection = pipe.start(cfg);
}

void myRealSense::getDepthImage(Mat & img, Mat& ir_ima_for_show, Mat & d_img)
{
	frames = pipe.wait_for_frames();
	
	rs2::frame ir_frame = frames.get_infrared_frame(1);
	depth_frame = frames.get_depth_frame();

	color_frame = frames.get_color_frame();
	if (color_frame) {
		// Creating OpenCV Matrix from a color image
		img = Mat(Size(1280, 720), CV_8UC3, const_cast<void*>(color_frame.get_data()), Mat::AUTO_STEP);
	}
	if (depth_frame) {
		ir_ima_for_show = Mat(Size(1280, 720), CV_8UC1, (void*)ir_frame.get_data(), Mat::AUTO_STEP);
		d_img = Mat(Size(1280, 720), CV_16UC1, const_cast<void*>(depth_frame.get_data()), Mat::AUTO_STEP);
	}
}
```

It worked well when I only connect one camera.
But if I connect one more camera, image acquiring is very slow. And eventually, an error below occurs.

> Unhandled exception at 0x00007FFFEC37D1D7 (opencv_world331d.dll) in MFC_3D_intelRelsense_D435_Panel_Aligment.exe: 0xC0000005: Access violation reading location 0x0000000000000024.
> 
> If there is a handler for this exception, the program may be safely continued.

Please help me to solve the problem.

MartyG-RealSense, 2020-10-29:
Hi @JANGSOONMYUN   I believe that you are asking how to put multiple cameras on multiple threads.   Is that correct, please?

If so, and you do not wish to use the **rs-multicam** code, then the long discussion in the link below is an excellent reference about programming access to multiple cameras in C++.

https://github.com/IntelRealSense/librealsense/issues/2219

In the middle of this discussion, @dorodnic the RealSense SDK Manager offers a script for processing with **pipeline and threads**:

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412351431

And finally at the end of the discussion, a finished script is posted by the RealSense user who created the discussion:

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468

JANGSOONMYUN, 2020-11-02:
@MartyG-RealSense Thanks to you I solved the problem. I appreciate.

closed 2020-11-02 by JANGSOONMYUN

MartyG-RealSense, 2020-11-02:
Great to hear - thanks for the update!