realsenseai/librealsense #7671: Unable to compile C++ examples

kylelscott, 2020-10-29:
Hello,
Thank you for tending to my issue. I am working off of a NVIDIA Jetson Nano and I was able to install librealsense and build pyrealsesne2 from source. I have the examples from the python wrapper using pyrealsense working well, however, when I try to compile the c++ examples in the examples directory I get some errors I do not understand. I usually develop in python, so I am a little new to compiling and running c++ files in linux terminal; but when I use "g++ foo.cpp" I get some debug information I do not understand how to solve. Can anyone help me, I was not able to find a similar issue already opened? The librealsense.so library files are also located in my /usr/local/lib directory, so I am not sure what I am missing. Thank you. Below is an example when I try to compile the rs-measure.cpp

`e$ g++ rs-measure.cpp
/tmp/ccUwrpTI.o: In function `set_viewport(rect const&)':
rs-measure.cpp:(.text+0x1a4c): undefined reference to `glViewport'
rs-measure.cpp:(.text+0x1a50): undefined reference to `glLoadIdentity'
rs-measure.cpp:(.text+0x1a58): undefined reference to `glMatrixMode'
rs-measure.cpp:(.text+0x1a88): undefined reference to `glOrtho'
/tmp/ccUwrpTI.o: In function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
rs-measure.cpp:(.text+0x1ae8): undefined reference to `glLoadIdentity'
rs-measure.cpp:(.text+0x1af0): undefined reference to `glPushAttrib'
rs-measure.cpp:(.text+0x1b1c): undefined reference to `glClearColor'
rs-measure.cpp:(.text+0x1b24): undefined reference to `glClear'
rs-measure.cpp:(.text+0x1b2c): undefined reference to `glMatrixMode'
rs-measure.cpp:(.text+0x1b30): undefined reference to `glPushMatrix'
rs-measure.cpp:(.text+0x1b64): undefined reference to `gluPerspective'
rs-measure.cpp:(.text+0x1b6c): undefined reference to `glMatrixMode'
rs-measure.cpp:(.text+0x1b70): undefined reference to `glPushMatrix'
rs-measure.cpp:(.text+0x1b98): undefined reference to `gluLookAt'
rs-measure.cpp:(.text+0x1bc8): undefined reference to `glTranslatef'
rs-measure.cpp:(.text+0x1be0): undefined reference to `glRotated'
rs-measure.cpp:(.text+0x1bf8): undefined reference to `glRotated'
rs-measure.cpp:(.text+0x1c08): undefined reference to `glTranslatef'
rs-measure.cpp:(.text+0x1c1c): undefined reference to `glPointSize'
rs-measure.cpp:(.text+0x1c24): undefined reference to `glEnable'
rs-measure.cpp:(.text+0x1c2c): undefined reference to `glEnable'
rs-measure.cpp:(.text+0x1c44): undefined reference to `glBindTexture'
rs-measure.cpp:(.text+0x1c98): undefined reference to `glTexParameterfv'
rs-measure.cpp:(.text+0x1ca8): undefined reference to `glTexParameteri'
rs-measure.cpp:(.text+0x1cb8): undefined reference to `glTexParameteri'
rs-measure.cpp:(.text+0x1cc0): undefined reference to `glBegin'
rs-measure.cpp:(.text+0x1d50): undefined reference to `glVertex3fv'
rs-measure.cpp:(.text+0x1d68): undefined reference to `glTexCoord2fv'
rs-measure.cpp:(.text+0x1d7c): undefined reference to `glEnd'
rs-measure.cpp:(.text+0x1d80): undefined reference to `glPopMatrix'
rs-measure.cpp:(.text+0x1d88): undefined reference to `glMatrixMode'
rs-measure.cpp:(.text+0x1d8c): undefined reference to `glPopMatrix'
rs-measure.cpp:(.text+0x1d90): undefined reference to `glPopAttrib'
/tmp/ccUwrpTI.o: In function `draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::vector<rs2_vector, std::allocator<rs2_vector> >&)':
rs-measure.cpp:(.text+0x20e8): undefined reference to `glLoadIdentity'
rs-measure.cpp:(.text+0x20f0): undefined reference to `glPushAttrib'
rs-measure.cpp:(.text+0x211c): undefined reference to `glClearColor'
rs-measure.cpp:(.text+0x2124): undefined reference to `glClear'
rs-measure.cpp:(.text+0x212c): undefined reference to `glMatrixMode'
rs-measure.cpp:(.text+0x2130): undefined reference to `glPushMatrix'
rs-measure.cpp:(.text+0x2164): undefined reference to `gluPerspective'
rs-measure.cpp:(.text+0x216c): undefined reference to `glMatrixMode'
rs-measure.cpp:(.text+0x2170): undefined reference to `glPushMatrix'
rs-measure.cpp:(.text+0x21a0): undefined reference to `glTranslatef'
rs-measure.cpp:(.text+0x21b8): undefined reference to `glRotated'
rs-measure.cpp:(.text+0x21d0): undefined reference to `glRotated'
rs-measure.cpp:(.text+0x21e0): undefined reference to `glTranslatef'
rs-measure.cpp:(.text+0x21e8): undefined reference to `glEnable'
rs-measure.cpp:(.text+0x21f0): undefined reference to `glLineWidth'
rs-measure.cpp:(.text+0x21f8): undefined reference to `glBegin'
rs-measure.cpp:(.text+0x224c): undefined reference to `glColor3f'
rs-measure.cpp:(.text+0x2268): undefined reference to `glVertex3f'
rs-measure.cpp:(.text+0x2278): undefined reference to `glEnd'
rs-measure.cpp:(.text+0x2280): undefined reference to `glLineWidth'
rs-measure.cpp:(.text+0x2290): undefined reference to `glColor3f'
rs-measure.cpp:(.text+0x22cc): undefined reference to `glMultMatrixf'
rs-measure.cpp:(.text+0x22d4): undefined reference to `glMultMatrixf'
rs-measure.cpp:(.text+0x22e8): undefined reference to `glPointSize'
rs-measure.cpp:(.text+0x22f0): undefined reference to `glEnable'
rs-measure.cpp:(.text+0x22f8): undefined reference to `glEnable'
rs-measure.cpp:(.text+0x2310): undefined reference to `glBindTexture'
rs-measure.cpp:(.text+0x2364): undefined reference to `glTexParameterfv'
rs-measure.cpp:(.text+0x2374): undefined reference to `glTexParameteri'
rs-measure.cpp:(.text+0x2384): undefined reference to `glTexParameteri'
[... 314 lines cut ...]
rs-measure.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x34): undefined reference to `glfwPollEvents'
rs-measure.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x58): undefined reference to `glfwGetFramebufferSize'
rs-measure.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x60): undefined reference to `glClear'
rs-measure.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x84): undefined reference to `glViewport'
rs-measure.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x88): undefined reference to `glPushMatrix'
rs-measure.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0xac): undefined reference to `glfwGetWindowSize'
rs-measure.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0xe0): undefined reference to `glOrtho'
/tmp/ccUwrpTI.o: In function `toggle::render(window const&)':
rs-measure.cpp:(.text._ZN6toggle6renderERK6window[_ZN6toggle6renderERK6window]+0x28): undefined reference to `glColor4f'
rs-measure.cpp:(.text._ZN6toggle6renderERK6window[_ZN6toggle6renderERK6window]+0x64): undefined reference to `glColor4f'
/tmp/ccUwrpTI.o: In function `toggle::render_circle(window const&, float)':
rs-measure.cpp:(.text._ZN6toggle13render_circleERK6windowf[_ZN6toggle13render_circleERK6windowf]+0x24): undefined reference to `glBegin'
rs-measure.cpp:(.text._ZN6toggle13render_circleERK6windowf[_ZN6toggle13render_circleERK6windowf]+0xe0): undefined reference to `glVertex2f'
rs-measure.cpp:(.text._ZN6toggle13render_circleERK6windowf[_ZN6toggle13render_circleERK6windowf]+0x114): undefined reference to `glVertex2f'
rs-measure.cpp:(.text._ZN6toggle13render_circleERK6windowf[_ZN6toggle13render_circleERK6windowf]+0x128): undefined reference to `glEnd'
/tmp/ccUwrpTI.o: In function `void rs2::frameset::foreach_rs<rs2::frameset::first_or_default(rs2_stream, rs2_format) const::{lambda(rs2::frame)#1}>(rs2::frameset::first_or_default(rs2_stream, rs2_format) const::{lambda(rs2::frame)#1}) const':
rs-measure.cpp:(.text._ZNK3rs28frameset10foreach_rsIZNKS0_16first_or_defaultE10rs2_stream10rs2_formatEUlNS_5frameEE_EEvT_[_ZNK3rs28frameset10foreach_rsIZNKS0_16first_or_defaultE10rs2_stream10rs2_formatEUlNS_5frameEE_EEvT_]+0x6c): undefined reference to `rs2_extract_frame'
/tmp/ccUwrpTI.o: In function `std::enable_if<std::is_base_of<rs2::frame, rs2::frame>::value, bool>::type rs2::frame_queue::poll_for_frame<rs2::frame>(rs2::frame*) const':
rs-measure.cpp:(.text._ZNK3rs211frame_queue14poll_for_frameINS_5frameEEENSt9enable_ifIXsrSt10is_base_ofIS2_T_E5valueEbE4typeEPS5_[_ZNK3rs211frame_queue14poll_for_frameINS_5frameEEENSt9enable_ifIXsrSt10is_base_ofIS2_T_E5valueEbE4typeEPS5_]+0x50): undefined reference to `rs2_poll_for_frame'
/tmp/ccUwrpTI.o: In function `void rs2::processing_block::start<rs2::frame_queue>(rs2::frame_queue)':
rs-measure.cpp:(.text._ZN3rs216processing_block5startINS_11frame_queueEEEvT_[_ZN3rs216processing_block5startINS_11frame_queueEEEvT_]+0x74): undefined reference to `rs2_start_processing'
/tmp/ccUwrpTI.o: In function `std::enable_if<std::is_base_of<rs2::frame, rs2::frameset>::value, bool>::type rs2::frame_queue::poll_for_frame<rs2::frameset>(rs2::frameset*) const':
rs-measure.cpp:(.text._ZNK3rs211frame_queue14poll_for_frameINS_8framesetEEENSt9enable_ifIXsrSt10is_base_ofINS_5frameET_E5valueEbE4typeEPS6_[_ZNK3rs211frame_queue14poll_for_frameINS_8framesetEEENSt9enable_ifIXsrSt10is_base_ofINS_5frameET_E5valueEbE4typeEPS6_]+0x50): undefined reference to `rs2_poll_for_frame'
collect2: error: ld returned 1 exit status
`

MartyG-RealSense, 2020-10-29:
Hi @kylelscott  Could you check whether built versions of the examples are already in the **/usr/local/bin** folder, please?

https://github.com/IntelRealSense/librealsense/issues/1559#issuecomment-382355756

juliahambright, 2020-11-03:
When you compile are you using -lrealsense2 or -glut? You might need more arguments in your command line. I had the same issue and adding -lrealsense2 fixed it.

So my command line to compile looked like this:
g++ -o runfile filetocompile.cpp -lrealsense2

MartyG-RealSense, 2020-11-03:
Thanks very much @juliahambright for your advice in this case!

MartyG-RealSense, 2020-11-10:
Hi @kylelscott   Do you still require assistance with this case, please?   Thanks!

kylelscott, 2020-11-10:
Hello,

I apoligize I haven’t been able to reply. I no longer need assistance with
this issue. Thank you for your help

closed 2020-11-10 by MartyG-RealSense