realsenseai/librealsense #7678: .bag to .ply with rs-convert tool

Makekihe, 2020-10-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | | 
| Operating System & Version |   Win (10)   | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.39.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robotics }                          | 

### Issue Description
I am trying to use the rs-convert tool to convert a .bag file (captured and stored using D435i and Intel RealSense Viewer) to a .ply file for further processing. This is done in the windows command prompt, based on the explanation found on your [rs-convert part of github](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert) and [#2671](https://github.com/IntelRealSense/librealsense/issues/2671).

My command looks like the following, and is executed from the location of the rs-convert.exe folder: 
`C:\Users\kimje\Desktop\rs-convert.exe --input C:\Users\kimje\Desktop\data.bag --output-ply C:\Users\kimje\Desktop\ply`
or (should be the same)
`C:\Users\kimje\Desktop\rs-convert.exe -i C:\Users\kimje\Desktop\data.bag -l C:\Users\kimje\Desktop\ply`

Result seen in command prompt:
[image]

My problem is however that after completion (which is almost instant) no .ply files appear in the selected folder. Am I doing something wrong, or is the fault somewhere else?
If the rs-convert tool does not work, is there any other method for performing this conversion from .bag to .ply?

Notes:
- The .bag file contains 43 frames of depth images that can we viewed from start to finish in the RealSense Viewer without any problems. 
- I've tried to use different variations of folder combinations, different names of files, and on different computers without any luck.
- I've also tried to convert from .bag to .csv (by just replacing the -l with -v), and this works without any problems (longer completion time).

### Final goal
The final goal of this is to use the depth data captured by the Intel Realsense D435i (either live or saved in .bag files) and represented in a point cloud format (.ply or .pcd) for post-processing using the Point Cloud Library in Visual Studios. 
The robotics project in which this is supposed to be a part of will be performed in ROS on Ubuntu.

MartyG-RealSense, 2020-10-30:
Hi @Makekihe  This has been an issue with rs-convert in the past, with some modes such as png generating files and others such as ply or csv not creating, even though the program is visibly converting and reports the number of converted frames at the end of the process 

The case in the link below highlights a number of users who have experienced this issue.

https://github.com/IntelRealSense/librealsense/issues/4171

I ran extensive tests with rs-convert today with two different Windows laptops whilst researching your case and could not generate files from any mode unfortunately, even though the extraction appeared to be processing.  Given the long history of issues with the rs-convert tool, there may not be a solution that can be provided.

An alternative bag extraction program is **rs_bag2image**, a tool created by a RealSense community member, though this may not be useful for your particular project as it extracts bag frames in png and jpg image formats rather than ply or csv

https://github.com/UnaNancyOwen/rs_bag2image

There was a C++ script developed by another community member for loading a bag and converting it to ply.

https://github.com/IntelRealSense/librealsense/issues/1963

Regarding other options, you are likely to find it easier if you save live camera frames to ply instead of bag data.  For example, in the python language there is   For example, the SDK has a Python sample program called **export_ply_example.py**:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

MartyG-RealSense, 2020-11-07:
Hi @Makekihe   Do you require further assistance with this case, please?  Thanks!

Makekihe, 2020-11-10:
No further assistance is required. Thank you for your quick and detailed reply!

We accepted that the tool did not work, and figure out a way to write our own script for solving the issue. 

I guess it could be nice for you to have working tools when they're included as part of the installation.

MartyG-RealSense, 2020-11-10:
Thanks very much for the update!  I do very much understand the frustration that you have experienced.

closed 2020-11-10 by MartyG-RealSense

akhawaja2014, 2021-03-29:
Hi Guys! I am going through the exact same problem. I click on the export option to export .ply data file but it does not appear any where in my directory. Can you please provide that script to go around this problem? Thanking you in advance
[image]

MartyG-RealSense, 2021-03-29:
Hi @akhawaja2014  Can you confirm please if you are left-clicking on the **Export** button and a file selector box pops up but you are unable to find a file with the name that you saved the ply as when you go to the chosen folder location afterwards, please?

[image]

akhawaja2014, 2021-03-29:
I can confirm you that no such box appears after I left click on "export". This is what happens after left clicking export option. It just disapears. Nothing appears in my working directory which is the path I gave for saving .bag recordings.
[image]

MartyG-RealSense, 2021-03-29:
@akhawaja2014  Is there any difference if you pause the stream before left-clicking the Export option?

[image]

akhawaja2014, 2021-03-30:
There is no difference. I tried that option earlier when you mentioned it to the other guy. Unfortunately it does not work sir.

MartyG-RealSense, 2021-03-30:
@akhawaja2014  Let's test bringing up a similar interface overlay in the Viewer to see whether other interfaces have the same problem.  Please left-click on the gear-wheel icon in the top corner of the Viewer, select the **Settings** option from its menu and see whether the Settings window overlay appears.

[image]

[image]

MartyG-RealSense, 2021-03-30:
Thanks very much.  And the ply export interface remains invisible if you press the **F8** key to enter fullscreen mode?  (pressing F8 again toggles the Viewer back to normal view).

akhawaja2014, 2021-03-30:
Yes. With F8 still the problem remains the same. Though it gives wierd depth errors by using F8 and I am not sure where it comes from, Quite unpredictable graphics. like this:
[image]

akhawaja2014, 2021-03-30:
I turn the camera and depth sensor off but still it shows arbitrary graphics. 
[image]

akhawaja2014, 2021-03-30:
and then sometimes using F8 changes the 3D space colour to white without any notification or asking?
[image]

akhawaja2014, 2021-03-30:
Also I found another problem in the interface. When I select the option to "ask me ". The interface does not allow me to to record. I see whatsoever no response by clicking record icon.
[image]

Now I am trying to record bag file but I cannot. I see no response.

[image]

MartyG-RealSense, 2021-03-30:
I note in your original message that you say "Can you please provide that script to go around this problem?"  Does that mean that you are interested in a program script to export a ply using a language such as C++ or Python, or would you prefer a solution in the Viewer?

akhawaja2014, 2021-04-01:
Here is what I want 

- The ability to record RGBD video and run and postprocess it on Python. Its equally good if I can have a python script.
- Initially I used export_ply_example.py as mentioned by you earlier but It is a point cloud for just one frame. I was thinking if in the gui there is an option that can save a rgbd video. The bag file it gives when I record can be just viewed for one frame in python.

MartyG-RealSense, 2021-04-01:
A RealSense user contributed a Python script for recording RGB and depth streams as .avi format video files.  Would that be helpful to you, please?

https://github.com/IntelRealSense/librealsense/issues/8528

akhawaja2014, 2021-04-06:
Thank you sp much, this helped :)

MartyG-RealSense, 2021-04-06:
Great to hear, @akhawaja2014 - I'm pleased that I could be of help.  Thanks for the update!

L-xn, 2022-09-16:
> 不需要进一步的帮助。感谢您快速详细的回复！
> 
> 我们承认该工具不起作用，并想办法编写我们自己的脚本来解决问题。
> 
> 我想，当它们作为安装的一部分包含在内时，拥有工作工具可能会很好。

I have the same problem. May I ask if you have solved this problem? Can you share your Python code.Thanks!