realsenseai/librealsense #7680: Realsense configuration file needs to be uploaded every time in Viewer everytime it gets disconnected

rheaajep, 2020-10-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am using Intel Realsense Model : D415. Currently I am uploading a json file to set certain parameters for the Camera in Realsense Viewer for allowing the camera to see our objects in certain lighting conditions properly. Though each time the camera gets disconnected, we have to upload the JSON file every time in realsense viewer. Is this normal in realsense or is there way around it ?

agrunnet, 2020-10-30:
Yes, this is normal behavior. The camera goes to default values after a power-down. 
Rewriting defaults is currently not supported, but is a good suggestion.

MartyG-RealSense, 2020-10-30:
Thanks @agrunnet for your help with this case!

closed 2020-11-15 by MartyG-RealSense