realsenseai/librealsense #7684: D455 depth image problem

m10507323, 2020-10-31:
Hi, currently I'm trying to get D455 depth image from the realsense viewer or python program,
For the viewer, I use different combination of frame rate and resolution, but all of them got the error bellow
[image]

Also, for my python code, no matter I use which resolution or frame rate, I still can't get the depth image output.
Here is my program
'''python
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
profile = pipeline.start(config)
try:
    while True:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        colorizer = rs.colorizer()
        colorized_depth = np.asanyarray(colorizer.colorize(depth).get_data())
        cv2.imshow('depth_colormap', colorized_depth)
        cv2.waitKey(1)
finally:
    pipeline.stop()
'''

And here is the error message from my python program,
'''python
Traceback (most recent call last):
  File "test.py", line 9, in <module>
    profile = pipeline.start(config)
RuntimeError: Couldn't resolve requests
'''
Any suggestion is appreciated, thanks a lot.

MartyG-RealSense, 2020-10-31:
Hi @m10507323    Regarding your RealSense Viewer problem, does the problem correct itself if you go to the **More** option at the top of the options side-panel of the Viewer and reset the camera by selecting **Hardware Reset** from the drop-down menu?

[image]

Also, may I confirm please that you got the **No frames teceived** message when launching depth after opening the Viewer without toggling the RGB and Motion Module modes first?  A stream  in the RealSense Viewer may sometimes stop displaying data and show this message if depth, RGB and Motion Module have all been activated and then toggled on and off.

Regarding the **Couldn't resolve requests** error in your Python program, this indicates that there is an instruction that librealsense could not understand or it is a streaming mode that is not currently supported by that camera model.  In this case, the use of **1024x768** for the depth stream is likely the cause.  The 400 Series cameras do not support this resolution.  It goes from 848x480 to 1280x720.

[image]

m10507323, 2020-11-01:
Hi @MartyG-RealSense ,
For the viewer part, I have already reset the hardware and only launched depth when I opening the Viewer,
but I still got the same error like previous picture.

And for the "Couldn't resolve requests" problem, after I change resolution to 1280*720 or 848*480,
I got another problem as bellow,

```python
Traceback (most recent call last):
  File "test.py", line 12, in <module>
    frames = pipeline.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
```
I also try all the method you mention in this issue, but the problem still not solved.
Do you have any suggestion for this problem? 
Thanks a lot.

MartyG-RealSense, 2020-11-01:
Do you still get the frame didn't arrive error if you use **poll_for_frames()** instead of wait_for_frames() ?

**frames = pipeline.poll_for_frames()**

m10507323, 2020-11-01:
Hi, I modified my program like bellow,
```python
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
# config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
profile = pipeline.start(config)
frames = pipeline.poll_for_frames()

try:
    while True:
        depth = frames.get_depth_frame()
        colorizer = rs.colorizer()
        colorized_depth = np.asanyarray(colorizer.colorize(depth).get_data())
        cv2.imshow('depth_colormap', colorized_depth)
        cv2.waitKey(1)
finally:
    pipeline.stop()
```
But got another error
```python
  File "test.py", line 16, in <module>
    colorized_depth = np.asanyarray(colorizer.colorize(depth).get_data())
RuntimeError: null pointer passed for argument "frame"
```

MartyG-RealSense, 2020-11-01:
And what happens if you use **try_wait_for_frames()**  - same as wait_for_frames() but does not throw an error and instead returns a 'False' state if the stream times-out.

m10507323, 2020-11-01:
If I change **frames = pipeline.poll_for_frames()** to **frames = pipeline.try_wait_for_frames()**
the error still the same
```python
File "test.py", line 16, in <module>
    colorized_depth = np.asanyarray(colorizer.colorize(depth).get_data())
RuntimeError: null pointer passed for argument "frame"
```

MartyG-RealSense, 2020-11-01:
Python is not my best language in terms of coding knowledge, so I apologize for any frustration.

Let's go back to the Viewer for the moment.  Are you able to get types of stream other than depth, such as infrared, RGB and Motion?

m10507323, 2020-11-01:
No, I've already try RGB and Depth stream, none of them can be worked.

MartyG-RealSense, 2020-11-01:
I was unable to reproduce the error with my own D455 in the Viewer using depth only.  Are you using the latest camera firmware?  That is **5.12.8.200** at the time of writing this.

m10507323, 2020-11-02:
Hi, after I update firmware to 5.12.8.200, it still have problem.
[image]

And here is the device info.
[image]
Thanks.

MartyG-RealSense, 2020-11-02:
Hi @m10507323    I have not been able to replicate your problems with D455 using depth only without activating any other streams.

I note the red '...' notifications in the top corner of this image:

[image]

Next time that you are able to generate these, could you click on the '...' to un-hide their contents so that I can see the errors that they are describing, please?

m10507323, 2020-11-03:
Hi @MartyG-RealSense ,
When I open the viewer, though I don't do anything, it still show the error like this.
[image]

After I start Depth stream, The viewer repeat these error again and again,
[image]
[image]
[image]

Thanks.

MartyG-RealSense, 2020-11-03:
What is the specification of the computer that you are using the camera with, please?   Is it a new or very old computer?

m10507323, 2020-11-04:
I've tried on two computer, one is i7-8750H wtih 1050Ti, another is i7-8700 with 1080Ti, but both of them have the same problem.

MartyG-RealSense, 2020-11-04:
Both of those computers (Intel 8th Generation processors) should be fine.  Thanks for the information.

Could you see if you experience the same problems with SDK 2.38.1 if you have not tried it already, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1

m10507323, 2020-11-05:
I have already tried version 2.38.1, but nothing difference.

MartyG-RealSense, 2020-11-05:
I had a case recently that had some similarities to yours (though the camera was D435i rather than D455).  

https://github.com/IntelRealSense/librealsense/issues/7621

I consulted with Intel about that case and they provided two recommendations for actions to try.  You have already performed one of their recommendations (updating to the latest SDK and firmware).  The other suggestion was to perform a Dynamic Calibration of the camera using the **Dynamic Calibrator** software.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

MartyG-RealSense, 2020-11-05:
In the case that I linked to in the comment above, performing dynamic calibration solved the problem for that RealSense user..

https://github.com/IntelRealSense/librealsense/issues/7621#issuecomment-722646676

m10507323, 2020-11-06:
Hi, I can't not test your method until next Thursday because I need to do other more emergency task.
After trying the method you mentioned, I'll give you some feedback right away, thanks.

MartyG-RealSense, 2020-11-06:
Okay, thanks very much.  I will make a note to keep this case open for the time being.   Good luck!

m10507323, 2020-11-10:
Hello, I just downgraded my firmware to 5.12.7.100 and use Dynamic Calibrator software (v2.11.0) to do the calibration.
After opening the calibrator software, I just click start calibration button and here is the result.
[image]

However, though I do this, it still can't show on the realsense viewer and get the same error message.
I'm confuse that did I use a right method to calibrate the model?
Thanks.

MartyG-RealSense, 2020-11-10:
I went through the Dynamic Calibrator tool with my D455.  After clicking the 'Start Calibration' button, calibration began.  Did your window change to show the screens below, please?

[image]

[image]

If you were able to progress the calibration process this far, then the **targeted calibration** process requires an image with bar-stripes on it to be printed out and the camera pointed to it.  Alternatively, an Android app that shows the bars on its screen can be used instead of a printed-out image.  

Using one of these targets would be necessary to complete the calibration procedure.  In te second image, the on-screen text says "Target is not detected in both left and right images!  Target may be too bright or dark, has reflection, too close or too far".

The printable version of the calibration image can be obtained from the PDF file link below.  The whole of page 1 of the document should be printed out and shown to the camera.

https://dev.intelrealsense.com/docs/dynamic-calibration-print-target

m10507323, 2020-11-11:
Hi, I think there are some troubles with the calibrator software.
I have tried many times, but the software just shut down after I click the start button.

MartyG-RealSense, 2020-11-11:
Are you are using Windows, please?  If you are, can you confirm if you have installed RealSense UWP driver **6.11.160.21**.  This driver has been reported to cause some errors, including a case of the Dynamic Calibrator being disrupted.  This driver does not need to be installed in order to use RealSense with Windows.

[image]

m10507323, 2020-11-11:
Yes, I'm using windows, but I never install the driver you mentioned.

MartyG-RealSense, 2020-11-11:
At this point I will refer the case to Intel for advice.  Thanks so much for your patience!

FrankCreen, 2020-11-11:
> Hi, currently I'm trying to get D455 depth image from the realsense viewer or python program,
> For the viewer, I use different combination of frame rate and resolution, but all of them got the error [...]

Hi ！I test the code below with a 75FPS, could you show your result? Thanks very much!

## Test Program
```
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

#include <opencv2/opencv.hpp>

#include <chrono>

using namespace std;
using namespace cv;

int main()
try
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 480, 270, RS2_FORMAT_BGR8, 90);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

    //Sensor Settings
    auto sensor1 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor1.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    auto sensor2 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor2.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

    // Display in a GUI
    const auto window_name = "Display Video";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    //用于计时
    //show time
    auto start = chrono::system_clock::now();
    int frame = 0;

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        //获取彩色图像帧
        rs2::frame color = data.get_color_frame();

        //保存彩色图像
        Mat image(Size(480, 270), CV_8UC3, (void *)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);

        frame++;
        //one loop end
        auto end = chrono::system_clock::now();
        auto duration = chrono::duration_cast<chrono::microseconds>(end - start);
        double timecost = double(duration.count()) * chrono::microseconds::period::num / chrono::microseconds::period::den;

        cout << frame / timecost << " FPS" << endl;
    }

    return 0;
}
catch (Exception ex)
{
    cout << "Error!!!" << endl;
    cerr << ex.msg << endl;
}
```

MartyG-RealSense, 2020-11-11:
Hi @FrankCreen    Please post your question as a new issue using the **New Issue** button at the link below and do not post it on multiple D455 cases if the question is unrelated to that case.   Thank you.

https://github.com/IntelRealSense/librealsense/issues/

MartyG-RealSense, 2020-11-11:
Hi @m10507323   I received a response from Intel.  They provided me with a test to check whether your case is related to an internal hardware issue with the camera.

Please go to the RealSense Viewer and set up the settings beow:

Stream type: Infrared
Resolution: 1280x800
FPS speed: 25

These settings are used by the camera for calibration.  Are you able to receive an image when using these settings, please?

[image]

[image]

m10507323, 2020-11-12:
Hi, I follow your setting, but it still got the same error with previous method
https://github.com/IntelRealSense/librealsense/issues/7684#issuecomment-720868206

For the infrared, I choose infrared1 and infrared2, but both of them can't work.
[image]

Thanks.

MartyG-RealSense, 2020-11-12:
Hi @m10507323   The advice that I have from Intel is that if you cannot receive an image in 1280x800 / 25 FPS Infrared (which is an unrectified calibration mode) then the camera should be returned for a replacement (RMA).

m10507323, 2020-11-12:
Ok, I'll contact my supplier and try to replace the D455 model.
Thanks for you help!

closed 2020-11-18 by m10507323