realsenseai/librealsense #7690: Depth Pipeline Collision [usb/ethernet]

3dsf, 2020-11-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |  |
|---------------------------------|------------------------------------------- | ------------------------------------------ |
|   | client  |  server  |
| Camera Model                       | d435i  | l515  |
| Firmware Version                   | upadated  |  updated  |
| Operating System & Version |   Fedora32  |  Jetpack 4.4 |
| Kernel Version (Linux Only)    |               5.8.16-200.fc32.x86_64                    |  4.9.140-tegra  |
| Platform                                 | x86_64 | arm64  |
| SDK Version                            |  2.39                         |   2.39  |
| Product                           | realsense-viewer          | rs-server  |
| Segment			|                          |    |

### Depth Frame pipeline collision
Running 435i locally and l515 remotely in realsense-viewer the depth frames collide
[example](https://youtu.be/T6DQX13T800)

If you look in the sidebar, you'll notice it calls the lidar, stereo.

extra detail :  remote camera is networked through a usb connection

RealSenseSupport, 2020-12-23:
@3dsf 

Interesting setup and configuration.

Are you still having issues or were you able to get past them?

*If we don’t hear from you in 7 days, this issue will be closed.

3dsf, 2020-12-27:
@RealSenseSupport 
Honestly, I haven't played with them since.  I was just letting you know that you likely have a hastily copied class; that was the conclusion I came to back then.  I would've looked at fixing it, but my c isn't what it should be.  
If I remember correctly, cuda processing is only implemented for display; are there plans for further implementations? 

Also, I might recommend that you include a script in the library to produce automated output of specifications as seen in the main post comment.  I feel less inclined to make issue posts when I have to complete similar.   _Yes, I understand that may be the point..._

RealSenseSupport, 2021-01-19:
Thanks for the feedback.  We will keep this in mind for the future.

closed 2021-01-19 by RealSenseSupport