realsenseai/librealsense #7691: Android: pipeline restart failure

rabits, 2020-11-01:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D415/D435i/D455) | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Android 11 (RP1A.201005.004) | 
| Kernel Version (Linux Only)    |  4.14.180                                       | 
| Platform                                 | Google Pixel 4 |
| SDK Version                            |  2.39 | 
| Language                            |  C++/Qt 5.14 | 
| Segment			| Smartphone | 

### Issue Description
During implementing dynamic switch for the pipeline configuration I found that there is no way to properly restart the pipeline.

Steps to reproduce:
1. Start the pipeline with some enabled configuration (Depth Z16 1280x720 30fps for example)
   ```
   I librs   : Found 1 RealSense devices (mask 255)
   I librs   : Request: Z16 Depth,
   I librs   : Resolved to: Z16 Depth,
   I librs   : endpoint 130 read buffer size: 1844224
   I librs   : USB pipe 130 reset successfully
   ```
2. Data is streaming well
3. pipeline->stop()
   ```
   I librs   : USB pipe 130 reset successfully
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   ```
4. pipeline->start(same_config) - the complete log, it stops to log after the last line:
   ```
   I librs   : Request: Z16 Depth,
   I librs   : Resolved to: Z16 Depth,
   I librs   : endpoint 130 read buffer size: 1844224
   I librs   : USB pipe 130 reset successfully
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   W librs   : control_transfer returned error, index: 768, error: Broken pipe, number: 32
   ```
5. No data transferred anymore

I tried to debug and figured out just that it uses USBHOST backend and closing/switch power for the uvc_device sensors. So I don't think Android supports some required functionality and can't restore the normal device functioning.

Did not found any useful info in logcat, so that's what I have for now.

MartyG-RealSense, 2020-11-02:
Hi @rabits   **control_transfer returned** errors that generate continuously can suggest that there is a serious communication problem with the device.  

May I confirm that the problem that you are experiencing is that Android streams successfully on the first start, but cannot resume when starting again after a stop?

rabits, 2020-11-03:
Hi @MartyG-RealSense , yeah - I'm rewriting my logic for Handy 3D Scanner to use the better way of choosing the streams to capture and yeah - it works good when it's started the first time (any number of streams), but if I want to change the configuration - there is no way to restart the pipeline with the new config, it just fails to work.

I'm using target SDK 29 (because it's a minimal req for PlayStore) and I know about the issue with USB on Android 10 (Handy3DScanner/issues#68) - and for my devices (pixel4/galaxyS10) it's fixed, no worries. I rechecked how it's working on Android 10/11 and Android 7 (Galaxy S8) - unfortunately quite the same story. Tried go back to 27 target SDK version, but no luck there... Also tried realsense 2.38.1 - and no luck there either - the same behavior.

So I think those tests shows it's something wrong with the pipeline restart process, but not with the Android platform (7,10,11).

MartyG-RealSense, 2020-11-03:
I did further research and found an Android case that had similar problems with resetting.  The RealSense user in that case found that they could not successfully reset their Android project with the librealsense API or by resetting the USB controller and the only method that worked for them was a complete system reboot.

https://github.com/IntelRealSense/librealsense/issues/4091

Another Android case had problems with start after stop:

https://community.intel.com/t5/Intel-NUCs/Fail-to-restart-stream-with-D410-RealSense-SDK-for-Android-on/td-p/481755

rabits, 2020-11-03:
Yeah, I heard about such issues - but in my case it's like the thread working on the getting the fresh frames from the pipeline just waiting forever after restart the pipeline. Previously I checked only for one run and disconnected the camera before run another try, but now I checked without disconnecting the camera and yep - if I run the app again and going through the steps - even the first one is not working:
```
I librs   : Found 1 RealSense devices (mask 255)
I librs   : Request: Z16 Depth,
I librs   : Resolved to: Z16 Depth,
W librs   : control_transfer returned error, index: 1, error: Broken pipe, number: 32
I chatty  : uid=10212(io.stateoftheart.handy3dscanner) qtMainLoopThrea identical 4 lines
W librs   : control_transfer returned error, index: 1, error: Broken pipe, number: 32
E librs   : Probe-commit control transfer failed with error: RS2_USB_STATUS_OTHER
W librs   : null pointer passed for argument "profile"
W Handy 3D Scanner: RealSensePlugin::RSDevice: Unable to start pipe with the current configuration:
W Handy 3D Scanner: Failed to resolve the request:
W Handy 3D Scanner: 	Format: Z16, width: 1280, height: 720
W Handy 3D Scanner: Into:
W Handy 3D Scanner: 	Formats:
W Handy 3D Scanner: 	 Z16
```
So the streaming becomes unusable until I will reconnect the camera via USB. Although it still shows the list of available streams and general info from the camera and that tells that the usb connection is not entirely broken, just the uvc streaming part of it.

Also tested with USB2.0 cable - the same behavior as for USB3.

Before I switched to librs 2.39 I used 2.20 (version prior to the android backend changes) and it acted the same way (stuck after stop/start), just less verbose.

So my current thoughts are about the stop process, that actually closes (or doing it wrong) the entire device descriptor - it's just a speculation, but that's where I stopped to dig because it becomes too complex very quickly for an app dev like me)

MartyG-RealSense, 2020-11-04:
I recall some cases that used the approach of starting with pipe.start() but performing the closing by closing the **sensor** instead of the pipeline.

https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

rabits, 2020-11-08:
So I figured out the issue - it was my logic.

First of all I dug into RS Camera and found that it's destroying the pipeline to restart it, so implemented the same way and fixed a couple of my app issues with streaming (https://github.com/state-of-the-art/Handy3DScanner/pull/75/commits/d9f60bb965dd341b4c2ccb8f6ba8709c65112658). And after that I found that `control_transfer returned error` issues are quite common and the streams are working even there is such errors.

So I moved back to not-destroying pipeline model and it's working well (https://github.com/state-of-the-art/Handy3DScanner/pull/75/commits/c8d6150205294e6f6b2b77135d522929d5788d7d).

Those errors are quite misleading, but overall yep - the issue could be closed, it was basically on my side.

MartyG-RealSense, 2020-11-08:
Great news that you found a solution @rabits - thanks so much for sharing your learnings about this subject with the RealSense community!    :)

closed 2020-11-08 by MartyG-RealSense

rabits, 2020-11-09:
Yeah, at least on 2.38.1 and android 11.