realsenseai/librealsense #7702: No device connected but lsusb can find device

MccAlfh, 2020-11-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |{ D435i }| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    {Linux (Ubuntu 16)}   | 
| Kernel Version (Linux Only)    |   4.4.58-20191018.kylin.server-generic )    | 
| Platform                                 |  FT1500 kylin.server-generic aarch64  |
| SDK Version                            |  {2.18， 2.34， 2.36}                          | 
| Language                            |  {C }                          | 
| Segment			|  {Robot }                          | 

I run the rs tools after finishing install the SDK，but it shows no device connect info as the following pics：
however it can be found from "lsusb"
[image]

This is first time i tried run D435i at a aarch64 platform，I remember i met the same problem at X86 platform and solved by the following approach：

----------
dorodnic commented on 16 Oct 2019 •
Hi @studennis911988
Sorry for all the troubles. Fortunately, not all is lost :)

What happened is firmware 5.10.6 has no idea D435i is a thing, since it was released prior to this product existing. This is why since the first release mentioning D435i we require minimum firmware version of 5.10.10.0.

Here's what you need to do:

    Git clone librealsense source code, I'll assume it's under ~/librealsense.
    Navigate to ~/librealsense/src/ds5/ds5-private.h
    Replace line 22 - const uint16_t RS430_PID = 0x0ad4; // AWG with const uint16_t RS430_PID = 0x0b01; // AWG
    Replace line 32 - const uint16_t RS430_MM_RGB_PID = 0x0b01; // AWGCT with const uint16_t RS430_MM_RGB_PID = 0x0ad4; // AWGCT
    Navigate to ~/librealsense/src/ds5/ds5-factory.cpp
    At line 851 (after an if statement and prior to ifdef APPLE) paste all_sensors_present = true;
    cd ~/librealsense && mkdir build && cd build && cmake .. && make
    Run newly compiled RealSense-Viewer (./tools/realsense-viewer/realsense-viewer) and update to recommended firmware. You can also use the custom ./tools/fw-update/rs-fw-update command line tool.
    Don't forget to undo changes to the SDK for future use

----------
But it seems not work at aarch64 Platform. 
and i've tried SDK[ 2.18, 2.34, 2.36] but still make no progress， how can i solve it?

MartyG-RealSense, 2020-11-03:
Hi @MccAlfh   It may be worth trying to install librealsense with the **RSUSB** method.  This method requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.   Instructions for the RSUSB method can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

MccAlfh, 2020-11-03:
> 
> 
> Hi @MccAlfh It may be worth trying to install librealsense from source code with the **RSUSB** method. This method requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching. Instructions for the RSUSB method can be found in the link below.
> 
> [#6368 (comment)](https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279)

It works， thank you！

closed 2020-11-03 by MccAlfh

MartyG-RealSense, 2020-11-03:
Great news @MccAlfh  - thanks very much for the update!