realsenseai/librealsense #7703: L515: Use Realsense Viewer to open the depth and color sensors . The viewer fps is unstable

Jaesirky, 2020-11-03:
Hello, we are conducting a collection project to collect customer purchase behavior in convenience store scenes through 9 L515s. During the trial collection process, we found that the captured video frame rate was faster than the actual one. Through Realsense Viewer, we found that Viewer Fps is not Stable. As shown below

After that, we solved the problem of the color sensor by adjusting the exposure rate of the color sensor, but the frame rate is still unstable when the depth and color sensors are turned on at the same time.

This is our environment:
Operating System: Win 10
host cpu: I7 9700
host disk: solid state drive
Relasense Viewer Version: 2.38.1
SDK Version: 2.38.1
Camera Model: L515
Firmware Version: 01.05.00.00

We now hope that the frame rate during acquisition can be stably maintained at around 30fps, and we hope to get some suggestions.

Jaesirky, 2020-11-03:
[image]

jagatfx, 2020-11-06:
@Jaesirky I had the same issue. I believe the issue is the Auto Exposure Priority setting for the RGB camera. From my understanding this overrides the hardware sync capabilities. When I disable this setting I am able to get stable framerates across sensors.

[image]

Jaesirky, 2020-11-18:
We used to record in the basement. When we moved to the first floor, we added lighting and adjusted the resolution and frame rate. Now the frame rate is stable.
Our configuration:
Depth: 720p 15Fps
Color: 1080p 15Fps auto exposure

closed 2020-12-23 by RealSenseSupport