realsenseai/librealsense #7708: Best practice for Windows build PR?

JTrantow, 2020-11-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { any} | 
| Firmware Version                   | (latest) | 
| Operating System & Version |   {Win10| 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  development
| Language                            |  {C/C#}                          | 
| Segment			|  others                           | 

### Issue Description
What is the best practice for testing Windows builds before a pull request?

I am building applications in Win10 and would like to share some code cleanup/compiler warning improvements.
I followed https://github.com/IntelRealSense/librealsense/blob/master/CONTRIBUTING.md. 
I can run the pr_check.sh and api_check.sh using the Ubuntu WSL. I changed the default CMake options to disable shared libs and to build unit_tests. 

With the development branch (none of my own changes), I tried running the unit tests on my PC with a D415 attached. When I run live_test, I get the attached output results and the test appears to hang. This may be consistent with my Travis CI failures after no response for 10 minutes? 

[unit_test_output.txt](https://github.com/IntelRealSense/librealsense/files/5483312/unit_test_output.txt)

What are the requirements to run live_test? Is it camera specific? 
**Should I worry about Travis CI failures that fail Windows testing but pass everything else?**

[image]

dorodnic, 2020-11-04:
Hi @JTrantow 
Thank you for the pull-request. There is some instability in Travis builds, partly because of Travis (when you see the exclamation mark) and partly because of issues in our unit-tests. These should be rare, and whenever one of the admins sees it, we immediately re-trigger the build

JTrantow, 2020-11-04:
Thanks.

I'm trying to shorten the development process and Travis CI takes a long
time to run.
Is it worth while to run live-test locally on my PC before running the
Travis CI (on my fork)?  I can just run live-test from the debugger but it
seems like the CI runs live-test several times with different options???
How can I do this locally?

Are the unit-tests designed to be high level multi-functional system tests
or are there unit tests at the individual function level?

RealSenseCustomerSupport, 2020-11-06:
----------------------------------------------

Hi JTrantow
The full test cases should included both individual function test  such as Advance mode test
and high level multi-functional system tests such as recording or sync test...
below I capture some portion of all test cases that use -l option to list all test cases..
there should more than 100 test cases available in different tag (name in bracket)
to run specific test case using name or tags, you can refer to below link

https://github.com/catchorg/Catch2/blob/devel/docs/command-line.md#specifying-which-tests-to-run

There are chance that the same test may sometimes pass some times fail ..
one way to see if it able to pass is to test that specific test case again..

====test Sync related test =====
C:\librealsense-development\build\Debug>live-test.exe Sync* -d yes
Running tests with the following parameters: live-test.exe Sync* -d yes
Filters: Sync*
12.037 s: Sync sanity
6.737 s: Sync different fps
8.080 s: Sync start stop
0.767 s: Syncer sanity with software-device device
0.435 s: Syncer clean_inactive_streams by frame number with software-device device
0.429 s: Syncer try wait for frames
===============================================================================
All tests passed (1416 assertions in 6 test cases)

=======use - l option to show ALL TEST CASES===
C:\librealsense-development\build\Debug>live-test.exe -l
Running tests with the following parameters: live-test.exe -l
All available test cases:
  Sync sanity
      [!mayfail][live]
  Sync different fps
      [!mayfail][live]
  Sync start stop
      [!mayfail][live]
  Device metadata enumerates correctly
      [live]
  Start-Stop stream sequence
      [live][using_pipeline]
  Start-Stop streaming  - Sensors callbacks API
      [live][using_callbacks]
  No extrinsic transformation between a stream and itself
      [live]
  Extrinsic transformation between two streams is a rigid transform
      [live]
  Extrinsic transformations are transitive
      [live]
  Toggle Advanced Mode
      [AdvMd][live]
  Advanced Mode presets
      [AdvMd][live]
  Advanced Mode JSON
   .......

JTrantow, 2020-11-06:
Thanks. I just starting building the unit_tests a few days ago. I will look
into the tests and how to run them under Win10.

RealSenseCustomerSupport, 2020-11-23:
----------------------------------------------

Dear @Jtrantow,
is there any good news that we could close this ticket ?
best

closed 2020-12-01 by RealSenseCustomerSupport