realsenseai/librealsense #7709: Cubemos integration into Unreal Engine 4.25 dll issue

elliotstronge, 2020-11-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC Unreal Engine 4.25 |
| SDK Version                            | Cubemos 3.0                        | 
| Language                            |  Unreal Engine 4.25 C++                  | 
| Segment			|  Skeleton Tracking                      | 

I'm having a lot of trouble getting the Cubemos SDK integrated into Unreal Engine. I've included all the source and binaries, but whenever I start Unreal Engine from Visual Studio 2019 it crashes with the error: 

`The procedure entry point SymEnumSymbolsExW could not be found in the dll C:\Windows\SYSTEM32\dbgeng.dll`

I believe that's just related to the Unreal Engine crash viewer. Looking through the logs I see the errors:

`LogWindows:   Missing import: cubemos_engine.dll`
`LogWindows:   Missing import: cubemos_base.dll`
`LogWindows:   Missing import: cubemos_skeleton_tracking.dll`

even though UE is looking in exactly the right place to find the dlls. I'm sure I have all the right settings, as in running 64 bit etc but it's just not recognising them. The sample programs included with the Cubemos SDK also work without any issues.

MartyG-RealSense, 2020-11-04:
Hi @elliotstronge   The Cubemos Skeleton Tracking SDK is the product of a company called **Cubemos**, and they are responsible for providing support for it as the skeleton SDK is not an Intel product.

To talk to Cubemos, please visit their website at https://www.cubemos.com/ and click on the **Got any questions?  I'm happy to help** button in the bottom corner of the page.

elliotstronge, 2020-11-04:
Ok thanks Marty.

closed 2020-11-04 by elliotstronge