realsenseai/librealsense #7711: Error in executing ./patch-arch.sh

ganglesmunne, 2020-11-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | {librealsense} | 
| Operating System & Version |   {Archlinux}  | 
| Kernel Version (Linux Only)    |  {5.9.3)                                         |                 

Hello,

I am trying to install librealsense in my archlinux, but when I executed ./patch-arch.sh,  the following error appears when I am patching:

- patch -p1 < ../../realsense-camera-formats.patch

patching file drivers/media/usb/uvc/Makefile
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/Makefile.rej
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 214 with fuzz 2 (offset 66 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 129 with fuzz 2 (offset 10 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 621.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej

The kernel version of my archlinux is **Linux 5.9.3-arch1-1** and the downloaded kernel version in the ./patch-arch.sh  for patching is **Linux 5.9.3**. I read that the problem could be a different kernel version between my version of linux and the patching kernel but in my case that is not true so I do not know what to do....

I will appreciate a lot your help!

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2020-12-17:
Hi @ganglesmunne 

Are you still running into this issue with the latest librealsense SDK 2.40?

ganglesmunne, 2020-12-18:
Hey! I had the same problem, but I solved adding in the lines where there are _patch -p1_  ----> _patch -p1 **--fuzz 3**_ .
I also had more problems which I solved following the instructions of this link:
https://github.com/IntelRealSense/librealsense/issues/4271

In the same time, I needed to run the patch everytime when I reboot the computer. However, surprisely when the kernel of Arch is uptate, then the patch becomes permenet.

RealSenseSupport, 2020-12-21:
Excellent you were able to solve the problem.

closed 2020-12-21 by RealSenseSupport