realsenseai/librealsense #7712: How long can the D435i last before being replaced?

maky-hnou, 2020-11-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (2.34.0) | 
| Operating System & Version |   {Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC/ NVIDIA Jetson Nano |
| SDK Version                            |  { 2.34.0 }                          | 
| Language                            |  {C++/Python }                          | 
| Segment			|  {others }                          | 

### Issue Description
I am working on a project using Intel Realsesne D435i and we want to do some endurance and aging tests to check how long can the D435i withstand/last.  
Did anyone of the developing team perform those tests?  
We tested the D435i for 15 hours, and it kept working without crashing. 
So we want to know, if anyone has the information, for how many hours can the D435i work before being totally out of work (or useless) and should be replaced?  
Thank you.

MartyG-RealSense, 2020-11-04:
Hi @maky-hnou   The 400 Series cameras can run indefinitely so long as they remain within recommended operating temperature tolerances.  If a problem does occur in a long-run session, it is likely to be related to the computer hardware, its operating system or the USB hardware (port / hub / controller / cable) rather than a problem with the camera hardware.

In regard to how long the camera can stream for in the same long-run session during real-world conditions: as stated above, the camera can run indefinitely as long as it remains within temperature tolerances and there are no other disruptive factors.  In real-world conditions, 4 days is the longest that I have heard of a RealSense application running for before a glitch occurs somewhere in the computer system.

In the link below, @dorodnic the RealSense SDK Manager comments on laser life expectancy over time.

https://github.com/IntelRealSense/librealsense/issues/3452#issuecomment-471492320

maky-hnou, 2020-11-04:
Hello @MartyG-RealSense , thank you for your reply.  
So as I understand, there is no **fixed** life expectancy period (i.e 1 year, 2 years, 5000 hours...) for the D400 Series. It all depends on the environment conditions (temperature, pressure, ...) and the the hardware connected to the camera (USB cable, USB port, PC) and the operating system. Am I right?  
Thank you again for your time.

MartyG-RealSense, 2020-11-04:
As ar as I know there are no fixed life expectancy estimates for the camera hardware.  

In regard to environment: in addition to internal operating temperature, external weather temperature could heat the camera's outer casing and have an influence on overall operating temperature.  Lab tests performed by Intel found that the operating temperature should not exceed 60 degrees C so long as the casing temperature remains below 44 degrees C.

The camera firmware driver should reduce laser power by half if the projector temperature exceeds 60 degrees for several seconds, and then shut the laser off if the temperature continues to be excessive.

https://github.com/IntelRealSense/librealsense/issues/5209#issuecomment-563358823

Safety certification details for the RealAense product range can be found here:

https://www.intelrealsense.com/regulatory-information/

maky-hnou, 2020-11-04:
Thank you so much.
I will close the issue as I got answers to my question.

closed 2020-11-04 by maky-hnou