realsenseai/librealsense #7717: Bug report in roi_sensor.m

ghxguo, 2020-11-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515                                                       | 
| Firmware Version                   |                                                               | 
| Operating System & Version |   win10                                                   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                         |
| SDK Version                            | 2                                                          | 
| Language                            |  Matlab                          | 
| Segment			| Others                          | 

in roi_sensorlm, line 31
it was:
            input = struct('min_x', int64(roi.min_x), 'min_y', int64(roi.min_y), 'max_x', int64(max_x), 'max_y', int64(max_y));
it should be:
            input = struct('min_x', int64(roi.min_x), 'min_y', int64(roi.min_y), 'max_x', int64(roi.max_x), 'max_y', int64(roi.max_y));

RealSenseSupport, 2020-12-17:
Hi @ghxguo 

Are you still running into this issue?  Is your configuration requiring the use of Matlab or can you use the basic librealsense library outside of Matlab wrapper?

RealSenseSupport, 2021-01-14:
Hi @ghxguo

Are you still running into this issue? Is your configuration requiring the use of Matlab or can you use the basic librealsense library outside of Matlab wrapper?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-21 by RealSenseSupport