realsenseai/librealsense #7719: 3d point cloud is only visible very close to camera

readblack1234, 2020-11-05:
|---|---|
|**librealsense**|2.35.2 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense SR305|
|**Serial Number**|636204001701|
|**Firmware Version**|3.27.3.0|
|**Camera Locked**|YES|
|**Product Line**|SR300|
|**Asic Serial Number**|636204001701|
|**Firmware Update Id**|636204001701|

Please provide a description of the problem

3d point cloud is only visible very close to camera.
-Object is completely compressed in a cone (3d cone), the distance it is measuring is like 100-200 mm only, beyond that if I place an object, it does not show that. 
same realsenseviewer on windows with same camera, shows correct image at distances normally 0.1 - 1 meter or more.
-I suspect issue is that image cone is generated and is distorting 3d point cloud. I am not sure if this is issue of firmware version, if so how can I update that?

MartyG-RealSense, 2020-11-05:
Hi @readblack1234   It sounds as though you are having problems with viewing the point cloud in Linux with the RalSense Viewer, but do not have that problem when using the Viewer in Windows.  Is that correct, please?

As firmware is stored inside the camera hardware and not on the computer (and therefore gets carried with the camera between machines), I would expect problems to also manifest on Windows if the issue was related to the firmware.

readblack1234, 2020-11-06:
This issue was reported earlier a well, I used this to resolve that
https://support.intelrealsense.com/hc/en-us/community/posts/360044304633-Is-depth-broken-in-my-SR305
https://github.com/IntelRealSense/librealsense/issues/6373#issuecomment-630584604
Basically, CUDA had to be turned off.

Now, I am having issue with building  specific "rs-pcl" binary from examples.
I am getting following error
"
~/librealsense/wrappers/pcl/build$ make
Scanning dependencies of target rs-pcl
[ 25%] Building CXX object pcl/CMakeFiles/rs-pcl.dir/rs-pcl.cpp.o
[ 50%] Linking CXX executable rs-pcl
/usr/bin/ld: /usr/local/lib/libglfw3.a(posix_thread.c.o): undefined reference to symbol 'pthread_getspecific@@GLIBC_2.17'
/lib/aarch64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
pcl/CMakeFiles/rs-pcl.dir/build.make:371: recipe for target 'pcl/rs-pcl' failed
make[2]: *** [pcl/rs-pcl] Error 1
CMakeFiles/Makefile2:85: recipe for target 'pcl/CMakeFiles/rs-pcl.dir/all' failed
make[1]: *** [pcl/CMakeFiles/rs-pcl.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2""

Is  there a fix for that?

readblack1234, 2020-11-06:
Also, attaching the console output when realsense sdk was installed. It shows libglfw3.a installed.
[231_installed.txt](https://github.com/IntelRealSense/librealsense/files/5503535/231_installed.txt)

MartyG-RealSense, 2020-11-06:
Reading the discussion in the link below may give you some ideas.

https://github.com/IntelRealSense/librealsense/issues/6784

MartyG-RealSense, 2020-11-15:
Hi @readblack1234   Do you still require assistance with this case, please?   Thanks!

readblack1234, 2020-11-15:
Thanks @MartyG-RealSense , I was able to resolve this by using CMake example in examples. That was very helpful.

MartyG-RealSense, 2020-11-16:
That is great to hear - thanks very much for the update !

xavirostudio, 2020-11-17:
I've commented on a closed issue #6373 - I believe CUDA can be made to work it's a matter of fixing how bit sift operations are done on GPU vs CPU.

MartyG-RealSense, 2020-11-17:
Hi @xavirostudio   I have now re-opened that case for another look.  Thanks for your report!

MartyG-RealSense, 2020-11-26:
Adding a note to keep this case open for a further time period.

MartyG-RealSense, 2020-11-26:
Hi @readblack1234   As you had a successdul outcome, I will close this case now.   Thanks!

@xavirostudio   As your CUDA question is on another case, I can detail with that there.  Thanks again for your report.

closed 2020-11-26 by MartyG-RealSense