realsenseai/librealsense #7720: Fail to install IntelRealSense_D400_series_win7.inf driver

MojamojaK, 2020-11-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC |
| SDK Version                            | 2.30.1 | 

### Issue Description

Hello,
I have been trying to install the win7 drivers for D400 cameras.
It used to work fine in the past (a few months ago), but now every time when I either use the provided installer or the inf file to install the driver, an error saying **"A problem was encountered while attempting to add the driver to the store."** would pop up.

When I use the inf file to install the driver, the following files are placed in the same directory
- IntelRealSense_D400_series_win7.inf
- IntelRealSense_D400_series_win7.cat
- WdfCoinstaller01011.dll
- winusbcoinstaller2.dll

I have also tried disabling signed driver enforcement and it would say that the driver was successfully installed, but the OS would still recognize the devices with the default Windows 10 driver

I hope to at least learn a workaround for this.

Here is the detupapi.dev.log file after failing to install the driver
```
[Device Install Log]
     OS Version = 10.0.18363
     Service Pack = 0.0
     Suite = 0x0100
     ProductType = 1
     Architecture = amd64

[BeginLog]

[Boot Session: 2020/11/05 12:22:06.500]
     Hardware Configuration: {2959a1a8-f65f-0000-0000-000000000000}

>>>  [Device Install (DiInstallDriver) - C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf]
>>>  Section start 2020/11/05 12:34:09.583
      cmd: "C:\Windows\System32\InfDefaultInstall.exe" "C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf"
     ndv: Flags: 0x00000000
     ndv: INF path: C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf
     inf: {SetupCopyOEMInf: C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf} 12:34:09.585
     inf:      Copy style: 0x00000000
     sto:      {Setup Import Driver Package: C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf} 12:34:09.587
     inf:           Provider: Intel(R) Corporation
     inf:           Class GUID: {88BAE032-5A81-49f0-BC3D-A4FF138216D6}
     inf:           Driver Version: 10/31/2019,2.30.00.0
     inf:           Catalog File: IntelRealSense_D400_series_win7.cat
     sto:           {Copy Driver Package: C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf} 12:34:09.591
     sto:                Driver Package = C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf
     sto:                Flags          = 0x00000007
     sto:                Destination    = C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}
     sto:                Copying driver package files to 'C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}'.
!    inf:                Section [uvc_install] is defined multiple times. Code = 1209, Line = 127
!    inf:                Section [winusb_install] is defined multiple times. Code = 1209, Line = 180
     flq:                Copying 'C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\WdfCoInstaller01011.dll' to 'C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}\WdfCoInstaller01011.dll'.
     flq:                Copying 'C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\WinUsbCoInstaller2.dll' to 'C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}\WinUsbCoInstaller2.dll'.
     flq:                Copying 'C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.cat' to 'C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}\IntelRealSense_D400_series_win7.cat'.
     flq:                Copying 'C:\Program Files (x86)\Intel RealSense SDK 2.0 (Win7)\drivers\IntelRealSense_D400_series_win7.inf' to 'C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}\IntelRealSense_D400_series_win7.inf'.
     sto:           {Copy Driver Package: exit(0x00000000)} 12:34:09.609
     pol:           {Driver package policy check} 12:34:09.614
     pol:           {Driver package policy check - exit(0x00000000)} 12:34:09.614
     sto:           {Stage Driver Package: C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}\IntelRealSense_D400_series_win7.inf} 12:34:09.615
!    inf:                Section [uvc_install] is defined multiple times. Code = 1209, Line = 127
!    inf:                Section [winusb_install] is defined multiple times. Code = 1209, Line = 180
     inf:                {Query Configurability: C:\Users\<User>\AppData\Local\Temp\{dc3771bf-79d0-2447-a381-c030c0691f44}\IntelRealSense_D400_series_win7.inf} 12:34:09.627
[... 15 lines cut ...]
!    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 12:34:09.710
     sig:                     {_VERIFY_FILE_SIGNATURE} 12:34:09.712
     sig:                          Key      = IntelRealSense_D400_series_win7.inf
     sig:                          FilePath = C:\Windows\System32\DriverStore\Temp\{74dcebd5-7323-0a47-b940-1b0042fa5130}\IntelRealSense_D400_series_win7.inf
     sig:                          Catalog  = C:\Windows\System32\DriverStore\Temp\{74dcebd5-7323-0a47-b940-1b0042fa5130}\IntelRealSense_D400_series_win7.cat
     sig:                          Success: File is signed in Authenticode(tm) catalog.
     sig:                          Error 0x800b0100: No signature was present in the subject.
!    sig:                          Verifying file against catalog 'IntelRealSense_D400_series_win7.cat' using Authenticode(tm) policy failed.
!    sig:                          Error 0x800b0100: No signature was present in the subject.
     sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 12:34:09.724
!!!  sig:                     Driver package catalog file does not contain a signature, and Code Integrity is enforced.
!!!  sig:                     Driver package failed signature validation. Error = 0xE0000247
     sto:                {DRIVERSTORE IMPORT VALIDATE: exit(0xe0000247)} 12:34:09.726
!!!  sig:                Driver package failed signature verification. Error = 0xE0000247
!!!  sto:                Failed to import driver package into Driver Store. Error = 0xE0000247
     sto:           {Stage Driver Package: exit(0xe0000247)} 12:34:09.728
     sto:      {Setup Import Driver Package - exit (0xe0000247)} 12:34:09.731
!!!  inf:      Failed to import driver package into driver store
!!!  inf:      Error 0xe0000247: A problem was encountered while attempting to add the driver to the store.
     inf: {SetupCopyOEMInf exit (0xe0000247)} 12:34:09.756
<<<  Section end 2020/11/05 12:34:09.757
<<<  [Exit status: FAILURE(0xe0000247)]

```

MojamojaK, 2020-11-05:
I have confirmed that this has been occurring on multiple Windows 10 PCs

MartyG-RealSense, 2020-11-05:
Hi @MojamojaK   I am not sure why you are trying to install a Windows 7 driver on Windows 10 PCs.  Could you provide more details about what you are trying to achieve, please?  Thanks very much.

MojamojaK, 2020-11-06:
Hi,
So I've been working on a project for around a year revolving the use of a lot (20+) realsense D400 series cameras on one Windows 10 system. Using this much devices on one single system really strains the it and would cause the devices to disconnect once in a while. 
After doing some extensive testing last year, I came to the conclusion that using the Windows 7 driver with RSUSB backend enabled, instead of the default Windows 10 driver would drastically improve USB connection stability at least on my environment.

My project works as smooth as butter on systems with the Windows 7 driver already installed, but I currently seem to not be able to install the driver properly on newly created Windows systems (with the same OS version) for some reason.

MojamojaK, 2020-11-06:
I have done some extra research and found the culprit (and a temporary workaround).
It seems like Windows Update (KB4577671) for Windows 10 1903 / 1909 has been messing around with how the Win7 driver security catalog should be signed.

Driver installation succeeds without any problem when this patch is uninstalled.

Here is the changelog: 
https://support.microsoft.com/en-us/help/4577671/windows-10-update-kb4577671

See
Symptom
> When installing a third-party driver, you might receive the error, “Windows can’t verify the publisher of this driver software”. You might also see the error, “No signature was present in the subject” when attempting to view the signature properties using Windows Explorer.

Workaround
> This issue occurs when one or more of the following is present in a driver package:
> An improperly formatted catalog file is identified during validation by Windows. Starting with this release, Windows will require the validity of DER encoded PKCS#7 content in catalog files. Catalogs files must be signed per section 11.6 of describing DER-encoding for SET OF members in X.690.
> A driver catalog file extension is not one of supported extensions.
> **If this happens you should contact the driver vendor or device manufacturer (OEM) and ask them for an updated driver to correct the issue.**

I hope this issue would be fixed since this would mean it would become impossible to install the Win7 driver onto a Windows 10 system in the future due to lack of Microsoft support.

MartyG-RealSense, 2020-11-06:
Thanks very much for the update about your workaround @MojamojaK

MartyG-RealSense, 2020-11-13:
Hi @MojamojaK   Do you still require assistance with this case, please?   Thanks!

MojamojaK, 2020-11-13:
I think we should leave this open unless your team don't intend to fix this issue permanently.
The only way this problem would be permanently fixed is for your team to properly format the driver installer's catalog file to what Microsoft specifies as written in the patch changelog.
> This issue occurs when one or more of the following is present in a driver package:
> An improperly formatted catalog file is identified during validation by Windows. Starting with this release, [...]

MartyG-RealSense, 2020-11-13:
I will highlight this driver signing issue to Intel.  Thanks very much.

MartyG-RealSense, 2020-11-19:
Hi @MojamojaK   Intel has responded that it is not recommended to use a Windows 7 driver on Windows 10, and that using of 20+ cameras on one PC is not validated.  So you are of course free to do so, but may need some subtle tweaks.  It is also suggested that you can use PCIe cards to expand USB bandwidth.  This might improve the system stability.

MartyG-RealSense, 2020-11-26:
Hi @MojamojaK   Do you require further assistance with this case, please?  Thanks!

MojamojaK, 2020-11-26:
This issues is obviously not resolved.
I mean I understand that intel hasn't validated the use of using 20+ cameras. I get it.
I am in fact using a lot of PCIe cards to improve stability.

This issue isn't about stability. Just the ability to install the driver to newer WIn10 systems.
I just want to know whether or not intel will consider fixing this issue or not. (I do hope they do)

MojamojaK, 2020-11-26:
I just checked with the latest release(2.40.0).

Seems like Intel has resolved the issue (Without stating in the Release notes).

closed 2020-11-26 by MojamojaK

MartyG-RealSense, 2020-11-26:
Great to hear that you had success with 2.40.0 - thanks for the update!