realsenseai/librealsense #7722: How to install librealsense and pyrealsense2 on the Jetson NX

lieuzhenghong, 2020-11-05:
There seems to be quite a lot of interest (and difficulty) in installing pyrealsense2 on the Jetson:
see issues #6964, #6820, #6980  and others.

I think this is because the documentation is not as clear as it could be.

Firstly, it should be clearly signposted that we need to build `librealsense` from source because the PyPi pip packages are not compatible with Arm processors. 

Secondly, one needs to refer to both READMEs and piece them together to get a full understanding of the install process:

[README1 in /blob/master/doc](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)
[README2 inside wrappers/python](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source)

Neither of the READMEs give a full set of instructions to follow; each of them is missing some piece.

I have come up with a list of instructions that works on my Jetson NX.

```bash
# Installs librealsense and pyrealsense2 on the Jetson NX running Ubuntu 18.04
# and using Python 3
# Tested on a Jetson NX running Ubuntu 18.04 and Python 3.6.9 on 2020-11-04

sudo apt-get update && sudo apt-get -y upgrade
sudo apt-get install -y --no-install-recommends \
    python3 \
    python3-setuptools \
    python3-pip \
    python3-dev

# Install the core packages required to build librealsense libs
sudo apt-get install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
# Install Distribution-specific packages for Ubuntu 18
sudo apt-get install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

# Install LibRealSense from source
# We need to build from source because
# the PyPi pip packages are not compatible with Arm processors.
# See link [here](https://github.com/IntelRealSense/librealsense/issues/6964).

# First clone the repository
git clone https://github.com/IntelRealSense/librealsense.git
cd ./librealsense

# Make sure that your RealSense cameras are disconnected at this point
# Run the Intel Realsense permissions script
./scripts/setup_udev_rules.sh

# Now the build
mkdir build && cd build
## Install CMake with Python bindings (that's what the -DBUILD flag is for)
## see link: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
## Recompile and install librealsense binaries
## This is gonna take a while! The -j4 flag means to use 4 cores in parallel
## but you can remove it and simply run `sudo make` instead, which will take longer
sudo make uninstall && sudo make clean && sudo make -j4 && sudo make install

## Export pyrealsense2 to your PYTHONPATH so `import pyrealsense2` works
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
```

If this is useful/correct, I would like it to be put in a section of the `README.md` to save other developers' time. Thank you.

dorodnic, 2020-11-05:
Hi @lieuzhenghong 
Thank you for the contribution. Feel free to submit pull-request on GitHub or one of the admins can merge it manually.

MartyG-RealSense, 2020-11-05:
Thanks very much @dorodnic - I will also track this case over time and keep it open until a PR is processed or it is added as documentation.

surefyyq, 2020-11-15:
rs-capture demo works and import pyrealsense2 successfully in python3 environment, but when I call rs.pipeline() and rs.config() it output 'has no attribute' error.
could anyone help please~

MartyG-RealSense, 2020-11-15:
Hi @surefyyq   Could you create a new question for your problem please by visiting the link below and clicking on the **New Issue** button.  Thanks!

https://github.com/IntelRealSense/librealsense/issues

ytfksw, 2020-12-07:
@lieuzhenghong I have encountered the same problem. The documentation needs to be updated. When will this be merged?

MartyG-RealSense, 2020-12-07:
Hi @ytfksw   There is not a time estimate that can be provided for the updating of documentation.  Cases tagged as documentation are kept open in the meantime so that they can be tracked.

MartyG-RealSense, 2021-01-20:
Hi @lamngoctam   The guide in case 6964 has been the most consistently successful for installing librealsense and the Python wrapper together on Jetson using CMake.

https://github.com/IntelRealSense/librealsense/issues/7905#issuecomment-737544294

One RealSense user who followed the installation guide in 6964 and had the 'No module named pyrealsense2' error found that the problem was resolved when they set the correct PYTHONPATH.

https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-713321354

Another RealSense user solved their problems simply by re-flashing their OS.

kithib, 2022-03-17:
hi,when i use this commend sudo apt-get install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
and it telled me that it can't install  libglfw3-dev ,how could i do with this.thanks

MartyG-RealSense, 2022-03-17:
Have you inputed the command below first, please:

`sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade`

kithib, 2022-03-18:
HI @MartyG-RealSense  Thanks for your help. I just followed you ,but it still occured!

`(base) nvidia@nvidia-desktop:~/librealsense/build$ sudo apt-get install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libglfw3-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libglfw3-dev' has no installation candidate
`

MartyG-RealSense, 2022-03-18:
Hi @kithib  As you refer to **nvidia@nvidia-desktop**, are you using Nvidia Jetson as your computing device like @lieuzhenghong  was in their instruction guide above, please?  

Which Ubuntu version are you using, please?  My understanding is that Ubuntu **18.04** is the most recent version supported by Nvidia JetPack at the time of writing this, so if you are using a newer Ubuntu such as 20.04 (Focal) then this may be causing you problems.  An Nvidia support forum member reports that JetPack will add 20.04 support in JetPack 5.

https://forums.developer.nvidia.com/t/jetpack-ubuntu-20-04/203197

kithib, 2022-03-20:
HI @MartyG-RealSense Thanks for your help.I use jetson NX . I solved last problems by downloading in ubuntu webset!
When I run cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
Is that my git problems?
-- Checking internet connection...
-- Failed to identify Internet connection
CMake Warning at CMakeLists.txt:15 (message):
  No internet connection, disabling BUILD_WITH_TM2

CMake Warning at CMakeLists.txt:21 (message):
  No internet connection, disabling IMPORT_DEPTH_CAM_FW

-- Info: REALSENSE_VERSION_STRING=2.50.0
-- Setting Unix configurations
-- Building libcurl enabled
-- using RS2_USE_LIBUVC_BACKEND
-- LIVE555: File exists, delete it to download the latest version (/home/nvidia/librealsense/third-party/live555/live555-latest.tar.gz)
-- LIVE555: Unpacking source
-- LIVE555: Source unpacked
CMake Warning at wrappers/CMakeLists.txt:10 (message):
  No internet connection.  Cloning Python bindings may fail

-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia/librealsense/build/external-projects/pybind11
[ 12%] Performing download step (git clone) for 'pybind11'
Cloning into 'pybind11'...
fatal: unable to access 'https://github.com/pybind/pybind11.git/': Could not resolve host: github.com
Cloning into 'pybind11'...
fatal: unable to access 'https://github.com/pybind/pybind11.git/': Could not resolve host: github.com
Cloning into 'pybind11'...
fatal: unable to access 'https://github.com/pybind/pybind11.git/': Could not resolve host: github.com
-- Had to git clone more than once:
          3 times.
CMake Error at /home/nvidia/librealsense/build/external-projects/pybind11/tmp/pybind11-gitclone.cmake:66 (message):
  Failed to clone repository: 'https://github.com/pybind/pybind11.git'

CMakeFiles/pybind11.dir/build.make:89: recipe for target 'src/pybind11-stamp/pybind11-download' failed
make[2]: *** [src/pybind11-stamp/pybind11-download] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/pybind11.dir/all' failed
make[1]: *** [CMakeFiles/pybind11.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
CMake Error at wrappers/python/CMakeLists.txt:22 (add_subdirectory):
  add_subdirectory given source
  "/home/nvidia/librealsense/build/third-party/pybind11" which is not an
  existing directory.

CMake Error at wrappers/python/CMakeLists.txt:53 (pybind11_add_module):
  Unknown CMake command "pybind11_add_module".

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/librealsense/build/CMakeFiles/CMakeOutput.log".

MartyG-RealSense, 2022-03-20:
Hi @kithib  Does it make a difference if you add **-DBUILD_WITH_TM2=false** to your CMake build instruction to stop the build process from trying to access an internet connection?

`cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DBUILD_WITH_TM2=false`

MartyG-RealSense, 2022-04-08:
Hi @kithib  Do you still require assistance with your installation problem, please?  Thanks!

henryrobbins, 2025-02-09:
Thanks so much! Writing to say these installation steps worked for me on the new Jetson Orin Nano Super. I also followed similar steps to construct a `Dockerfile`. I'm including the relevant parts here.

```
FROM python:3.11-slim AS build

# Prevent interactive inputs when installing packages
ENV DEBIAN_FRONTEND=noninteractive
ENV TZ=America/New_York
ENV SUDO=command

# Install packages
RUN apt-get update && apt-get install -y \
    curl wget unzip cmake make g++ gcc \
    # Required for librealsense and pyrealsense2
    # https://github.com/IntelRealSense/librealsense/issues/7722#issue-736687248
    git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev \
    libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev \
    && rm -rf /var/lib/apt/lists/*

# =============================================================================

# Build librealsense and pyrealsense2 from source
# This is necessary as there is no pre-built wheel for Python 3.11 on ARM64.
# https://github.com/IntelRealSense/librealsense/issues/7722#issue-736687248

# Install
RUN wget https://github.com/IntelRealSense/librealsense/archive/refs/tags/v2.55.1.zip \
    && unzip v2.55.1.zip \
    && rm -f v2.55.1.zip

# Build
RUN cd librealsense-2.55.1 && \
    mkdir build && \
    cd build && \
    cmake ../ \
        -DBUILD_PYTHON_BINDINGS:bool=true \
        -DBUILD_EXAMPLES:BOOL=OFF \
        -DBUILD_GRAPHICAL_EXAMPLES:BOOL=OFF \
        -DBUILD_TOOLS:BOOL=OFF \
        -DBUILD_UNIT_TESTS:BOOL=OFF \
        -DBUILD_WITH_TM2:BOOL=OFF \
        -DPYTHON_EXECUTABLE=/usr/local/bin/python \
        -DPYTHON_INSTALL_DIR=/usr/local/lib/python3.11/site-packages \
    && make -j$(nproc) \
    && make install \
    # Ran into issues with DPYTHON_INSTALL_DIR, so manually copied the .so files
    && cp Release/pyrealsense2.*.so* /usr/local/lib/python3.11/site-packages

# Make pip aware of librealsense2
RUN mkdir -p /usr/local/lib/python3.11/site-packages/pyrealsense2.dist-info \
    && echo "Metadata-Version: 2.1" > /usr/local/lib/python3.11/site-packages/pyrealsense2.dist-info/METADATA \
    && echo "Name: pyrealsense2" >> /usr/local/lib/python3.11/site-packages/pyrealsense2.dist-info/METADATA \
    && echo "Version: 2.55.1.6486" >> /usr/local/lib/python3.11/site-packages/pyrealsense2.dist-info/METADATA
```

In my case, I only needed `pyrealsense2` in my container and was trying to keep it slim. Hence the following flags:

```
-DBUILD_EXAMPLES:BOOL=OFF
-DBUILD_GRAPHICAL_EXAMPLES:BOOL=OFF
-DBUILD_TOOLS:BOOL=OFF
-DBUILD_UNIT_TESTS:BOOL=OFF
-DBUILD_WITH_TM2:BOOL=OFF
```

Also, I did run into some issues with `DPYTHON_INSTALL_DIR` where the generated `.so` files weren't getting installed there after `make install`. I resolved this by manually copying them. 

Lastly, `pip` didn't recognize the `pyrealsense2` package. I hackily manually created `pyrealsense2.dist-info` to resolve that. I'm sure there's a better way -- please let me know!

MartyG-RealSense, 2025-02-10:
Hi @henryrobbins   Thanks so much for sharing the details of the procedure that worked for you!

I note that you are using Python 3.11.  The **aarch64** pip packages for use on Arm architecture devices such as Jetson only currently support up to Python 3.10.  Thanks too for highlighting your workaround.  It is also possible to create .whl packages for not yet supported Python versions by building the pyrealsense2 wrapper from source code and then uploading it as your own Python package.  The link below has a guide for creating and uploading packages to PyPi.

https://www.turing.com/kb/how-to-create-pypi-packages

MartyG-RealSense, 2026-10-23:
This open GitHub issue is being closed due to its significant age.

closed 2026-10-23 by MartyG-RealSense