realsenseai/librealsense #7728: librealsense2-dkms not found

wongjon, 2020-11-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | v2.31 | 
| Operating System & Version |   Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    |  5.3.0-1036-raspi2  | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  2   | 
| Language                            |  C++/pcl     | 
| Segment			|  Robot   | 

### Issue Description

I am trying to install the librealsense2-dkms but it says that the "package is unable to be located." I am following this guide: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md. Everything works up until the point I run ```sudo apt-get install librealsense2-dkms```. I've run ```sudo apt update``` as well as other commands to ensure my repositories are up to date. I am also fairly new to Linux so there might be something that I'm missing. Thank you!

[image]

MartyG-RealSense, 2020-11-06:
Hi @wongjon   I hope that the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/3895#issuecomment-489023921

As you are using Pi 4, the **Acrobotic** guide in the link below is a good one if you are only installing librealsense and not additional wrappers such as ROS.

If you anticipate that you will be using the RealSense ROS wrapper as well as librealsense, I would recommend the **eleccelerator** guide.

https://github.com/IntelRealSense/realsense-ros/issues/1375#issuecomment-691482673

wongjon, 2020-11-06:
Hello MartyG

Thank you for providing some resources. I will try them out. I do notice that #3895 is for the Raspberry Pi 3 and not the Raspberry Pi 4B. Also, the eleccelerator guide you mentioned is for a Raspbian Buster OS and not Ubuntu. Will these make a significant difference? 

I have also found https://github.com/IntelRealSense/librealsense/issues/7082 which seems to have the same problem. Do you know if that issue was resolved since I followed the same link that was given to him.

MartyG-RealSense, 2020-11-06:
Yes, I edited elecellerator out of the comment after its oiginal posting because you are using Ubuntu rather than Raspbian.  So I would recommend focusing on the **Acrobotic** guide if you do not need ROS.

You could also set up librealsense by using a pre-made SD card image for Pi 4 that is supplied by Intel, though I believe that it does not have any wrappers such as Python and ROS included in the build.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-3-preparing-the-sd-card

The mention in https://github.com/IntelRealSense/librealsense/issues/3895#issuecomment-489023921 about x86/AMD64 refers to how Raspberry Pi boards have **Arm** processors rather than the x86/x64 ones typically used in desktop and laptop PCs.  So a package built for x86/AMD64 machines would not work on a newer Pi model if its processor was still an Arm one.

wongjon, 2020-11-12:
I tried the above Acrobatic guide (I skipped the Python since I don't think I need the Python wrapper). If I do a "sudo apt-get install librealsense2-dkms" or a "dpkg -L librealsense" the dkms package is still not found. Instead, what I did to get librealsense2-dkms was to install it on my desktop and transfer the package manually via a USB drive. I moved the package into /usr/src at root. 

Now, when I run the realsense-viewer, the stereo module works but the RGB camera does not work at all and the Gyroscope stops working after about 30 seconds. I really only need the depth camera though. 

Errors that I get when running the stereo module is:  12/11 20:42:50,844 WARNING [281472082633152] (types.cpp:49) Out of frame resources! and  12/11 20:42:50,796 ERROR [281472082633152] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!. 

If I run the RGB module, I get a constant stream of this:  12/11 20:43:50,540 ERROR [281472250388928] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!

When I'm running my algorithm, it sees that it can detect the camera so the dependencies should be correct. However, it looks like my algorithm isn't receiving any information from the camera and thus the algorithm is failing. I'm wondering if you have any ideas why my algorithm can't receive anything from the camera. The algorithm works fine on my desktop so it shouldn't be an issue with the code itself. I have a snapshot of the output below. The serial number for my camera is 846112071342 so it can detect the camera properly. 

[image]
[image]

MartyG-RealSense, 2020-11-13:
Apologies for the delay in responding further, as I was carefully considering your case.  Given that you have all three stream types active (depth, RGB and Motion Module), it looks as though it is not an issue related to **librealsense2-dkms**.  Instead, it seems to be related to a known issue that can occur on some computers in the RealSense Viewer after activating the Motion Module.

It can be worked around when programming in C++ or Python but I am not aware of a fix for the Viewer.  The link below has an example of such a case where the RGB stream has **No Frames Received**.

https://github.com/IntelRealSense/librealsense/issues/7584

wongjon, 2020-11-13:
Thanks for the suggestion. If I restart the realsense viewer and only turn on RGB instead of all 3 it still does not work though. Would the No Frames Received still fix that problem?

MartyG-RealSense, 2020-11-13:
You may need to reboot the computer to correct the problem once it occurs the first time after starting all 3 streams, as the problem with the RGB can persist if you exit the Viewer and then relaunch it, even if you then only select the RGB.

wongjon, 2020-11-17:
I noticed that in the ticket #7584 that you linked mentioned "There are fixes that can be programmed if creating your own applications in C++ or Python." Since I am creating my own application, what are the fixes that can be implemented?

MartyG-RealSense, 2020-11-17:
You can either have two separate pipelines (IMU on one pipeline and depth / color on the other) or use **callbacks**.  For Python, separate pipelines tends to work best, whilst callbacks are the preferred choice of C++ RealSense users.

Here is a Python example of separate pipelines for IMU and for depth / color:

https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238

https://github.com/IntelRealSense/librealsense/issues/6031#issuecomment-599072265

C++ callback example:

https://github.com/IntelRealSense/librealsense/tree/master/examples/callback

MartyG-RealSense, 2020-12-05:
Hi @wongjon   Do you require further assistance with this case, please?   Thanks!

wongjon, 2020-12-05:
No, the callback example worked. Thank you so much for the help!

closed 2020-12-05 by wongjon

MartyG-RealSense, 2020-12-05:
Great news @wongjon - thanks very much for the update!

ev-mp, 2021-08-18:
The issue reported by @wongjon was effectively fixed in v2.43.0 via https://github.com/IntelRealSense/librealsense/pull/8328
(DSO-15250)