realsenseai/librealsense #7729: librealsense2-using vcpkg

Zlavine, 2020-11-06:
### Issue Description
I am trying to develop my D415 camera, so I bulid librealsense2 by using vcpkg according to the development document. Now I have some question.

> 1.  I found that  some third-party files( imgui, glfw and so on )  in realsense2 file building through vcpkg, but I can`t include these third-party files in VS2019, as well as I have included librealsense2, So, do I need to install these additional third-party libraries?

> 2.  I`ve tried installing third-party libraries(imgui, glfw, opengl) via vcpkg, but I can`t find the flies need for the examples("init.c" "imgui_impl_glfw.cpp" "imgui_impl_glfw.h" and so on ).

MartyG-RealSense, 2020-11-07:
Hi @Zlavine   I carefully researched your case.  It may be best, I believe, to treat it as a question relating to Visual Studio library linking rather than a vcpkg-related question.  You may therefore find the link below useful for setting up linkages in Visual Studio when creating your own project.

https://github.com/IntelRealSense/librealsense/issues/7106#issuecomment-678922995

MartyG-RealSense, 2020-11-23:
Case closed due to no further comments received.

closed 2020-11-23 by MartyG-RealSense

freepeasantry2021, 2021-12-06:
from the readme.md, seems it's not related with vs.
Just follow the readme.md then encounter it.
Need update readme or other action?