realsenseai/librealsense #7730: How do I save some of the data I need?

Chat1ng, 2020-11-06:
I got a realsense SR300, which was successfully connected to my computer and worked properly.Then I installed Intel RealSense SDK Sample Browser (2016 R2). In the SR300 Sample, I found the Eye Tracking module and Facial Landmark (78 Point) module I wanted and successfully ran them.

I would like to know that in the next step, I need to use the module mentioned above to record and save the Facial Landmark point with 78 Facial Landmark points 1) color video stream and 2) depth video stream),and 3) the three-dimensional coordinates of 78 feature points of the face in the video stream,4)world coordinates/head World coordinates/head deflections, etc., what do I need to do?Is there anything I can refer to?

MartyG-RealSense, 2020-11-06:
Hi @Chat1ng  The discussion in the link below has a zip file attached to its page that contains an archive of the official Intel PDF programming manuals from 2014 for that SDK.

https://community.intel.com/t5/Items-with-no-label/RealSense-2014-SDK-manuals-and-script-samples/td-p/482383?profile.language=en 

The official documentation website for 2016 R2 is at the link below:

https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_devguide_introduction.html

MartyG-RealSense, 2020-11-13:
Hi @Chat1ng   Do you still require assistance with this case, please?   Thanks!

Chat1ng, 2020-11-13:
I'm trying. Things should get better.Thank you very much!

MartyG-RealSense, 2020-11-13:
Thanks very much @Chat1ng for the update.  I will close this case now.   Please do feel welcome to make a new case if you have another question in the future.  Good luck!

closed 2020-11-13 by MartyG-RealSense